MagicMirrorOrg/MagicMirror #3423: Initializing buttons using 'MMM-Buttons' for notification sending to 'MMM-StopwatchTimer' induces an error :(

dummmylitt, 2024-04-10:
Hello all,

I have tried uninstalling and installing sdetweil's Buttons repository as well as rewiring my button to different GPIO pins but all lead to the same result: an error that I have a really hard trouble deciphering and cannot find the solution anywhere else so I decided to try here. Any help is so appreciated as I am at my wit's end. 

`[ERROR] Whoops! There was an uncaught exception... 
[2024-04-09 21:41:42.803] [ERROR] Error: EINVAL: invalid argument, write
    at Object.writeSync (node:fs:917:3)
    at Object.writeFileSync (node:fs:2294:26)
    at exportGpio (/home/group6/MagicMirror/modules/MMM-Buttons/node_modules/onoff/onoff.js:18:8)
    at new Gpio (/home/group6/MagicMirror/modules/MMM-Buttons/node_modules/onoff/onoff.js:172:36)
    at Class.intializeButton (/home/group6/MagicMirror/modules/MMM-Buttons/node_helper.js:74:19)
    at Class.intializeButtons (/home/group6/MagicMirror/modules/MMM-Buttons/node_helper.js:90:18)
    at Class.socketNotificationReceived (/home/group6/MagicMirror/modules/MMM-Buttons/node_helper.js:26:18)
    at Socket.<anonymous> (/home/group6/MagicMirror/js/node_helper.js:98:11)
    at Socket.emit (node:events:514:28)
    at Socket.emitUntyped (/home/group6/MagicMirror/node_modules/socket.io/dist/typed-events.js:69:22)
    at /home/group6/MagicMirror/node_modules/socket.io/dist/socket.js:704:39
    at process.processTicksAndRejections (node:internal/process/task_queues:77:11) {
  errno: -22,
  syscall: 'write',
  code: 'EINVAL'
} 
[2024-04-09 21:41:42.804] [ERROR] MagicMirror² will not quit, but it might be a good idea to check why this happened. Maybe no internet connection? 
[2024-04-09 21:41:42.805] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MagicMirrorOrg/MagicMirror/issues 
[2024-04-09 21:41:42.806] [INFO]  updatenotification: Updater Class Loaded! 
`

Thank you <3

sdetweil, 2024-04-10:
I posted a reply in the forums..

as MM doesn't use gpio itself, this is a module problem..

khassel, 2024-06-28:
unrelated to mm core

closed 2024-06-28 by khassel