MagicMirrorOrg/MagicMirror #3424: [weather] openweathermap One Call 2.5 will be finally closed in June 2024

khassel, 2024-04-17:
via mail:

"Dear OpenWeather user,

We would like to inform you about our plans to complete the migration from One Call 2.5 to One Call 3.0 that started 2 years ago. In line with this, access to One Call 2.5 will be finally closed in June 2024. 

One Call 3.0 provides you with significantly advanced business-demanded features such as a minute forecast for 1 hour, 4 days forecast for any timestamp, a daily forecast for 8 days, including a human-readable summary, historical data (45-year archive) for timestamps, and with daily aggregation, 1.5-year long-term forecast with daily aggregation, and others."

We have still 2.5 in the provider file as default, must be changed to 3.0.

mlcampbe, 2024-04-26:
I too got the email about the change to the onecall 3.0 and have tried to do some testing and it seems that we need to change the apiVersion from 2.5 to 3.0 but we also need to change the weatherEndpoint from /weather to /onecall (at least when using the default type=current).

For example, this call works fine:
curl -v "https://api.openweathermap.org/data/3.0/onecall?lat=xxx&lon=yy&appid=zzzz"

but this fails with a 404 error:
curl -v "https://api.openweathermap.org/data/3.0/weather/?lat=xxx&lon=yy&appid=zzzz"

Is this correct in that we need to change both the apiVersion and the weatherEndpoint parameters to use the 3.0 calls after June?

sdetweil, 2024-04-26:
this issue says, we need to change to fix the coming api change.  whatever  it takes.

you have pointed out another thing to address.

dkallen78, 2024-07-04:
Oh hey, I did a pull request for this! Easy change (I think)

rejas, 2024-08-09:
Seems the old API is shutdown now for good. Switching to the new one isnt working outofthebox, so I think this needs a little more work:

https://openweathermap.org/api shows only the onecall api being free. 
- so the weatherendpoint needs to be per default "/onecall" since the old wont work

https://openweathermap.org/api/one-call-3 shows a much simpler params:
- only lat/long, but no location can be given

Can anyone confirm my observations? Or does openweathermap work for you?

sdetweil, 2024-08-09:
I think changing the endpoint to /onecall with v3.0 in the url works the same

closed 2024-09-30 by khassel