MagicMirrorOrg/MagicMirror #3441: Error magicmirror 2.27 after upgrade - Blank black screen

NeelChandra001, 2024-05-03:
Hello
--
## I found an ERROR in MagicMirror in my installation.

**Platform**: npm - 10.1.0  hardware - Raspberry Pi 3 b.
**Node Version**:  20.
**MagicMirror² Version**: 2.27
**Description**:  screen shot attached below.
[image]

**Steps to Reproduce**: started MagicMirror 2.27 with the npm command - npm start run
**Expected Results**: MagicMirror 2.27 was expected to be up and running without manual intervention or fixing any errors.

**Actual Results**: 

Error occurs when MagicMirror is starting ... shown below ...

[2024-05-03 16:07:56.930] [LOG]   [MMM-ClearLog] Starting node helper 
[2024-05-03 16:07:56.931] [LOG]   Connecting socket for: MMM-RRS 
[2024-05-03 16:07:56.933] [LOG]   [MMM-RRS] Starting node helper 
[2024-05-03 16:07:56.935] [LOG]   Connecting socket for: EXT-YouTubeCast 
[2024-05-03 16:07:56.936] [LOG]   Starting module helper: EXT-YouTubeCast 
[2024-05-03 16:07:56.938] [LOG]   Connecting socket for: MMM-TouchSwipe 
[2024-05-03 16:07:56.940] [LOG]   Starting module helper: MMM-TouchSwipe 
[2024-05-03 16:07:56.944] [LOG]   Sockets connected & modules started ... 
[2024-05-03 16:07:57.884] [LOG]   Launching application. 
****[2024-05-03 16:08:07.005] [INFO]  System information:
### SYSTEM:   manufacturer: Raspberry Pi Foundation; model: Raspberry Pi 3 Model B Plus Rev 1.3; raspberry: [object Object]; virtual: false
### OS:       platform: linux; distro: Raspbian GNU/Linux; release: 11; arch: arm; kernel: 6.1.47-v7+
### VERSIONS: electron: 29.3.1; used node: 20.9.0; installed node: 20.8.0; npm: 10.1.0; pm2: 5.3.1
### OTHER:    timeZone: Australia/Sydney; ELECTRON_ENABLE_GPU: undefined**** 

--> MagicMirror 2.27 is stuck on black screen.

**Configuration**: config.js is clean and working.

**Additional Notes**: upgraded the MagicMirror to version 2.27 and node versio 20 using the script provided raspberrypi.sh script on the web.

Thanks.

KristjanESPERANTO, 2024-05-03:
Unfortunately, I do not see any error message. The part in blue is system information and not an error message.

Does it work if you deactivate all modules except for the clock?

sdetweil, 2024-05-03:
also post the 
~/MagicMirror/installers/upgrade.log

are you using pmw to launch h and this US the stdout.log
if using pm2 do
```
pm2 stop all
cd ~/MagicMirror 
npm start
```

typically black screen is a. module missing a dependency

NeelChandra001, 2024-05-06:
Hi - I will try again - re-installing MM - Thank you.

sdetweil, 2024-05-06:
@NeelChandra001 no. install won't help
 we removed the libraries the module needs  because they are no longer used by the base MagicMirror 

the module needs to be fixed.  but old so it likely won't

you can get around  by installing the missing libs.    or running my upgrade script with force as the parameter instead of apply  and my code will check and install missing libs we know about

khassel, 2024-06-24:
As already mentioned, the black screen is caused by a module. For further troubleshooting please use the forum.

closed 2024-06-24 by khassel