MagicMirrorOrg/MagicMirror #3447: newsfeed enclosure feature

bugsounet, 2024-05-10:
Hi, i plan to add to newsfeed default module an enclosure picture of the item displayed.

What do you think about this feature?

sdetweil, 2024-05-10:
more info.. 
enclosure picture of the item displayed

doesn't mean anything to me..

give examples..
does this mean it can't fit on the bottom_center anymore (too tall?)

bugsounet, 2024-05-12:
The picture like this (example of one of my modules):

[image]

many feed have an enclosured picture related of article why not display it ?

sdetweil, 2024-05-12:
ok, got it now.

I personally don't think the base should take on this kind of technical debt. That's what the 3rd party ecosystem is for. we give examples of what can be done in a functional way, we don't do every solution.

bugsounet, 2024-05-12:
ok, it's just a proposal ;)

rejas, 2024-05-13:
I wouldnt close it so fast, cause I like the idea. We always add new functionality to our code modules, so why not this? Maybe disable it per default, so the "slick" look is preserved.

bugsounet, 2024-05-18:
@rejas: Ok, i will see