MagicMirrorOrg/MagicMirror #3465: Compliments module special days config

WallysWellies, 2024-06-13:
I asked this on the forums and was advised to log a bug.

**Platform**: RaspberryPi 5 running raspberryPi OS. Current version of MagicMirror software.

When adding config to enable the "special days" feature of the default compliments module it will seemingly add that days comment to the existing array. In my case that array is considerable so special messages such as "Happy Birthday" or "Merry Christmas" etc. are lost in a sea of compliments.

My expectation of this feature would be that on the configured day it shows only the compliments specified in that area of the config. For example:

`"....-12-25": ["Happy Christmas!"]`

I would expect the only compliment on the 25th December to be "Happy Christmas!". That is not the case though. I am currently adding a line of code in the compliments.js file to empty the array on a special day and only include the configured comments:

`// Add compliments for special days`
`for (let entry in this.config.compliments) {`
`  if (new RegExp(entry).test(date)) {`
`    compliments = []; // Empty the array if it's a special day in config.js`
`    Array.prototype.push.apply(compliments, this.config.compliments[entry]);`
`  }`
`}`

This is not necessarily a "bug" as it may well be the intended behaviour, however I would suggest making this clear in the docs and, if I may be so bold, adding a new config option that allows this behaviour if the user wants it. I am not confident in my coding or GitHub skills to suggest such a change but it seems like it could be quite simple.

WallysWellies, 2024-06-13:
Forum link: [https://forum.magicmirror.builders/topic/18793/default-compliments-module-special-days](url)

sdetweil, 2024-06-15:
I agree with your assessment.  if you have a specific  event list  they should be not be combined with the general list.

rejas, 2024-06-16:
I agree also. Do you want to create a PR for that yourself with some help?

WallysWellies, 2024-06-16:
I would be happy to give it a go! Let me do some testing and reading up on how to GitHub and I’ll get back to you in the next couple of days.

Cheers for the encouragement :)

WallysWellies, 2024-06-17:
I have done what I think I should have - forked, edited, and submitted a pull request referencing this issue. Please let me know if I have done something wrong.

I have another PR ready to go that updates the documentation with the proposed change but I'll hold fire on that until I know if I am doing this right.

PR #3471 (2024-06-24): Update compliments module
`Fixes #3465`

Add config option `specialDayUnique` that defaults to `false` and causes special day compliments to be added to the existing compliments array. Setting this option to `true` will only show the compliments that have been configured for that day.
commit aefb3a0b (2024-06-24): Update compliments module (#3471)

`Fixes #3465`

Add config option `specialDayUnique` that defaults to `false` and causes
special day compliments to be added to the existing compliments array.
Setting this option to `true` will only show the compliments that have
been configured for that day.

---------
closed 2024-06-30 by khassel