MagicMirrorOrg/MagicMirror #3466: OpenMeteo weather provider uses "apparent temperature" in forecast data instead of actual temperatures

btoconnor, 2024-06-14:
## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Running magicmirror in a vm, display is on a rpi4

**Node Version**: v20.11.1

**MagicMirror² Version**: 2.27.0

**Description**:

The OpenMeteo provider uses the "apparent temperature" for the forecast temperature, max and min temperatures.  The apparent temperature is the "feels like", which is different from the actual temperature.

**Steps to Reproduce**: Set up a forecast module with openmeteo as the provider.  Temperatures will differ from actual temperatures.

**Expected Results**: I expect to get real temperatures.

**Actual Results**: I get the "feels like" / "apparent" temperatures

**Additional Notes**:

The code to fix this is straightforward - I'm happy to open a PR.  I've already fixed it locally.  This was opened in #3273 but was closed by the author without a fix.

```javascript
diff --git a/modules/default/weather/providers/openmeteo.js b/modules/default/weather/providers/openmeteo.js
index 5c4b2322..3a6be27a 100644
--- a/modules/default/weather/providers/openmeteo.js
+++ b/modules/default/weather/providers/openmeteo.js
@@ -398,9 +398,9 @@ WeatherProvider.register("openmeteo", {
                        currentWeather.windFromDirection = weather.winddirection_10m_dominant;
                        currentWeather.sunrise = weather.sunrise;
                        currentWeather.sunset = weather.sunset;
-                       currentWeather.temperature = parseFloat((weather.apparent_temperature_max + weather.apparent_temperature_min) / 2);
-                       currentWeather.minTemperature = parseFloat(weather.apparent_temperature_min);
-                       currentWeather.maxTemperature = parseFloat(weather.apparent_temperature_max);
+                       currentWeather.temperature = parseFloat((weather.temperature_2m_max + weather.temperature_2m_min) / 2);
+                       currentWeather.minTemperature = parseFloat(weather.temperature_2m_min);
+                       currentWeather.maxTemperature = parseFloat(weather.temperature_2m_min);
                        currentWeather.weatherType = this.convertWeatherType(weather.weathercode, currentWeather.isDayTime());
                        currentWeather.rain = parseFloat(weather.rain_sum);
                        currentWeather.snow = parseFloat(weather.snowfall_sum * 10);
```

khassel, 2024-06-14:
> I'm happy to open a PR. I've already fixed it locally

Thanks!

Looking at the code I think we have the same problem with current temp:

```js
			currentWeather.temperature = parseFloat(weather.apparent_temperature);
			currentWeather.minTemperature = parseFloat(weathers.daily[h].apparent_temperature_min);
			currentWeather.maxTemperature = parseFloat(weathers.daily[h].apparent_temperature_max);
```

So if you provide a PR for forecast can you please include the same change for current?

btoconnor, 2024-06-14:
Happy to, yeah.  I'll get that posted soon.

closed 2024-06-30 by khassel