MagicMirrorOrg/MagicMirror #3467: Repeating calendar event shifted by 1 day when RRule is processed

morozgrafix, 2024-06-15:
## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Raspberry Pi 2 - Raspbian GNU/Linux 12 (bookworm). Also reproduced and debugged on OS X - Sonoma 14.5 with 125.0.6422.142 

**Node Version**: v20.14.0

**MagicMirror² Version**: 2.27.0

**Description**:  Repeating weekly calendar event is offest by 1 day when processed by Magic Mirror. For example when event is scheduled on Thursday in the calendar, it appears to be scheduled for Friday when displayed on MM. I ran through the `debug.js` and it seems that RRules are affecting it. Full logs and steps to reproduce are below. Possibly related to #3342 and #3291 fixes. My MM installations are set to `America/Los_Angeles` timezone. 
The calendar is set in the same timezone, event is set in the same timezone. `modules/default/calendar/calendarfetcherutils.js` around line 300 seems to be adding an extra day.

**Steps to Reproduce**: 
1 . Add following ics file to the calendar config or as `url` variable in `debug.js` [Gist for basic.ics](https://gist.githubusercontent.com/morozgrafix/71379538d328e565130a89d6eb6d3137/raw/32ad67a4d0e76ceaab54e1d3d5bd55d65dc494b4/basic.ics)
2. Launch MM and observe the `Test Entries` to show up on Fridays at 17:30 or run `modules/default/calendar/debug.js` and observe the output.

**Expected Results**: Repeating calendar events to be displayed to occur on Thursday at 17:30

**Actual Results**: Events are displayed to occur on Fridays at 17:30

**Configuration**:  Plain vanilla `config.js` based on `config.js.sample` with only addition of the sample calendar:
```
/* Config Sample
 *
 * For more information on how you can configure this file
 * see https://docs.magicmirror.builders/configuration/introduction.html
 * and https://docs.magicmirror.builders/modules/configuration.html
 *
 * You can use environment variables using a `config.js.template` file instead of `config.js`
 * which will be converted to `config.js` while starting. For more information
 * see https://docs.magicmirror.builders/configuration/introduction.html#enviromnent-variables
 */
let config = {
	address: "localhost",	// Address to listen on, can be:
	// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
	// - another specific IPv4/6 to listen on a specific interface
	// - "0.0.0.0", "::" to listen on any interface
	// Default, when address config is left out or empty, is "localhost"
	port: 8080,
	basePath: "/",	// The URL path where MagicMirror² is hosted. If you are using a Reverse proxy
	// you must set the sub path here. basePath must end with a /
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"],	// Set [] to allow all IP addresses
	// or add a specific IPv4 of 192.168.1.5 :
	// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
	// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
	// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false,			// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "",	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "",	// HTTPS Certificate path, only require when useHttps is true

	language: "en",
	locale: "en-US",
	logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
	timeFormat: 24,
	units: "metric",

	modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
[... 189 lines cut ...]
BEGIN:VEVENT
DTSTART;TZID=America/Los_Angeles:20230803T173000
DTEND;TZID=America/Los_Angeles:20230803T190000
RRULE:FREQ=WEEKLY
EXDATE;TZID=America/Los_Angeles:20230817T173000
EXDATE;TZID=America/Los_Angeles:20230914T173000
EXDATE;TZID=America/Los_Angeles:20230921T173000
EXDATE;TZID=America/Los_Angeles:20230928T173000
EXDATE;TZID=America/Los_Angeles:20231026T173000
EXDATE;TZID=America/Los_Angeles:20231123T173000
EXDATE;TZID=America/Los_Angeles:20231207T173000
EXDATE;TZID=America/Los_Angeles:20240215T173000
EXDATE;TZID=America/Los_Angeles:20240222T173000
EXDATE;TZID=America/Los_Angeles:20240321T173000
DTSTAMP:20240614T041122Z
UID:78C53B9F-2152-4824-838D-F52000D3332B
CREATED:20230801T193617Z
LAST-MODIFIED:20240321T191415Z
SEQUENCE:0
STATUS:CONFIRMED
SUMMARY:Test Event
TRANSP:OPAQUE
END:VEVENT
END:VCALENDAR
```

sdetweil, 2024-06-15:
yes the rrule processor has trouble with some date combinations.

not our code. and we can't tell as we don't process the ics file in MagicMirror code

morozgrafix, 2024-06-15:
I did a little bit more digging and the extra day is added in `calendarfetcherurils.js` around line 300:

`// RRule can generate dates with an incorrect recurrence date. Process the array here and apply date correction.
					if (hasByWeekdayRule) {
						Log.debug("Rule has byweekday, checking for correction");
						dates.forEach((date, index, arr) => {
							// NOTE: getTimezoneOffset() is negative of the expected value. For America/Los_Angeles under DST (GMT-7),
							// this value is +420. For Australia/Sydney under DST (GMT+11), this value is -660.
							const tzOffset = date.getTimezoneOffset() / 60;
							const hour = date.getHours();
							if ((tzOffset < 0) && (hour < -tzOffset)) { // east of GMT
								Log.debug(`East of GMT (tzOffset: ${tzOffset}) and hour=${hour} < ${-tzOffset}, Subtracting 1 day from ${date}`);
								arr[index] = new Date(date.valueOf() - oneDayInMs);
							} else if ((tzOffset > 0) && (hour >= (24 - tzOffset))) { // west of GMT
								Log.debug(`West of GMT (tzOffset: ${tzOffset}) and hour=${hour} >= 24-${tzOffset}, Adding 1 day to ${date}`);
								arr[index] = new Date(date.valueOf() + oneDayInMs);
							}
						});
			 `

if I’m not mistaken the time zone is set to null just before that code is executed. I think if calendar/event is matching the local time zone the adjustment shouldn’t be made. I tried to comment out the code above and it displays the date correctly, but of course it breaks some of the existing tests. I think that it needs additional checks before execution. Maybe checking that event timezone doesn’t match local timezone (although I didn’t dig into the code much yet)

thanks

sdetweil, 2024-06-15:
that's my code trying to deal with this problem.  no good fix yet

morozgrafix, 2024-10-05:
Bump, too keep it open as it’s still an issue.

Pollard-J, 2024-10-07:
yea ive just updated mine and it is still displaying full day events as 2 days.

sdetweil, 2024-10-07:
see this.. waiting on user test for submitting PR here and node-ical
all testcases in both run successfully..

here is a test version of the fixes for all kinds of date problems.

best to make a new folder and git clone there

git  clone https://github.com/sdetweil/MagicMirror
git checkout fixcaldates
npm run install-mm
copy your config.js and custom.css from the prior folder
and the non-default modules you have installed..

this ONLY changes the default calendar
but DOES ship an updated node-ical library too

if you need to fall back, just rename the folders around again so that
your original is called MagicMirror

all the testcases for node-ical and MagicMirror execute successfully.

the 'BIG' change here is to get the local NON-TZ dates for the
rrule.between()

all the checking and conversion code is commented out or not used
the node-ical fixes are for excluded dates (exdate) values being adjusted for DST/STD time.. waiting to submit that PR

one fix in calendar.js for checking if a past date was too far back,
but it never checked to see IF the event date was in the past.. (before today) so it chopped off too many

and one change in calendarfetcher.js to put out a better diagnostic message of the parsed data.. (exdate was excluded cause JSON stringify couldn't convert the complex structure)

sdetweil, 2024-10-07:
@Pollard-J can u try my new fork?

and post a calendar event I can test with.. thx
BEGIN:VEVENT
...
END:VEVENT

Pollard-J, 2024-10-08:
@sdetweil  i just tried this but i'm not that good when i come into problems. I cloned your file of MagicMirror, but it wouldn't execute the next 2 lines properly.
git checkout fixcaldates
npm run install-mm

ill just wait till its released. thank you for trying to help me.

sdetweil, 2024-10-08:
sorry i left out a critical step
after the git clone 
``` 
cd MagicMirror 
```

morozgrafix, 2024-10-09:
@sdetweil thanks for the fix. I just tested the issue and `fixcaldates` branch is displaying events correctly. I only tested with `config.js` from original description of this bug. Repeated events should occur every Thursday at 17:30. Below are screenshots from `master` (not fixed) and `fixcaldates` (fixed) branches. Thank you for digging into this issue, hopefully other issues can also be tested on that branch and your PR goes through soon.

Before:
[image]

After:
[image]

sdetweil, 2024-10-09:
cool, thx for the feedback. 

we wouldn't release til january 1. so keep using and report anything else you see i need to fix

sdetweil, 2024-10-11:
fixed a bunch more testcases..  updated tests

both mm and node-ical changed

```
cd MagicMirror
git pull
rm -rf node_modules
npm run install-mm
```

sdetweil, 2024-10-17:
try again, note ne branch name 

here is an updated test version of the fixes for all kinds of calendar date problems.

NOTE: the changed branch name
NOTE: this used the node-cal@0.19.0 library UNCHANGED

best to make a new folder and git clone there

git clone https://github.com/sdetweil/MagicMirror
cd MagicMirror
git checkout fixcaldates2 // <------ note this is a changed branch name
npm run install-mm
copy your config.js and custom.css from the prior folder
and the non-default modules you have installed…

this ONLY changes the default calendar
but DOES ship an updated node-ical library too

if you need to fall back, just rename the folders around again so that
your original is called MagicMirror

all the testcases for node-ical and MagicMirror execute successfully.

the ‘BIG’ change here is to get the local NON-TZ dates for the
rrule.between()

all the checking and conversion code is commented out or not used
the node-ical fixes are for excluded dates (exdate) values being adjusted for DST/STD time… waiting to submit that PR

one fix in calendar.js for checking if a past date was too far back,
but it never checked to see IF the event date was in the past… (before today) so it chopped off too many

and one change in calendarfetcher.js to put out a better diagnostic message of the parsed data… (exdate was excluded cause JSON stringify couldn’t convert the complex structure)

I added the tests you all have documented

please re-pull and checkout the new branch (I deleted the old branch)
and npm run install-mm again

bdalton12, 2024-12-02:
My research into my buggy MagicMirror calendar dates has led me here.  Great to see there are people working on this!

sdetweil, 2024-12-02:
Try the code 

 coming Jan 1

closed 2025-01-01 by sdetweil