MagicMirrorOrg/MagicMirror #3480: [updatenotification] use of this module creates pm2 daemon process(es)

khassel, 2024-06-24:
When running mm with updatenotification module (or running tests where this module is active) we get

```bash
node@ec51cfe176dd:/opt/magic_mirror$ npx jest tests/e2e/env_spec.js
 PASS   e2e  tests/e2e/env_spec.js (7.22 s)
  App environment
    ✓ get request from http://localhost:8080 should return 200 (217 ms)
    ✓ get request from http://localhost:8080/nothing should return 404 (27 ms)
    ✓ should show the title MagicMirror² (204 ms)

Test Suites: 1 passed, 1 total
Tests:       3 passed, 3 total
Snapshots:   0 total
Time:        7.456 s
Ran all test suites matching /tests\/e2e\/env_spec.js/i.

node@ec51cfe176dd:/opt/magic_mirror$ ps aux
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
node         1  0.0  0.0   2472   884 pts/0    Ss   22:52   0:00 /usr/bin/tini -- ./entrypoint.sh bash
node         7  0.0  0.0   4188  3384 pts/0    S    22:52   0:00 bash
node       193  3.6  0.8 1039160 66032 ?       Ssl  22:53   0:00 PM2 v5.4.1: God Daemon (/home/node/.pm2)
node       200  3.1  0.7 1006164 64880 ?       Ssl  22:53   0:00 PM2 v5.4.1: God Daemon (/home/node/.pm2)
node       211  0.0  0.0   8088  3868 pts/0    R+   22:53   0:00 ps aux
```

@bugsounet  I'm not familiar with pm2, so my question is: Can we avoid these `God Daemon` processes?

bugsounet, 2024-06-26:
sure it's not really logical to have 2 `pm2` process running

I will inspect

bugsounet, 2024-06-26:
```sh
bugsounet@PC-Bugsounet:~/MagicMirror/modules$ npx jest tests/e2e/env_spec.js && grep pm2
 PASS   e2e  ../tests/e2e/env_spec.js
  App environment
    ✓ get request from http://localhost:8080 should return 200 (37 ms)
    ✓ get request from http://localhost:8080/nothing should return 404 (14 ms)
    ✓ should show the title MagicMirror² (201 ms)

Test Suites: 1 passed, 1 total
Tests:       3 passed, 3 total
Snapshots:   0 total
Time:        1.802 s
Ran all test suites matching /tests\/e2e\/env_spec.js/i.
```

```
bugsounet@PC-Bugsounet:~/MagicMirror/modules$ ps aux | grep PM2
bugsoun+    3687  0.0  0.7 1299924 61668 ?       Ssl  17:02   0:02 PM2 v5.4.1: God Daemon (/home/bugsounet/.pm2)
bugsoun+   84793  0.0  0.0   9104  2304 pts/0    S+   20:31   0:00 grep --color=auto PM2
bugsounet@PC-Bugsounet:~/MagicMirror/modules$ 
```

I don't create any process with PM2.
I just check pm2 app presence and check if MM² is on pm2 "status" list
--> if condition match set "flag" to use PM2 for restart MM² (with MM² process id in PM2)

So, i don't know why you have double process, maybe double install of PM2 ??? (I don't think PM2 must avoid this)

khassel, 2024-06-26:
Thanks so far, will test this again ...

closed 2024-06-28 by khassel