MagicMirrorOrg/MagicMirror #3492: [core] cleanup questions

khassel, 2024-07-01:
Looking through the code I found 2 directories which are not used by mm core:

- `installers` with file `mm.sh`, comment from inside this file: `This file is still here to keep PM2 working on older installations.`
- `splashscreen` with some files which I think were used in older install scripts

So maybe @sdetweil or @rejas can tell me for what they are still in this repo. If they are needed by the install script from @sdetweil can they be moved into that repo?

Thanks!

sdetweil, 2024-07-01:
both are used by the installer.. I wiill move them .. and remove the folder .. scripts will add it back as required

sdetweil, 2024-07-01:
will have to look at the doc too.. I think one might be referenced there..

khassel, 2024-07-01:
found no related things in the doc repo, `mm.sh` is mentioned but in a common way and not that it exists in the mm repo. So nothing to do there.

sdetweil, 2024-07-01:
cool.. thx

closed 2024-07-01 by khassel