MagicMirrorOrg/MagicMirror #3503: [WARN] No /home/*user*/MagicMirror/js/../modules/MMM-Jast/MMM-Jast.js found for module: MMM-Jast

CeletroZ, 2024-07-20:
Hi I'm very new to Magic Mirror, and I'm trying to get new modules for my display. I've run through the "Installing the Module" and gone through npm install. When I paste the module into the js it looks like it's not being recognized? 

Please help!

sdetweil, 2024-07-20:
seems wrong place 
see this

https://forum.magicmirror.builders/topic/4231/how-to-add-modules-for-absolute-beginners

khassel, 2024-07-20:
from the issue template:

>## I'm having troubles installing or configuring MagicMirror
>Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

closed 2024-07-20 by khassel