MagicMirrorOrg/MagicMirror #3505: fix logging of systeminformation for raspberry pi

khassel, 2024-07-22:
Our systeminformation on startup on a raspberry pi:

```bash
[2024-07-22 21:03:15.102] [INFO]  System information:
### SYSTEM:   manufacturer: Raspberry Pi Foundation; model: Raspberry Pi 4 Model B Rev 1.5; raspberry: [object Object]; virtual: false
### OS:       platform: linux; distro: Debian GNU/Linux; release: 12; arch: arm64; kernel: 6.1.21-v8+
### VERSIONS: electron: 31.2.1; used node: 20.15.0; installed node: 22.4.1; npm: 10.8.1; pm2:
### OTHER:    timeZone: Europe/Berlin; ELECTRON_ENABLE_GPU: undefined
```

So the object `raspberry: [object Object]` should be corrected. Using `JSON.stringify` results in

```json
{"manufacturer":"Sony UK","processor":"BCM2711","type":"4B","revision":"1.5"}
```

Options:
- we use `JSON.stringify` and add the info to the line
- we omit these infos because they are not of interest
- ?

I'm for removing the object, what do you think @sdetweil @rejas @KristjanESPERANTO ?

rejas, 2024-07-22:
I'd go for the stringify

KristjanESPERANTO, 2024-07-23:
I don't know if the information is really helpful, the information in the model field seems to be enough. From my point of view, we could remove it.

If we do stringify, the line will be too long, so we should add a line break somewhere. Maybe like this:

```sh
[2024-07-22 21:03:15.102] [INFO]  System information:
### SYSTEM:    manufacturer: Raspberry Pi Foundation; model: Raspberry Pi 4 Model B Rev 1.5; virtual: false
### raspberry: "manufacturer":"Sony UK","processor":"BCM2711","type":"4B","revision":"1.5"
### OS:        platform: linux; distro: Debian GNU/Linux; release: 12; arch: arm64; kernel: 6.1.21-v8+
### VERSIONS:  electron: 31.2.1; used node: 20.15.0; installed node: 22.4.1; npm: 10.8.1; pm2:
### OTHER:     timeZone: Europe/Berlin; ELECTRON_ENABLE_GPU: undefined
```

rejas, 2024-07-28:
ok, then lets remove it

PR #3507 (2024-07-29): remove raspberry object from systeminformation
fixes #3505
closed 2024-09-30 by khassel