MagicMirrorOrg/MagicMirror #3521: Improvement to #3502

ryan-d-williams, 2024-08-12:
I discovered the issue described in #3502 this morning and applied a patch locally. While creating an issue with the fix, I noticed it had already been solved. Looking over the [commit](https://github.com/MagicMirrorOrg/MagicMirror/commit/780e4e2e064b40ae10fb7faca536b440502433d3), it looks like that fix was applied pretty far downstream (by downstream I mean the function stack is `start` -> `loadModules` -> `loadModule`) and inside `loadModule` we are performing an unnecessary file system operation. It seems as though we can prevent this loading of the same module well upstream in the `start` function with a one-line change (this is from lines ~260 in `app.js` with the only change being the new if statement - the rest of the code is copied for readability):
```javascript
for (const module of config.modules) {
  if (module.disabled) continue;
  if (module.module) {
  if (Utils.moduleHasValidPosition(module.position) || typeof (module.position) === "undefined") {
    if(!modules.includes(module.module)){  // ONLY NEW ADDITION
      modules.push(module.module);
    }
  } else {
    Log.warn("Invalid module position found for this configuration:", module);
  }
  } else {
    Log.warn("No module name found for this configuration:", module);
  }
}
```

This fix should prevent any unnecessary filesystem access and prevent extraneous code from being run on startup.

Additionally, while tracking down this bug, I found the following code in the base `node_helper.js` around line 85: 
```javascript
io.of(this.name).on("connection", (socket) => {
	// add a catch all event.
	const onevent = socket.onevent;
	socket.onevent = function (packet) {
		const args = packet.data || [];
		onevent.call(this, packet); // original call
		packet.data = ["*"].concat(args);
		onevent.call(this, packet); // additional call to catch-all
	};

	// register catch all.
	socket.on("*", (notification, payload) => {
		if (notification !== "*") {
			this.socketNotificationReceived(notification, payload);
		}
	});
});
```

This is a fairly old method of catch-all event handling in socketIO. It may have been borrowed from a [2015 SO answer](https://stackoverflow.com/a/29957288/3312395). New versions of socketIO make this much simpler [[docs](https://socket.io/docs/v4/listening-to-events/#catch-all-listeners)]:
```javascript
io.of(this.name).on("connection", (socket) => {
	// register catch all.
	socket.onAny((notification, payload) => {
		this.socketNotificationReceived(notification, payload);
	});
});
```

I'm happy to submit pull requests for both of these topics, but I wanted to make sure it's discussed here since they are both critical areas of the setup.

Please let me know if I am missing anything!

sdetweil, 2024-08-12:
but you don’t know if it has a node_helper til you get down there.  but whatever. 

socket

this code WAS developed in 2015 or 2016, was like this when i got here in 2017.

ryan-d-williams, 2024-08-12:
@sdetweil 

Right, but all of that code is just layers of abstraction to load the node_helper for the module. Might as well stop it before it begins since we already know for sure it has been loaded rather than stopping it at the very end.

For the socket improvement, no defense needed. Not trying to assign blame, just want to make the project better!

sdetweil, 2024-08-12:
on the socket thing , in general i am opposed to changing code unless there is a technical reason. cause there is a new way is not valid. 

on the other, submit a pr.

ryan-d-williams, 2024-08-12:
@sdetweil 

Will do! 

There is a technical reason for that change, the old code sends twice as much traffic through the websocket network as it needs to (it duplicates the request and puts a "*" on it to ensure it is caught). The new code does not have this special message so the websocket messages are cut in half. Not a huge deal for a network this size, but still cuts the required messages down. Sorry I didn't make that explicit in my original text. 

I think the other technical argument to be made is that the old method is no longer documented. I can't find any reference to `onevent` anywhere in their documentation. Keeping up with the documented functions should keep the project more maintainable long-term.

PR #3523 (2024-08-18): Improve duplicate module filtering. Update SocketIO catch-all API.
- [x] Base your pull requests against the `develop` branch.
- [x] Include these infos in the description:
> - Does the pull request solve a **related** issue?
Yes - solves #3521

> - If so, can you reference the issue like this `Fixes #<issue_number>`?
Fixes #3521 (also mentioned in commit message)

> - What does the pull request accomplish? Use a list if needed.

> > - Updates duplicate module filter method (upstream vs downstream - see #3502)
> > - Updates socket io catchall functionality to new API [[docs](https://socket.io/ [... cut]
commit cc1d4ab2 (2024-08-18): Improve duplicate module filtering. Update SocketIO catch-all API. (#3523)

- [x] Base your pull requests against the `develop` branch.
- [x] Include these infos in the description:
> - Does the pull request solve a **related** issue?
Yes - solves #3521

> - If so, can you reference the issue like this `Fixes
#<issue_number>`?
Fixes #3521 (also mentioned in commit message)

> - What does the pull request accomplish? Use a list if needed.

> > - Updates duplicate module filter method (upstream vs downstream -
see #3502)
> > - Updates socket io catchall functionality to new API
[[docs](https://s [... cut]
closed 2024-09-30 by khassel