MagicMirrorOrg/MagicMirror #3525: Add warning message to MagicMirrorOS

tfischer4765, 2024-08-24:
MagicMirrorOS, while a cool idea, seems to be unmaintained and quite outdated, to the point where packages on the host system can't be updated easily without expert knowledge. 

A warning should be added to the installation method in https://docs.magicmirror.builders/getting-started/installation.html#magicmirroros to notify users of potential problems and security issues using it.

sdetweil, 2024-08-24:
its not part of the base MagicMirror. 

its developed and owned by a user just
like you.

tfischer4765, 2024-08-24:
The documentation I linked however IS owned and operated by the MagicMirror core team, is it not?

sdetweil, 2024-08-24:
yes. in this case we should remove that entry.

rejas, 2024-08-24:
> yes. in this case we should remove that entry.

I dont want to remove that useful information. As we already state on top of the paragraph:

"The following installation methods are not maintained by the MagicMirror² core team. Use these scripts and methods at your own risk."

Maybe its enough to put this into a yellow Attention Box to make it clearer?

sdetweil, 2024-08-24:
i personally hate that sentence, "use at your own risk", given how much time i put into install and upgrade

khassel, 2024-08-25:
> MagicMirrorOS, while a cool idea, seems to be unmaintained and quite outdated

it was updated ~2 weeks ago and uses actual raspian os lite images as base

khassel, 2024-09-24:
@tfischer4765 

no feedback: I will close this
otherwise: I will move this to the documentation repo

closed 2024-09-30 by khassel