MagicMirrorOrg/MagicMirror #3532: calendar e2e test fails

khassel, 2024-09-12:
```bash
FAIL tests/e2e/modules/calendar_spec.js (77.317 s)
  ● Calendar module › Events from multiple calendars › should show multiple events with the same title and start time from different calendars
    expect(received).resolves.toBe()
    Received promise rejected instead of resolved
    Rejected to value: [Error: expect(received).toHaveLength(expected)·
    Expected length: 22
    Received length: 20
    Received object: [<tr class="event-wrapper normal event"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">TestEvent</td><td class="time light ">Mar 9th</td></tr>, <tr class="event-wrapper normal event"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">TestEvent</td><td class="time light ">Mar 9th</td></tr>, <tr class="event-wrapper normal event"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">TestEvent</td><td class="time light ">Sep 9th</td></tr>, <tr class="event-wrapper normal event"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">TestEvent</td><td class="time light ">Sep 9th</td></tr>, <tr class="event-wrapper normal event"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">TestEvent</td><td class="time light ">Mar 9th</td></tr
       96 |
       97 | 		it("should show multiple events with the same title and start time from different calendars", async () => {
    >  98 | 			await expect(testElementLength(".calendar .event", 22)).resolves.toBe(true);
          | 			      ^
       99 | 		});
      100 | 	});
      101 |
      at expect (node_modules/expect/build/index.js:113:15)
      at Object.expect (tests/e2e/modules/calendar_spec.js:98:10)
```

This seems to be a data problem. Running this test 10 days back the test succeeds.

Running the config and checking result in the browser results in 20 calendar entries.

Quickfix would be to change expected 22 to now 20. But I will try to dig deeper and find a real fix.

khassel, 2024-09-12:
There are 2 calender events for the Sept. 09 2024 which are missing now (because events in the past are not displayed).

khassel, 2024-09-12:
I'm to far away from the calendar stuff. So I will change the expected value from 22 to 20 for getting the tests running again.

We need another test for this which works not only for a limited time (next date is Mar. 03 2027). The used test data `calendar_test.ics` is used in several other tests so we need new data for this test, maybe with events far away in the future.

sdetweil, 2024-09-12:
you can set the date for this test, right? just set back to august 15, 2024

khassel, 2024-09-12:
no, in `e2e` you cannot set date, would be require a move to `electron`

sdetweil, 2024-09-12:
future dates only go out a year from the current day

khassel, 2024-09-12:
from debugging: max date of events displayed in this test is 2049-09-09 ...

will not invest more time in this

PR #3533 (2024-09-13): hotfix for calendar_spec.js (used data now returns 20 events)
fixes #3532
commit 3380314c (2024-09-13): hotfix for calendar_spec.js (used data now returns 20 events) (#3533)

fixes #3532
PR #3540 (2024-09-15): fixes calendar test by moving it from e2e to electron with fixed date
and refactor tests/electron/modules/calendar_spec.js

fixes #3532
commit 0faefd10 (2024-09-15): fixes calendar test by moving it from e2e to electron with fixed date (#3540)

and refactor tests/electron/modules/calendar_spec.js

fixes #3532
closed 2024-09-15 by khassel