MagicMirrorOrg/MagicMirror #3537: new node-ical version v0.19.0 breaks a bunch of calendar tests

khassel, 2024-09-14:
electron tests:

```bash
 FAIL   electron  tests/electron/modules/calendar_spec.js (32.671 s)
  Calendar module
    Test css classes
      ✓ has css class dayBeforeYesterday (2646 ms)
      ✓ has css class yesterday (2981 ms)
      ✓ has css class today (2666 ms)
      ✓ has css class tomorrow (2650 ms)
      ✓ has css class dayAfterTomorrow (2710 ms)
    rrule
      ✕ Issue #3393 recurrence dates past rrule until date (2669 ms)
    Exdate: LA crossover DST before midnight GMT
      ✕ LA crossover DST before midnight GMT should have 2 events (2669 ms)
    Exdate: LA crossover DST at midnight GMT local STD
      ✕ LA crossover DST before midnight GMT should have 2 events (2709 ms)
    Exdate: LA crossover DST at midnight GMT local DST
      ✕ LA crossover DST before midnight GMT should have 2 events (2670 ms)
    Exdate: SYD crossover DST before midnight GMT
      ✕ LA crossover DST before midnight GMT should have 2 events (2822 ms)
    Exdate: SYD crossover DST at midnight GMT local STD
      ✕ LA crossover DST before midnight GMT should have 2 events (2699 ms)
    Exdate: SYD crossover DST at midnight GMT local DST
      ✕ SYD crossover DST at midnight GMT local DST should have 2 events (2648 ms)

  ● Calendar module › rrule › Issue #3393 recurrence dates past rrule until date

    expect(received).toBe(expected) // Object.is equality

    Expected: 1
    Received: 2

      58 |                      expect(elem).not.toBeNull();
      59 |                      const cnt = await loc.count();
    > 60 |                      expect(cnt).toBe(1);
         |                                  ^
      61 |              });
      62 |      });
      63 |

      at Object.toBe (tests/electron/modules/calendar_spec.js:60:16)

  ● Calendar module › Exdate: LA crossover DST before midnight GMT › LA crossover DST before midnight GMT should have 2 events

    expect(received).toBe(expected) // Object.is equality

    Expected: 2
    Received: 3

      81 |                      expect(elem).not.toBeNull();
      82 |                      const cnt = await loc.count();
    > 83 |                      expect(cnt).toBe(2);
         |                                  ^
      84 |              });
      85 |      });
      86 |

      at Object.toBe (tests/electron/modules/calendar_spec.js:83:16)

  ● Calendar module › Exdate: LA crossover DST at midnight GMT local STD › LA crossover DST before midnight GMT should have 2 events

    expect(received).toBe(expected) // Object.is equality

    Expected: 2
    Received: 4

      94 |                      expect(elem).not.toBeNull();
      95 |                      const cnt = await loc.count();
    > 96 |                      expect(cnt).toBe(2);
[... 188 lines cut ...]
      at it (tests/e2e/modules/calendar_spec.js:67:3)
      at describe (tests/e2e/modules/calendar_spec.js:49:2)
      at Object.describe (tests/e2e/modules/calendar_spec.js:4:1)

  ● Calendar module › Custom configuration › should show two custom icons for repeating events

    expect(received).resolves.toBe()

    Received promise rejected instead of resolved
    Rejected to value: [Error: expect(received).toHaveLength(expected)·
    Expected length: 2
    Received length: 3
    Received object: [<span class="fas fa-fw fa-undo" />, <span class="fas fa-fw fa-undo" />, <span class="fas fa-fw fa-undo" style="padding-left: 5px;" />]]

      70 |
      71 |              it("should show two custom icons for repeating events", async () => {
    > 72 |                      await expect(testElementLength(".calendar .event .fa-undo", 2)).resolves.toBe(true);
         |                            ^
      73 |              });
      74 |
      75 |              it("should show two custom icons for day events", async () => {

      at expect (node_modules/expect/build/index.js:113:15)
      at Object.expect (tests/e2e/modules/calendar_spec.js:72:10)
```

bugsounet, 2024-09-28:
@khassel

Are you sure about this ?
because we use node-ical v0.19.0 in package-lock on develop branch

see [there](https://github.com/MagicMirrorOrg/MagicMirror/blob/develop/package-lock.json#L26)  and [there](https://github.com/MagicMirrorOrg/MagicMirror/blob/develop/package-lock.json#L9102)

khassel, 2024-09-28:
Thanks. Will check this again later ...

khassel, 2024-09-28:
sadly the errors still exist.

We had a problem with `package-lock.json`, fix for this is [on the way](https://github.com/MagicMirrorOrg/MagicMirror/pull/3566).

sdetweil, 2024-09-29:
I tested with node-ical 19, and the problem is that the EXDATE entries are not returned in the parsed data 
I opened https://github.com/jens-maus/node-ical/issues/338
to document the problem. 

all of the failing testcases have EXDATE entries

sdetweil, 2024-09-30:
I think I have fix for that problem (they were returned but cononverted to UTC time so never will match) 

BUT I think I have a fix for all the date conversion problems. (all our testcases pass with all the conversions commented out)

want more testing..

sdetweil, 2024-10-07:
see this.. waiting on user test  for submitting PR here and node-ical 
all testcases in both run successfully.. 

here is a test version of the fixes for all kinds of date problems.

best to make a new folder and git clone there 
```
git  clone https://github.com/sdetweil/MagicMirror
git checkout fixcaldates
npm run install-mm
```
copy your config.js and custom.css from the prior folder
and the non-default modules you have installed..

this ONLY changes the default calendar
but DOES ship an updated node-ical library too

if you need to fall back, just rename the folders around again so that 
your original is called MagicMirror

all the testcases for node-ical and MagicMirror execute successfully. 

the 'BIG' change here is to get the local NON-TZ dates for the 
rrule.between()

all the checking and conversion code is commented out or not used
the node-ical fixes are for excluded dates (exdate) values being adjusted for DST/STD time..   waiting to submit that PR 

one fix in calendar.js for checking if a past date was too far back, 
but it never checked to see IF the event date was in the past..  (before today)  so it chopped off too many

and one change in calendarfetcher.js to put out a better diagnostic message of the parsed data.. (exdate was excluded cause JSON stringify couldn't convert the complex structure)

khassel, 2024-10-11:
I tested your branch and I still get some test errors:

```bash
 FAIL   unit  tests/unit/modules/default/calendar/calendar_fetcher_utils_bad_rrule.js
  Calendar fetcher utils test
    filterEvents
      ✕ no events, not crash (53 ms)

  ● Calendar fetcher utils test › filterEvents › no events, not crash

    expect(received).toHaveLength(expected)

    Expected length: 0
    Received length: 2
    Received array:  [{"class": undefined, "description": "ongoingEvent", "endDate": "1728689046900", "fullDayEvent": false, "geo": false, "location": false, "startDate": "1728681846899", "title": "ongoingEvent"}, {"class": undefined, "description": "upcomingEvent", "endDate": "1728692646900", "fullDayEvent": false, "geo": false, "location": false, "startDate": "1728689046900", "title": "upcomingEvent"}]

      24 |                      );
      25 |
    > 26 |                      expect(filteredEvents).toHaveLength(0);
         |                                             ^
      27 |              });
      28 |      });
      29 | });

      at Object.toHaveLength (tests/unit/modules/default/calendar/calendar_fetcher_utils_bad_rrule.js:26:27)

 FAIL   electron  tests/electron/modules/calendar_spec.js (56.808 s)
  Calendar module
    Test css classes
      ✓ has css class dayBeforeYesterday (2326 ms)
      ✓ has css class yesterday (2312 ms)
      ✓ has css class today (2258 ms)
      ✓ has css class tomorrow (2309 ms)
      ✓ has css class dayAfterTomorrow (2280 ms)
    Events from multiple calendars
      ✓ should show multiple events with the same title and start time from different calendars (2270 ms)
    rrule
      ✓ Issue #3393 recurrence dates past rrule until date (2340 ms)
    Exdate: LA crossover DST before midnight GMT
      ✓ LA crossover DST before midnight GMT should have 2 events (2270 ms)
    Exdate: LA crossover DST at midnight GMT local STD
      ✓ LA crossover DST before midnight GMT should have 2 events (2330 ms)
    Exdate: LA crossover DST at midnight GMT local DST
      ✓ LA crossover DST before midnight GMT should have 2 events (2330 ms)
    Exdate: SYD crossover DST before midnight GMT
      ✓ LA crossover DST before midnight GMT should have 2 events (2250 ms)
    Exdate: SYD crossover DST at midnight GMT local STD
      ✓ LA crossover DST before midnight GMT should have 2 events (2289 ms)
    Exdate: SYD crossover DST at midnight GMT local DST
      ✓ SYD crossover DST at midnight GMT local DST should have 2 events (2309 ms)
    sliceMultiDayEvents
      ✓ Issue #3452 split multiday in Europe (2269 ms)
    germany timezone
      ✓ Issue #unknown fullday timezone East of UTC edge (2249 ms)
    germany all day repeating moved (recurrence and exdate)
      ✓ Issue #unknown fullday timezone East of UTC event moved (2270 ms)
    chicago late in timezone
      ✕ Issue #unknown rrule US close to timezone edge (20071 ms)

  ● Calendar module › chicago late in timezone › Issue #unknown rrule US close to timezone edge

    thrown: "Exceeded timeout of 20000 ms for a test.
    Add a timeout value to this test to increase the timeout, if this is a long-running test. See https://jestjs.io/docs/api#testname-fn-timeout."

      166 |
      167 |     describe("chicago late in timezone", () => {
    > 168 |             it("Issue #unknown rrule US close to timezone edge", async () => {
          |             ^
      169 |                     await helpers.startApplication("tests/configs/modules/calendar/chicago_late_in_timezone.js", "01 Sept 2024 10:38:00 GMT-06:00", ["js/electron.js"], "Europe/Berlin");
      170 |                     await expect(doTestTableContent(".calendar .event", ".time", "10th.Sep, 20:15")).resolves.toBe(true);
[... 99 lines cut ...]

      at expect (node_modules/expect/build/index.js:113:15)
      at Object.expect (tests/e2e/modules/calendar_spec.js:104:10)

  ● Calendar module › FullDayEvent Single day, should show Today › should contain in total two events

    expect(received).resolves.toBe()

    Received promise rejected instead of resolved
    Rejected to value: [Error: expect(received).toHaveLength(expected)·
    Expected length: 2
    Received length: 4
    Received object: [<tr class="event-wrapper normal event yesterday"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">daily full days</td><td class="time light ">Today</td></tr>, <tr class="event-wrapper normal event today" style="opacity: 1;"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">daily full days</td><td class="time light ">Today</td></tr>, <tr class="event-wrapper normal event tomorrow" style="opacity: 0.6666666666666667;"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">daily full days</td><td class="time light ">Tomorrow</td></tr>, <tr class="event-wrapper normal event dayAfterTomorrow" style="opacity: 0.33333333333333337;"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">daily full days</td><td class="time light ">Monday</td></tr>]]

      116 |             });
      117 |             it("should contain in total two events", async () => {
    > 118 |                     await expect(testElementLength(".calendar .event", 2)).resolves.toBe(true);
          |                           ^
      119 |             });
      120 |     });
      121 |

      at expect (node_modules/expect/build/index.js:113:15)
      at Object.expect (tests/e2e/modules/calendar_spec.js:118:10)
```

sdetweil, 2024-10-11:
that the electron tests, right?  I didnt run e2e

sdetweil, 2024-10-11:
you DID npm install to get the update node-ical, right?  I have a short term github link to my fork of that

sdetweil, 2024-10-11:
i'll go over the e2e tests.. only ran electron due to the date constraints

khassel, 2024-10-12:
these tests errored:

```bash
tests/unit/modules/default/calendar/calendar_fetcher_utils_bad_rrule.js
tests/electron/modules/calendar_spec.js
tests/e2e/modules/calendar_spec.js
```

> you DID npm install to get the updated node-ical, right?

sure, used this:

```yaml
services:
  magicmirror:
    image: registry.gitlab.com/khassel/magicmirror:develop_debug22
    ports:
      - "8080:8080"
    stdin_open: true
    tty: true
    command:
      - /bin/bash
      - -c
      - |
        # comment out if needed:
        rm -rf ./*
        rm -rf ./.*
        git clone -b fixcaldates https://github.com/sdetweil/MagicMirror.git .
        npm install
        mv config/config.js.sample config/config.js
        sed -i 's|address:.*|address: "0.0.0.0",|g;s|ipWhitelist:.*|ipWhitelist: [],|g' config/config.js
        touch ${MM_CUSTOMCSS_FILE:-"css/custom.css"}
        Xvfb :99 -screen 0 1024x768x16 &
        export DISPLAY=:99
        /bin/bash
```

sdetweil, 2024-10-17:
thx..

new branch

here is an updated test version of the fixes for all kinds of calendar date problems.

NOTE: the changed branch name
NOTE: this used the node-cal@0.19.0 library UNCHANGED

best to make a new folder and git clone there

git clone https://github.com/sdetweil/MagicMirror
cd MagicMirror
git checkout fixcaldates2 // <------ note this is a changed branch name
npm run install-mm
copy your config.js and custom.css from the prior folder
and the non-default modules you have installed…

this ONLY changes the default calendar
but DOES ship an updated node-ical library too

if you need to fall back, just rename the folders around again so that
your original is called MagicMirror

all the testcases for node-ical and MagicMirror execute successfully.

the ‘BIG’ change here is to get the local NON-TZ dates for the
rrule.between()

all the checking and conversion code is commented out or not used
the node-ical fixes are for excluded dates (exdate) values being adjusted for DST/STD time… waiting to submit that PR

one fix in calendar.js for checking if a past date was too far back,
but it never checked to see IF the event date was in the past… (before today) so it chopped off too many

and one change in calendarfetcher.js to put out a better diagnostic message of the parsed data… (exdate was excluded cause JSON stringify couldn’t convert the complex structure)

I added the tests you all have documented

please re-pull and checkout the new branch (I deleted the old branch)
and npm run install-mm again

sdetweil, 2024-11-01:
node-ical version 0.20. posted to npm

will update pr next week when back home

closed 2025-01-01 by sdetweil