MagicMirrorOrg/MagicMirror #3552: npm run config:check needs to  support new JS idioms, like chaining (?.)

sdetweil, 2024-09-23:
because config.js enables code, some developers may use it 

see  forum
https://forum.magicmirror.builders/topic/19001/picture-in-bg-on-a-cell-related-to-someone-s-birthday/5?_=1726690411207

I don't know if the work on eslint 9  can be used to solve this

khassel, 2024-09-23:
I already tested the config provided in the forum topic with no problems, `npm run config:check` was successful on `develop` branch.

The other point is if the change in [this PR](https://github.com/MagicMirrorOrg/MagicMirror/pull/3450/files) (concerning `exit` instead `return`) should be reverted because with this change mm will not start anymore if the config check fails ...

@KristjanESPERANTO @rejas

sdetweil, 2024-09-23:
cool on the check working on develop

i thought that was the point of the check change , to NOT  start if check failed, and check on every start

KristjanESPERANTO, 2024-09-24:
> I already tested the config provided in the forum topic with no problems

Yes, I think I "accidentally" fixed that by updating `env` in 06f6fbf49ba46c5fbe764d86e16656497f3d6375

> i thought that was the point of the check change , to NOT start if check failed

I also thought so. But I'm no longer quite sure anymore because of this chaining example. I would be fine with either decision (starting and NOT starting).

sdetweil, 2024-09-24:
well starting starts something other than what the user defined

there is an alternative for chaining.  

anyhow w pm2 starting for new users they have no idea what is going on.  not see the logs etc

i like the no start

khassel, 2024-09-24:
so it stays as it is, can we close the issue?

sdetweil, 2024-09-24:
yes,

closed 2024-09-24 by sdetweil