MagicMirrorOrg/MagicMirror #3557: Not opening in WebView but in browser

ibrahim5albakri, 2024-09-24:
install.log

install starting  - Tue Sep 24 21:14:39 +03 2024
installing on aarch64 processor system
the os is PRETTY_NAME="Debian GNU/Linux 12 (bookworm)" NAME="Debian GNU/Linux" VERSION_ID="12" VERSION="12 (bookworm)" VERSION_CODENAME=bookworm ID=debian HOME_URL="https://www.debian.org/" SUPPORT_URL="https://www.debian.org/support" BUG_REPORT_URL="https://bugs.debian.org/"
               total        used        free      shared  buff/cache   available
Mem:            7810         584        5493          45        1900        7225
Swap:            199           0         199
[96mUpdating packages ...[0m
Hit:1 http://deb.debian.org/debian bookworm InRelease Hit:2 http://archive.raspberrypi.com/debian bookworm InRelease Hit:3 http://deb.debian.org/debian-security bookworm-security InRelease Hit:4 http://deb.debian.org/debian bookworm-updates InRelease Reading package lists...
apt-get update  completed ok
Reading package lists...
Building dependency tree...
Reading state information...
pv is already the newest version (1.6.20-1).
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
apt-get upgrade  started
apt-get upgrade result =rc=0 Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages have been kept back:
  chromium-browser chromium-browser-l10n chromium-codecs-ffmpeg-extra
  raspi-utils rpi-chromium-mods
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
[96mInstalling helper tools ...[0m
Reading package lists...
Building dependency tree...
Reading state information...
curl is already the newest version (7.88.1-10+deb12u7).
wget is already the newest version (1.21.3-1+b1).
git is already the newest version (1:2.39.5-0+deb12u1).
build-essential is already the newest version (12.9).
unzip is already the newest version (6.0-28).
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
installing on bookworm
[96mCloning MagicMirror ...[0m
[92mCloning MagicMirror Done![0m
[96mInstalling dependencies ...[0m
> magicmirror@2.28.0 postinstall > npm run install-vendor && npm run install-fonts && echo "MagicMirrorÂ² installation finished successfully! > " > magicmirror@2.28.0 install-vendor > echo "Installing vendor files ... > " && cd vendor && npm install --loglevel=error --no-audit --no-fund --no-update-notifier Installing vendor files ... added 10 packages in 4s > magicmirror@2.28.0 install-fonts > echo "Installing fonts ... > " && cd fonts && npm install --loglevel=error --no-audit --no-fund --no-update-notifier Installing fonts ... added 2 packages in 2s MagicMirrorÂ² installation finished successfully! > magicmirror@2.28.0 prepare > [ -f node_modules/.bin/husky ] && husky || echo no husky installed. no husky installed. added 433 packages in 5m 102 packages are looking for funding run `npm fund` for details
[92mDependencies installation Done![0m
package.json update for aarch64 or Electron missing, completed ok
setting up initial config.js
[96mCheck plymouth installation ...[0m
[96mSplashscreen: Checking themes directory.[0m
[96mSplashscreen: Create theme directory if not exists.[0m
[92mSplashscreen: Changed theme to MagicMirror successfully.[0m
gsettings screen saver already disabled
disable screensaver via lightdm.conf
disable screensaver via lxsession
Found: wayfire.ini
wayfire screen saver already disabled
[92mWe're ready! Run [1m[97mDISPLAY=:0 npm start[0m[92m from the ~/MagicMirror directory to start your MagicMirror.[0m
install completed - Tue Sep 24 21:26:26 +03 2024

sdetweil, 2024-09-24:
>package.json update for aarch64 or Electron missing, completed o

so, elevtron didn't install.. 

that means I setup for using chrome (or chromium)  over the server mode
BUT I dont check if chrome is installed..

did you setup to use pm2 to autostart? 

if so, from a terminal window do 
pm2 stop all 

then do 
which chrome
which chromium

if they don't return anything then they are not installed

you can also use firefox, midori or push  (ee the doc on my script at the end descrives what to do 

we can confirm 
```sh
cd ~/MagicMirror
npm start 
```
and look at the messages

ibrahim5albakri, 2024-09-24:
> >package.json update for aarch64 or Electron missing, completed o
> 
> so, electron didn't install.. 
> 
> that means I setup for using chrome (or chromium)  over the server mode
> BUT I dont check if chrome is installed..
> 
> did you setup to use pm2 to autostart? 
> 
> if so, from a terminal window do 
> pm2 stop all 
> 
> then do 
> which chrome
> which chromium
> 
> if they don't return anything then they are not installed
> 
> you can also use firefox, midori or push  (ee the doc on my script at the end descrives what to do 
> 
> we can confirm 
> ```sh
> cd ~/MagicMirror
> npm start 
> ```
> and look at the messages
> 
> 

No, I did not do autorun, but when I run an instruction npm start 
A white screen appears for a few seconds and then disappears The screen freezes

sdetweil, 2024-09-24:
did you check from chrome? or chromium?

ibrahim5albakri, 2024-09-24:
Yes it works correctly

sdetweil, 2024-09-24:
can you show the output of npm start

closed 2024-09-27 by ibrahim5albakri