MagicMirrorOrg/MagicMirror #3572: [Weather module] Zeros are no longer hidden in forecast

CGabo, 2024-10-01:
Hi

In the newer release of MM Weather Module (forecast), the zeros are no longer hidden.

Version: 2.24.0: If the value is zero, it will be hidden aka blanked out.

Version: 2.29.0:  All numbers are displayed, including the zeros

Maybe this was changed in the previous version, I can't say because I skipped some versions.
But the old version looked nicer when the zeros were not displayed.

See please the picture.

[image]

Is it possible to bring it back or maybe there could be an option in the config?

Thank you so much for your great job!
Have a nice day.

CGabo, 2024-10-01:
Please see also:
Precipitation Probability not displayed when it is 0% #3345

This was changed in Release 2.27.0 #3410, but as I mentioned, I mean the 0 mm, not the 0%
Thanks

khassel, 2024-10-01:
thank you for the research. You want it not displayed, the one who opened the old issue wants to see 0 values ...

So only way I see is to add a param ...

CGabo, 2024-10-01:
Yes that would be great!
Maybe the change could be included in a future release?

Honestly, which representation is better?
From a distance, you can see that there is no rain.
Even if the probability tends towards zero ;-)

Thanks again
Gabo

khassel, 2024-10-01:
> Honestly, which representation is better?

that's a matter of taste ... I agree with you and prefer empty values.

> Maybe the change could be included in a future release?

the next release is in 3 months, feel free to implement it ;) (otherwise I will do it at some point)

CGabo, 2024-10-02:
Ok, I give it a try. 
Should I leave it open until I implemented the change?

khassel, 2024-10-02:
> Ok, I give it a try.

👍

> Should I leave it open until I implemented the change?

yes, please.

Procedure is:
- issue is solved by a PR and merged on `develop` branch
- with next release `develop` is merged on `master` and we close all solved issues

khassel, 2025-03-27:
should be fixed with https://github.com/MagicMirrorOrg/MagicMirror/pull/3748

closed 2025-04-01 by rejas