MagicMirrorOrg/MagicMirror #3574: Weather Module not loading since MM update

Denn2705, 2024-10-02:
Hi all,

i did the MM Update few days ago (Notifikation in the bottom bar) via git pull. Until the Update everything worked fine, but since the Update was done, the Weather Module is not loading any more. All other modules are fine after the update. I dont see anything in the logs. Would be great if someone could help me figuring out whats going on.

System: Raspi Zero 2W with PoE LAN Hat
Node: v22.9.0
OS: Debian 12 Bookworm latest update/upgrade
Electron: v32.1.2
MagicMirror Version: 2.29.0

Weather Module API is OpenWeather Map
API Key is valid and checked.

Config-Part of Weather:
```
		{
			module: "weather",
			position: "bottom_right",
			header: "Wetter Vorhersage",
			config: {
				weatherProvider: "openweathermap",
				type: "forecast",
				maxNumberOfDays: 5,
				locationID: "2831948", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "*****CENSORED*******",
				weatherEndpoint: "/forecast",
				tableClass: 'medium',
				colored: 'true',
				fade: true,
				fadePoint: 0.5,
				
			}
```

Log output if restarting MM:

> 0|mm  | [2024-10-02 12:37:42.326] [LOG]   Starting MagicMirror: v2.29.0 
> 0|mm  | [2024-10-02 12:37:42.427] [LOG]   Loading config ...
> 0|mm  | [2024-10-02 12:37:42.436] [LOG]   config template file not exists, no envsubst
> 0|mm  | [2024-10-02 12:37:43.863] [INFO]  Checking config file /home/pi/MagicMirror/config/config.js ...
> 0|mm  | [2024-10-02 12:37:44.087] [INFO]  Your configuration file doesn't contain syntax errors :)
> 0|mm  | [2024-10-02 12:37:44.090] [INFO]  Checking modules structure configuration ...
> 0|mm  | [2024-10-02 12:37:44.205] [INFO]  Your modules structure configuration doesn't contain errors :)
> 0|mm  | [2024-10-02 12:37:44.212] [LOG]   Loading module helpers ...
> 0|mm  | [2024-10-02 12:37:44.218] [LOG]   No helper found for module: alert.
> 0|mm  | [2024-10-02 12:37:44.240] [LOG]   Initializing new module helper ...
> 0|mm  | [2024-10-02 12:37:44.242] [LOG]   Module helper loaded: updatenotification
> 0|mm  | [2024-10-02 12:37:44.245] [LOG]   No helper found for module: clock.
> 0|mm  | [2024-10-02 12:37:44.701] [LOG]   Initializing new module helper ...
> 0|mm  | [2024-10-02 12:37:44.706] [LOG]   Module helper loaded: calendar
> 0|mm  | [2024-10-02 12:37:44.717] [LOG]   No helper found for module: MMM-AutoDimmer.
> 0|mm  | [2024-10-02 12:37:44.723] [LOG]   No helper found for module: weather.
> 0|mm  | [2024-10-02 12:37:44.741] [LOG]   Initializing new module helper ...
> 0|mm  | [2024-10-02 12:37:44.742] [LOG]   Module helper loaded: netatmo
> 0|mm  | [2024-10-02 12:37:44.743] [LOG]   All module helpers loaded.
> 0|mm  | [2024-10-02 12:37:44.764] [LOG]   Starting server on port 8080 ...
> 0|mm  | [2024-10-02 12:37:56.138] [LOG]   Server started ...
> 0|mm  | [2024-10-02 12:37:56.163] [LOG]   Connecting socket for: updatenotification
> 0|mm  | [2024-10-02 12:37:56.165] [LOG]   Starting module helper: updatenotification
> 0|mm  | [2024-10-02 12:37:56.166] [LOG]   Connecting socket for: calendar
> 0|mm  | [2024-10-02 12:37:56.168] [LOG]   Starting node helper for: calendar
> 0|mm  | [2024-10-02 12:37:56.169] [LOG]   Connecting socket for: netatmo
> 0|mm  | [2024-10-02 12:37:56.170] [LOG]   Netatmo helper started ...
> 0|mm  | [2024-10-02 12:37:56.172] [LOG]   Sockets connected & modules started ...
> 0|mm  | [2024-10-02 12:37:58.210] [LOG]   Launching application.
> 0|mm  | [2024-10-02 12:38:09.296] [INFO]  System information:
> 0|mm  | ### SYSTEM:   manufacturer: Raspberry Pi Foundation; model: Raspberry Pi Zero 2 W Rev 1.0; virtual: false
> 0|mm  | ### OS:       platform: linux; distro: Raspbian GNU/Linux; release: 12; arch: arm; kernel: 6.6.51+rpt-rpi-v7
> 0|mm  | ### VERSIONS: electron: 29.1.6; used node: 20.9.0; installed node: 22.9.0; npm: 10.8.3; pm2: 5.3.1
> 0|mm  | ### OTHER:    timeZone: Europe/Berlin; ELECTRON_ENABLE_GPU: undefined
> 0|mm  | [2024-10-02 12:38:18.041] [ERROR] Netatmo: clientId not set in config.
> 0|mm  | [2024-10-02 12:38:18.116] [LOG]   Create new calendarfetcher for url: http://p71-caldav.icloud.com/published/2/MjA0MDUwMTIxMjA.....*CENSORED* - Interval: 604800000
> 0|mm  | [2024-10-02 12:38:18.786] [LOG]   Netatmo: using token.json file
> 0|mm  | [2024-10-02 12:38:18.794] [LOG]   Netatmo: Initialized
> 0|mm  | [2024-10-02 12:38:18.847] [INFO]  updatenotification: Updater Class Loaded!
> 0|mm  | [2024-10-02 12:38:18.848] [INFO]  updatenotification: Checking PM2 using...
> 0|mm  | [2024-10-02 12:38:21.265] [INFO]  Checking git for module: MMM-AutoDimmer
> 0|mm  | [2024-10-02 12:38:21.642] [INFO]  Checking git for module: netatmo
> 0|mm  | [2024-10-02 12:38:21.819] [INFO]  Checking git for module: MagicMirror
> 0|mm  | [2024-10-02 12:38:21.972] [INFO]  updatenotification: [PM2] You are not using pm2
> 0|mm  | [2024-10-02 12:38:22.050] [LOG]   Netatmo: Authenticated
> 0|mm  | [2024-10-02 12:38:22.053] [LOG]   Netatmo: token.json was written successfully
> 0|mm  | [2024-10-02 12:38:22.109] [LOG]   Netatmo: New Token Expire Wednesday, October 2, 2024 3:38 PM
> 0|mm  | [2024-10-02 12:38:32.845] [ERROR] Moment Timezone has no data for GMT+0200. See http://momentjs.com/timezone/docs/#/data-loading/.
> 0|mm  | [2024-10-02 12:38:32.846] [ERROR] Moment Timezone has no data for GMT+0200. See http://momentjs.com/timezone/docs/#/data-loading/.
> 0|mm  | [2024-10-02 12:38:32.847] [ERROR] Moment Timezone has no data for GMT+0200. See http://momentjs.com/timezone/docs/#/data-loading/.
> 0|mm  | [2024-10-02 12:38:33.062] [INFO]  Calendar-Fetcher: Broadcasting 40 events from http://p71-caldav.icloud.com/published/2/MjA0MDUwMTIxMjA0MD........*CENSORED*.

I dont even see the weather module loading in the log-File? But to be honest, i didnt compeared if it was listed before the update in the log... Unfortunatly i didnt stored any older log files.
The Display shows endless "loading". Can someone have a look on this or maybe guide me into deeper analysis?

[image]

Denn2705, 2024-10-02:
Small Update: I switched now from OpenWeatherMap Config to Pirateweather, and the module is working fine again.

Seems there is an Issue with the OpenWeatherMap config.

`		{
			module: "weather",
			position: "bottom_right",
			header: "Wetter Vorhersage",
			config: {
				weatherProvider: "pirateweather",
				type: "forecast",
				apibase: "https://api.pirateweather.net",
				weatherEndpoint: "/forecast",
				apiKey: "***CENSORED***",
				lat: "***CENSORED***",
				lon: "***CENSORED***",
				tableClass: 'medium',
				colored: 'true',
				fade: true,
				fadePoint: 0.5,
				
			}
		},
`

[image]

JHenzi, 2024-10-02:
I'm getting the same with OpenWeatherMap:

```
Could not load data ...  TypeError: Cannot read properties of undefined (reading 'humidity')
    at Class.generateWeatherObjectFromCurrentWeather (openweathermap.js:132:53)
    at openweathermap.js:37:28
```

and 

```
Could not load data ...  TypeError: forecasts is not iterable
    at Class.generateForecastHourly (openweathermap.js:184:26)
    at Class.generateWeatherObjectsFromForecast (openweathermap.js:149:16)
    at openweathermap.js:57:22
```

Doesn't seem right to switch providers to fix this.

sdetweil, 2024-10-02:
@JHenzi can you post the module info for weather from config.js  xxx out your apikey

SirStepheno, 2024-10-02:
They switched the default API version to 3.0, don't know why I doens't work anymore.

But when you add `apiVersion: "2.5"` to the config, it works again.

JHenzi, 2024-10-02:
The addition of `apiVersion` tagging works. Here's my config:

```
{
			module: "weather",
			position: "top_right",
			config: {
				weatherProvider: "openweathermap",
				type: "current",
				location: "Cincinnati",
				locationID: "4508722", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "XXXX",
				apiVersion: "2.5"
			}
		},
		{
			module: "weather",
			position: "top_left",
			header: "Weather Forecast",
			config: {
				weatherProvider: "openweathermap",
				type: "forecast",
				location: "Cincinnati",
				locationID: "4508722", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "XXXX",
				apiVersion: "2.5"
			}
		}
```

Denn2705, 2024-10-02:
Closed the Issue as fixed with adding the apiVersion: „2.5“

JHenzi, 2024-10-02:
This isn't fixed! API Version 2.5 is deprecated. It was supposed to be retired in June of 2024. This needs a real fix, everyone has to migrate to version 3.0. 

Your default weather service is going to stop working and does NOT work without a code change already.

sdetweil, 2024-10-02:
@JHenzi we changed because of their notice.  3.0 is the only official api

Denn2705, 2024-10-02:
ReOpened due to request for an update to api 3.0

MarcLandis, 2024-10-02:
Nothing needs to be fixed with the code. You need to get a new V3 API key at openweathermap.com

Denn2705, 2024-10-02:
> Nothing needs to be fixed with the code. You need to get a new V3 API key at openweathermap.com

I can’t choose an api key version on my free profile. 
And generating a new key does not fix the issue. 
So you think only solution is to upgrade to API V3.0 billing Methode? 

[image]

gregimagines, 2024-10-03:
I have an API 3.0 account and the weather is not loading. The same API is working with other modules for weather (/scottcl88/MMM-DynamicWeather) but the default weather module is not working after the update.

If you setup a payment method you can limit it to 1000 per day so you never have to pay. I've had it setup for months and have yet to get a bill from them or see anything. It's just like the free account but API 3.0.

sdetweil, 2024-10-03:
before 2.29 (yesterday) the default weather module used 2.5 in the url

chrismaverick, 2024-10-03:
Oh thank you, @JHenzi this was driving me crazy before I thought to look here. 

I have three magic mirrors running. 1000 is probably still enough, but still. I kinda think the default option for a standard model probably shouldn't be to an API service that even just MIGHT require a fee. To much danger for an unsuspecting user who just accidentally does something wrong.

sdetweil, 2024-10-03:
@chrismaverick  we provide different providers

chrismaverick, 2024-10-03:
@sdetweil yes, I know. But the module defaults to openweathermap, so I am presuming that's the most likely to be used by those who are least able to understand what is happening.

sdetweil, 2024-10-04:
we did that because it gave the best results now so long ago, and was free. still is under most circumstances

DJPAUL90, 2024-10-08:
MM doesnt start when i was adding apiVersion: "2.5" to the config.js 

But when i change apiVersion: "3.0" here

~/MagicMirror/modules/default/weather/providers/openweathermap.js-

to "2.5" it solved the problem and weather was working again

sdetweil, 2024-10-08:
so you missed a comma after and maybe before

2.5 WILL  end sometime. was supposed to be june

don't edit our files. all config changes go  in config.js

KristjanESPERANTO, 2024-10-08:
> 2.5 WILL end sometime. was supposed to be june

I recieved this yesterday per email: "_We would like to inform you that we started One Call 2.5 deprecation, and your access to the product will be stopped on October 14, 2024._"

sdetweil, 2024-10-08:
thanks, i did not receive that notice

DJPAUL90, 2024-10-08:
> so you missed a comma after and maybe before
> 
> 2.5 WILL end sometime. was supposed to be june
> 
> don't edit our files. all config changes go in config.js

I think i did it right. It looked like the post above. But it didnt worked. Sorry i am a beginner dont know so much about MM. I also subscribbed to API 3.0 on openweather but i dont know what to do now. I dont get any special "3.0" API. They only send me this 

[image]

sdetweil, 2024-10-08:
many posters font post the 'exact' text, so you have to know. sometimes they don't post it correctly and the forum software changes the quote marks too. you need to know how to discover and fix it

the 3.0 api also requires an additional  config parm
``` 
weatherEndpoint:"/onecall",
```

Killator, 2024-10-08:
> @chrismaverick we provide different providers

Hello TeaMM,

I just discover the OpenWeather API issue... 👎 

Just a quick question: I have a Netatmo weather station, could you please make it an official data provider for the module ? 

I am not a developer, but if necessary, I can provide API keys to help you... And make an official request for change...

It would be so "magic" 😇

AWSW-de, 2024-10-15:
Hello, 
I have the same problem after the MM done today. I have checked my API key as well and it is reported as active still. The same key is working on one of my weather stations too without a problem https://github.com/G6EJD/ESP32-e-Paper-Weather-Display so I don’t see the point here for the API key as the issue. 

Thanks for a fix in advance =)

sdetweil, 2024-10-15:
@AWSW-de  not what we want, but what we get. 

OW has changed their business model and api to match .

you MUST  supply a credit card to get access to the new api, ( w an uncharged count so far of 1000 requests per day) and you must change the parameters to our weather module, 

as location and locationID no longer work. the api requires latitude and longitude now. 

you must also supply a new parm
weatherEndoint:"/onecall"

as they only provide one now

as they told us in emails, this transition has been going on for 2 years , and they are making it final. 
some have had emails saying this week is it.

sdetweil, 2024-10-15:
possible

JHWelch, 2024-10-19:
If someone comes here and is just looking for it, here is 
## How to Update to a 3.0 Key

You will need a credit card to do this, but you can limit your calls to the free limit and theoretically you will never be charged.

1. Go to [OpenWeatherMap's Billing Plan page] in the account section.
2. Under "One Call API 3.0" click "Subscribe"
3. Enter your payment details
4. (Optional) Update your "Calls per day" to `1000` to avoid being charged
5. Update your existing config

I use two modules, and my new config looks like this
```js
[
  {
    module: "weather",
    position: "top_left",
    config: {
      weatherProvider: "openweathermap",
      weatherEndpoint:"/onecall",
      type: "current",
      lat: 40.748440,
      lon: -73.985664,
      apiKey: "$WEATHER_API_KEY"
    }
  },
  {
      module: "weather",
      position: "top_left",
      header: "Weather Forecast",
      config: {
        weatherProvider: "openweathermap",
        weatherEndpoint:"/onecall",
        type: "forecast",
        lat: 40.748440,
        lon: -73.985664,
        apiKey: "$WEATHER_API_KEY"
      }
  },
]
```

Sub out your Key and your lat/lon (these are not mine)

The api key at first was getting 401 errors, but after a bit it started working, the same key I was using before. [Their FAQ mentions](https://openweathermap.org/faq#error401) it can take a few hours

I am not worried about usage. The default `updateInterval` is `600000` or 10 minutes, meaning with two modules like this I will be hitting less than 300 calls a day if I'm doing math right.

If I'm wrong or missed something let me know and I can edit this.

z-master42, 2024-11-09:
@JHWelch works so far. But how do I get the name of the city back into the header? I now have the name of the city from the corresponding time zone or the capital, i.e. Berlin/Europe.

sdetweil, 2024-11-09:
@z-master42  you need to change the config

set whatever you want in the module header, above the config:{
and set appendLocationNameToHeader:false,
in the config section

z-master42, 2024-11-09:
Damn. I must have gone through the documentary five times. Either I always missed it, or I just didn't realise the function of `appendLocationNameToHeader`.

Thank you!

princemaxwell, 2025-01-02:
After updating from 2.28.0 to 2.30.0 my weather plugin wasn't loading correctly.

The documentation for the weather plugin is not up to date because it says the default value is: apiVersion = '2.5' and weatherEndpoint = '/weather'. 
Here: https://docs.magicmirror.builders/modules/weather.html#openweathermap-options

I saw, that the default was changed to apiVersion = 3.0 and weatherEndpoint = '/onecall' in 2.29.0. I then read the Openweathermap API information. I believe that the API version 2.5 remains, only the /onecall call has been deprecated.

Or have I misunderstood something?

Therefore, after the update to 2.30.0, I adjusted my config as follows and lo and behold, everything works:

```
		{
			module: "weather",
			position: "top_right",
			config: {
				type: "current",
				location: "...",
				locationID: "...",
				apiKey: "...",
				apiVersion: "2.5",
				weatherEndpoint: "/weather",
				degreeLabel: true,
				decimalSymbol: ",",
				showWindDirection: true,
				showWindDirectionAsArrow: false,
				showHumidity: true,
				showFeelsLike: true,
				showSun: true,
			}
		},
		{
			module: "weather",
			position: "top_right",
			header: "Wetter Vorhersage",
			config: {
				type: "forecast",
				location: "...",
				locationID: "...",
				apiKey: "...",
				apiVersion: "2.5",
				weatherEndpoint: "/forecast",
				degreeLabel: true,
				decimalSymbol: ",",
				appendLocationNameToHeader: false,
				fade: false,
				maxNumberOfDays: 5,
				
			}
		},
```

SirStepheno, 2025-01-02:
The OpenWeatherMap must be update to version 3.0 instead of your 2.5. I subscribed to the 3.0 version today and after half an hour it worked. For an explaination how to upgrade, see the reaction of @JHWelch above.

sdetweil, 2025-01-02:
@princemaxwell  tgat is not how we understood the communication from OpenWeather. they have long said they will be discontinuing their 2.5 api (over 2 years)

princemaxwell, 2025-01-02:
@SirStepheno @sdetweil 
I understood that this only applies to OneCall. OneCall API 2.5 was discontinued in June.

For current & forecast, the API documentation still describes 2.5, see here:
https://openweathermap.org/current
https://openweathermap.org/forecast5

It works again for me with the config I just posted.

sdetweil, 2025-01-02:
as I said, that is not the way we read the communications.

princemaxwell, 2025-01-02:
OK, maybe I misunderstood it. I got my information from the current FAQ on the API and it still describes the call to 2.5 and after I adjusted it, it works for me with 2.5, let's see how long.

Nevertheless, the current documentation should still be adjusted to the new default values.

sdetweil, 2025-01-02:
yes

princemaxwell, 2025-01-02:
i can contribute this, but don't know what was changed else.

sdetweil, 2025-01-02:
@princemaxwell  fork the doc repo, https://github.com/MagicMirrorOrg/MagicMirror-Documentation

 switch to develop branch, make a new branch, update as you think,
commit, push, submit pr against doc develop branch

sdetweil, 2025-01-12:
I added a pinned post in the forum 

https://forum.magicmirror.builders/topic/19323/version-2-30-0-weather-module-stuck-loading-worked-fine-before-upgrade

gregimagines, 2025-01-28:
I was having issues with this after the update, but was able to use API 3. I did have to change from using a locationId to lon and lat instead. Once I swapped the location id to my longitude and latitude everything is up and running smoothly with api3 and the latest updates on MM.

sdetweil, 2025-01-29:
@gregimagines yes as the linked forum article says

closed 2025-04-12 by github-actions[bot]

mishu28nmv, 2025-06-17:
So this is my working config:
`{
			module: "weather",
			position: "top_right",
			header: "XXX, DE",
			config: {
				weatherProvider: "openweathermap",
				weatherEndpoint: "/onecall",
				type: "current",
				appendLocationNameToHeader: false,
				lon: 9.XXXXXX,
				lat: 48.XXXXXX,
				apiKey: "XXXX"
				}
		},
		{
			module: "weather",
			position: "top_right",
			header: "XXX, DE",
			config: {
				weatherProvider: "openweathermap",
				weatherEndpoint: "/onecall",
				type: "forecast",
				appendLocationNameToHeader: false,
				lon: 9.XXXXXX,
				lat: 48.XXXXXX,
				apiKey: "XXXX",
				maxNumberOfDays: 8,
				colored: true,
				showPrecipitationAmount: true,
				fade: true,
				fadePoint: 1
				}
		},`

[image]

PR #3583 (2024-10-11): Change default for weatherEndpoint according to API 3.0
since API 3.0 is default, weatherEndpoint should be set to "/onecall"
Fixes #3574

ATTENTION: since lat / lon  defaults to 0 / 0, the weather plugins works after this patch, but shows the weather from https://de.wikipedia.org/wiki/Null_Island if lat / lon is not manually set.
commit 7489e51a (2024-10-11): Change default for weatherEndpoint according to API 3.0 (#3583)

since API 3.0 is default, weatherEndpoint should be set to "/onecall"
Fixes #3574

ATTENTION: since lat / lon defaults to 0 / 0, the weather plugins works
after this patch, but shows the weather from
https://de.wikipedia.org/wiki/Null_Island if lat / lon is not manually
set.

---------