MagicMirrorOrg/MagicMirror #3579: EXT Pir error while command vocal

anideb1975, 2024-10-05:
0|MagicMirror  | [2024-10-05 12:08:02.508] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MagicMirrorOrg/MagicMirror/issues
0|MagicMirror  | [2024-10-05 12:08:03.097] [ERROR] Whoops! There was an uncaught exception...
0|MagicMirror  | [2024-10-05 12:08:03.097] [ERROR] TypeError: this.pir is not a function
0|MagicMirror  |     at Timeout._onTimeout (/home/vincenzodigi/MagicMirror/modules/EXT-Pir/components/pirLib.js:6:2899)
0|MagicMirror  |     at listOnTimeout (node:internal/timers:581:17)
0|MagicMirror  |     at process.processTimers (node:internal/timers:519:7)
0|MagicMirror  | [2024-10-05 12:08:03.098] [ERROR] MagicMirror² will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
0|MagicMirror  | [2024-10-05 12:08:03.101] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MagicMirrorOrg/MagicMirror/issues
0|MagicMirror  | [2024-10-05 12:08:03.752] [ERROR] Whoops! There was an uncaught exception...
0|MagicMirror  | [2024-10-05 12:08:03.753] [ERROR] TypeError: this.pir is not a function
0|MagicMirror  |     at Timeout._onTimeout (/home/vincenzodigi/MagicMirror/modules/EXT-Pir/components/pirLib.js:6:2899)
0|MagicMirror  |     at listOnTimeout (node:internal/timers:581:17)
0|MagicMirror  |     at process.processTimers (node:internal/timers:519:7)
0|MagicMirror  | [2024-10-05 12:08:03.775] [ERROR] MagicMirror² will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
0|MagicMirror  | [2024-10-05 12:08:03.776] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MagicMirrorOrg/MagicMirror/issues
0|MagicMirror  | [2024-10-05 12:08:04.103] [ERROR] Whoops! There was an uncaught exception...
0|MagicMirror  | [2024-10-05 12:08:04.107] [ERROR] TypeError: this.pir is not a function
0|MagicMirror  |     at Timeout._onTimeout (/home/vincenzodigi/MagicMirror/modules/EXT-Pir/components/pirLib.js:6:2899)
0|MagicMirror  |     at listOnTimeout (node:internal/timers:581:17)
0|MagicMirror  |     at process.processTimers (node:internal/timers:519:7)
0|MagicMirror  | [2024-10-05 12:08:04.122] [ERROR] MagicMirror² will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
0|MagicMirror  | [2024-10-05 12:08:04.123] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MagicMirrorOrg/MagicMirror/issues

bugsounet, 2024-10-05:
[Fixed](https://github.com/bugsounet/EXT-Pir/pull/78)

khassel, 2024-10-05:
@anideb1975 

from the issue template:

> ## I'm having troubles installing or configuring foreign modules
> Please open an issue in the module repository or ask for help in the [forum](https://forum.magicmirror.builders/)

closed 2024-10-05 by khassel