MagicMirrorOrg/MagicMirror #3582: Wayland support

goldyfruit, 2024-10-08:
Hi,

Raspberry Pi OS Bookworm is now defaulting to Wayland and Wayfire and because of this change MagicMirror is not able to start via the standard `npm run start` as its looking for `DISPLAY` to be set which doesn't exist anymore on Wayland.

Maybe adding a `start-wayland` option to `package.json` will be a requirement in order to allow user running on modern distribution to use MagicMirror.

This is how I'm running it  and its working very well so far:

```
`WAYLAND_DISPLAY="${DISPLAY:=wayland-1}" ./node_modules/.bin/electron js/electron.js --enable-features=UseOzonePlatform --ozone-platform=wayland`
```

KristjanESPERANTO, 2024-10-08:
I'm running Debian 11 with Wayland but without Wayfire and I have no problems with `npm run start`. So maybe Wayfire is the "problem" and not Wayland?

goldyfruit, 2024-10-08:
I'm running "pure" Wayland _(not xwayland)_, same issue happens with Sway.

KristjanESPERANTO, 2024-10-08:
Oh okay, I just checked: Electron is running under xwayland on my system.

I'm just thinking out loud: Would it be possible to put a check into "npm run start" to see if Wayland or X is running and call the right command? This way we could also check if Windows is running (they also need another command).

bugsounet, 2024-10-22:
hmm, interesting :)

I just write a test script for test it

----
```
pi@raspberrypi:~ $ ./test.sh 
Running: Wayland
pi@raspberrypi:~ $ 
```
----
```
pi@raspberrypi2:~ $ ./test.sh 
Running: X11
pi@raspberrypi2:~ $ 
```
----

now the question is... what language to use for code it ?
bash ? or node ?

* if we use bash we have linux / mac compatibilities BUT not windows
* if we use node I think we have both compatibility

Or ... just add un new start line with something like this: `npm start wayland`, `npm start windows`

khassel, 2024-10-22:
> just add un new start line with something like this: npm start wayland, npm start windows

I think this is sufficient.

The wayland scenario is an edge case at the moment, in all raspian os images with desktop XWayland is installed which works with X11 as usual.

bugsounet, 2024-10-22:
@khassel ok ;)

@KristjanESPERANTO , do you want to code it ?  (i'm lazy mouahah)

sdetweil, 2024-10-22:
a few years ago i wrote a neutral wrapper for start, post install see #2571 
https://raw.githubusercontent.com/sdetweil/MMM-Config/refs/heads/main/exec_script.js

sdetweil, 2024-10-22:
so you don't need different start, and couldn't have multiple postinstall clauses

khassel, 2024-10-22:
I think we
- should prevent current start commands
- can add new start lines for windows, wayland, ...
- can add new start line for autodetect
- should not force users to use autodetect with every start (e.g. there is no need for it in a container)

KristjanESPERANTO, 2024-10-23:
I created a draft PR #3594 for further investigation and discussion.

It is interesting to note that `npm run start:wayland` seems to run more smoothly (ticker texts) than `npm run start` on my Debian system. But I haven't investigated this further, and my tired senses may be deceiving me. Since X11 development is dead, I'll try this as my new default.

closed 2025-01-01 by sdetweil