MagicMirrorOrg/MagicMirror #3597: date based electron tests are using today instead of specific date

khassel, 2024-10-23:
see https://github.com/MagicMirrorOrg/MagicMirror/pull/3596

The helper is not working (anymore) so every (electron) test running with a specific date does not run with this date but uses today.

Above PR should be reverted if this is solved.

sdetweil, 2024-10-23:
yes, the 'work around' is to code the Date() setting IN the config.js file for the tescase instead of in the calendar_spec.js runner (which uses helper) 
my newly added testcases have it in the config.js testcase file.. and also in the runner.. copied other working testcases.. 

in the config.js for the testcase works for local debugging

khassel, 2024-10-23:
but you have to be careful coding this in config because there are config's which are used in several tests ...

sdetweil, 2024-10-23:
yes, this one (ICS) 
```
auth-default.js:						url: "http://localhost:8080/tests/mocks/calendar_test.ics",
basic-auth.js:						url: "http://localhost:8080/tests/mocks/calendar_test.ics",
changed-port.js:						url: "http://localhost:8010/tests/mocks/calendar_test.ics",
default.js:						url: "http://localhost:8080/tests/mocks/calendar_test.ics"
fail-basic-auth.js:						url: "http://localhost:8020/tests/mocks/calendar_test.ics",
old-basic-auth.js:						url: "http://localhost:8080/tests/mocks/calendar_test.ics",
show-duplicates-in-calendar.js:						url: "http://localhost:8080/tests/mocks/calendar_test.ics" 
```
but each has diff .js file, which contains the Date() function

sdetweil, 2024-10-23:
those other tests are used in e2e  and do not care about the number of events posted..

sdetweil, 2024-10-25:
i have merged develop now that #3599 is merged

BUT the testcases in Electron  still have  Date().... in them  did you intend for these to be changed too so that only date in runner applies?

```
grep "new Date" *.js
3_move_first_allday_repeating_event.js:	return new Date("01 Oct 2024 10:38:00 GMT+2:00").valueOf();
berlin_end_of_day_repeating.js:	return new Date("08 Oct 2024 12:30:00 GMT+02:00").valueOf();
berlin_multi.js:	return new Date("08 Oct 2024 12:30:00 GMT+02:00").valueOf();
berlin_whole_day_event_moved_over_dst_change.js:	return new Date("08 Oct 2024 12:30:00 GMT+02:00").valueOf();
chicago_late_in_timezone.js:	return new Date("01 Sept 2024 10:38:00 GMT-05:00").valueOf();
diff_tz_start_end.js:	return new Date("08 Oct 2024 12:30:00 GMT-07:00").valueOf();
end_of_day_berlin_moved.js:	return new Date("08 Oct 2024 12:30:00 GMT+02:00").valueOf();
event_with_time_over_multiple_days_non_repeating_display_end.js:	return new Date("15 Sep 2024 12:30:00 GMT").valueOf();
event_with_time_over_multiple_days_non_repeating_no_display_end.js:	return new Date("15 Sep 2024 12:30:00 GMT").valueOf();
exdate_and_recurrence_together.js:	return new Date("08 Oct 2024 12:30:00 GMT+07:00").valueOf();
exdate_la_at_midnight_dst.js:	return new Date("19 Oct 2023 12:30:00 GMT-07:00").valueOf();
exdate_la_at_midnight_std.js:	return new Date("19 Oct 2023 12:30:00 GMT-07:00").valueOf();
exdate_la_before_midnight.js:	return new Date("19 Oct 2023 12:30:00 GMT-07:00").valueOf();
exdate_syd_at_midnight_dst.js:	return new Date("14 Sep 2023 12:30:00 GMT+10:00").valueOf();
exdate_syd_at_midnight_std.js:	return new Date("14 Sep 2023 12:30:00 GMT+10:00").valueOf();
exdate_syd_before_midnight.js:	return new Date("14 Sep 2023 12:30:00 GMT+10:00").valueOf();
fullday_event_over_multiple_days_nonrepeating.js:	return new Date("15 Sep 2024 12:30:00 GMT").valueOf();
germany_at_end_of_day_repeating.js:	return new Date("01 Oct 2024 10:38:00 GMT+2:00").valueOf();
rrule_until.js:	return new Date("07 Mar 2024 10:38:00 GMT-07:00").valueOf();
show-duplicates-in-calendar.js:	return new Date("15 Sep 2024 12:30:00 GMT").valueOf();
sliceMultiDayEvents.js:	return new Date("01 Sept 2024 10:38:00 GMT+2:00").valueOf();
```

sdetweil, 2024-10-25:
also in helper 
we don't need this  asits all about setting the date, which is done a different way now
```
	for (const win of global.electronApp.windows()) {
		const title = await win.title();
		expect(["MagicMirror²", "DevTools"]).toContain(title);
		if (title === "MagicMirror²") { <---- here  -------------------
			global.page = win;
			if (systemDate) {
				await global.page.evaluate((systemDate) => {
					Date.now = () => {
						return new Date(systemDate).valueOf();
					};
				}, systemDate);
			}
		}  <---- thru here  -------------------------
	}

sdetweil, 2024-10-25:
my helper suggestion however causes failure to locate content

khassel, 2024-10-25:
I did several tests yesterday and as I wrote already in the PR, I think we need both, the date mocking in the browser and the date mocking in the backend (maybe except for modules running completly in the browser as weather).

But I'm happy to merge a simpler solution ...

khassel, 2024-10-25:
> but the Date... in the actual testcases is not needed now..

do you mean this e.g. in `exdate_syd_at_midnight_dst.js`:

```js
Date.now = () => {
	return new Date("19 Oct 2023 12:30:00 GMT-07:00").valueOf();
};
```

was not aware of setting dates in the configs.

So now I understand your grep statement, will you provide a PR for removing this? Otherwise I can do this ...

khassel, 2024-10-25:
as test stuff is running on `develop` we don't have to wait for this until next release, so closing this now as completed.

PR #3599 (2024-10-25): fix electron tests mocking dates
fixes #3597 

Changes:
- electron tests: add mocking to `electron.js` for mocking the server side, before only the browser side was mocked
- publish "/tests/configs" and "/tests/mocks" always in `server.js`, this reverts a change done with latest release, we need this for debugging (otherwise you get on the screen that your config has errors but config check is successful ...)
- revert hotfix in `tests/configs/modules/calendar/show-duplicates-in-calendar.js`
- fix `tests/configs/modules/calendar/custom.js` to allow events in the past (~~I don't know how this co [... cut]
commit cfa5c0d1 (2024-10-25): fix electron tests mocking dates (#3599)

fixes #3597 

Changes:
- electron tests: add mocking to `electron.js` for mocking the server
side, before only the browser side was mocked
- publish "/tests/configs" and "/tests/mocks" always in `server.js`,
this reverts a change done with latest release, we need this for
debugging (otherwise you get on the screen that your config has errors
but config check is successful ...)
- revert hotfix in
`tests/configs/modules/calendar/show-duplicates-in-calendar.js`
- fix `tests/configs/modules/calendar/custom.js` to allow events in the
past (~~I don't know ho [... cut]
closed 2024-10-25 by khassel