MagicMirrorOrg/MagicMirror #3603: Default calendar: nextDaysRelative not working

evroom, 2024-10-26:
Default calendar: the nextDaysRelative parameter seems to not work.
The timeformat remains absolute for events on today and tomorrow.

```
$ grep version /home/admin/MagicMirror/package.json
	"version": "2.30.0-develop",
```

```
nextDaysRelative
If this is set to true, the appointments of today and tomorrow are displayed relatively,
even if the timeformat is set to absolute.
```

timeFormat: "absolute",
nextDaysRelative: false,
[image]

timeFormat: "absolute",
nextDaysRelative: true,
[image]

sdetweil, 2024-10-26:
can you post config please

evroom, 2024-10-26:
Did the git pull on approx. Sat 26 Oct 10:45:00 CEST 2024.

config.js (removed personal entries):

[config.js.txt](https://github.com/user-attachments/files/17530200/config.js.txt)

sdetweil, 2024-10-26:
nextDaysRelative is only for full day events  AND timeFormat : "absolute" according to the code

you only showed duration events, not full day

evroom, 2024-10-26:
Okay, that is not mentioned in the documentation.

But indeed that is what it does:

[image]

sdetweil, 2024-10-26:
i think its encoded in the name 
next  DAYS relative

evroom, 2024-10-26:
For a Sherlock Holmes like you perhaps :-)
It wasn't clear for a regular user like me.
I read next days as today, tomorrow, day after tomorrow.
The text narrowed this down to today and tomorrow. 

Opened this anyway:
https://github.com/MagicMirrorOrg/MagicMirror-Documentation/issues/267

Please close this issue (#3603) when it is meant to be for fullday events only.
Otherwise the issue will address the fact that it is not working non-fullday events events.

sdetweil, 2024-10-26:
relative only matters for full day events

Sunday, 16:00-18:00  displayed as Tomorrow seems illogical to me

evroom, 2024-10-26:
With `timeFormat: "relative"` you'll get:
[image]
Makes sense to me.

So if I get the same result with `timeFormat: "absolute"` and `nextDaysRelative` then I think that would be nice.

But when 'we' leave it be, then it would be nice for a re-wording in the calendar document just make it more clear.

PS
```
timeFormat: "relative",
dateFormat: "ddd MMM D - HH:mm",
dateEndFormat: "ddd MMM D - HH:mm",
fullDayEventDateFormat: "ddd MMM D",
showEnd: true,
showEndsOnlyWithDuration: true,
nextDaysRelative: false
```
Does `showEnd` apply to timed events too, or only to fullday event ?
Because with the above settings, I do not see the date end.
All my latest testing was only done with `timeFormat: "absolute"`, whereas my "production" MagicMirror uses `relative` (which I need to give a second thought).

sdetweil, 2024-10-26:
absolute and releative produce different results

>Does showEnd apply to timed events too, or only to fullday event ?

see #3602 

fixed in test version
I have just updated the doc (develop branch for nextDaysRelative and showEndsOnlyWithDuration

closed 2025-01-01 by sdetweil