MagicMirrorOrg/MagicMirror #3609: New RaspiOS releases dates Oct 22 2024, use node 23

sdetweil, 2024-10-29:
we don't install on node 23 due to the engines check.. 

i deleted package-lock and change engines to  || 23

and it seems to run fine

develop does

```
node -v
v23.1.0
```

testcases  on develop fail on macos Sonoma 14.5, run ok on ubuntu 22.04

[image]

```
Mac-mini:MagicMirror sam$ npm run test:electron

> magicmirror@2.30.0-develop test:electron
> NODE_ENV=test jest --selectProjects electron -i --forceExit

Running one project: electron
 PASS   electron  tests/electron/env_spec.js (5.545 s)
  Electron app environment
    ✓ should open browserwindow (3408 ms)
  Development console tests
    ✓ should open browserwindow and dev console (1842 ms)

 FAIL   electron  tests/electron/modules/calendar_spec.js (175.894 s)
  Calendar module
    Test css classes
      ✓ has css class dayBeforeYesterday (2766 ms)
      ✕ has css class yesterday (20046 ms)
      ✕ has css class today (20041 ms)
      ✕ has css class tomorrow (20048 ms)
      ✕ has css class dayAfterTomorrow (20037 ms)
    Events from multiple calendars
      ✓ should show multiple events with the same title and start time from different calendars (2655 ms)
    rrule
      ✓ Issue #3393 recurrence dates past rrule until date (2403 ms)
    Exdate: LA crossover DST before midnight GMT
      ✕ LA crossover DST before midnight GMT should have 2 events (20036 ms)
    Exdate: LA crossover DST at midnight GMT local STD
      ✓ LA crossover DST before midnight GMT should have 2 events (2664 ms)
    Exdate: LA crossover DST at midnight GMT local DST
      ✓ LA crossover DST before midnight GMT should have 2 events (2380 ms)
    Exdate: SYD crossover DST before midnight GMT
      ✕ LA crossover DST before midnight GMT should have 2 events (20042 ms)
    Exdate: SYD crossover DST at midnight GMT local STD
      ✕ LA crossover DST before midnight GMT should have 2 events (20037 ms)
    Exdate: SYD crossover DST at midnight GMT local DST
      ✕ SYD crossover DST at midnight GMT local DST should have 2 events (20047 ms)
    sliceMultiDayEvents
      ✓ Issue #3452 split multiday in Europe (2595 ms)

  ● Calendar module › Test css classes › has css class yesterday

    thrown: "Exceeded timeout of 20000 ms for a test.
    Add a timeout value to this test to increase the timeout, if this is a long-running test. See https://jestjs.io/docs/api#testname-fn-timeout."

      33 | 		});
      34 |
    > 35 | 		it("has css class yesterday", async () => {
         | 		^
      36 | 			await helpers.startApplication("tests/configs/modules/calendar/custom.js", "02 Jan 2030 12:30:00 GMT");
      37 | 			await expect(doTest(".yesterday")).resolves.toBe(true);
      38 | 		});

[... 208 lines cut ...]
      ✕ should render sunset (97 ms)

  ● Weather module › Current weather with sunset › should render sunset

    expect(received).resolves.toBe()

    Received promise rejected instead of resolved
    Rejected to value: [Error: expect(received).toBe(expected) // Object.is equality·
    Expected: "3:45 pm"
    Received: "7:00 am"]

      25 |
      26 | 		it("should render sunset", async () => {
    > 27 | 			await expect(weatherHelper.getText(".weather .normal.medium span:nth-child(4)", "3:45 pm")).resolves.toBe(true);
         | 			      ^
      28 | 		});
      29 | 	});
      30 | });

      at expect (node_modules/expect/build/index.js:113:15)
      at Object.expect (tests/electron/modules/weather_spec.js:27:10)

Test Suites: 3 failed, 1 passed, 4 total
Tests:       12 failed, 15 passed, 27 total
```

sdetweil, 2024-10-29:
should we put out a point release on package.json
i will fix my installer  in the morning
i will also update package.json

khassel, 2024-10-29:
I installed this new raspian and there is no node preinstalled. 

```bash
pi@pi4-argon:~ $ node -v
-bash: node: command not found
```
If I install with apt I get an outdated 

```bash
pi@pi4-argon:~ $ node -v
v18.19.0
```

sdetweil, 2024-10-29:
what image did uninstall? i did 64 bit full used the imager for pi4

sdetweil, 2024-10-29:
false alarm   user didn't show me all the info .. they installed and got node 23 manually..

then did npm install, which failed..

sdetweil, 2024-10-29:
i fixed my script, but won't change it back..

now to find where they save the screensaver info on new bookworm as the wayfire.ini is gone and labwc doesn't do it natively

khassel, 2024-10-29:
so can we close this?

sdetweil, 2024-10-29:
yes

closed 2024-10-29 by sdetweil