MagicMirrorOrg/MagicMirror #3610: in test mode is there a way for browser side to know in test?

sdetweil, 2024-10-29:
in helper side one can test the  process.env.JEST_WORKER_ID
not available in module side

I am helping user with new compliments parm test

khassel, 2024-10-29:
from `tests/e2e/helpers/global-setup.js`:

```js
exports.getDocument = () => {
	return new Promise((resolve) => {
		const url = `http://${config.address || "localhost"}:${config.port || "8080"}`;
		jsdom.JSDOM.fromURL(url, { resources: "usable", runScripts: "dangerously" }).then((dom) => {
			dom.window.name = "jsdom";
			global.window = dom.window;
			// Following fixes `navigator is not defined` errors in e2e tests, found here
			// https://www.appsloveworld.com/reactjs/100/37/mocha-react-navigator-is-not-defined
			global.navigator = {
				useragent: "node.js"
			};
			dom.window.fetch = fetch;
			dom.window.onload = () => {
				global.document = dom.window.document;
				resolve();
			};
		});
	});
};
```

we are setting here `dom.window.name = "jsdom";` so this could maybe used in the e2e-tests.

sdetweil, 2024-10-29:
I meant the code running inside MM.. a module

to test the refreshable compliments remote file and have it locally we need to 'change' the configured url on the fly to it will pull a different file.  but i have another way... if the url starts with tests and there is another parm (url2...)

khassel, 2024-10-29:
> I meant the code running inside MM.. a module

and I meant you maybe can use 

```js
if (global.window.name === "jsdom")` {
  console.log("running in test");
}
```

sdetweil, 2024-11-01:
ok, thx

sdetweil, 2024-11-08:
```

khassel, 2024-11-08:
`In the browser environment, global variables are attached to the `window` object.`

so for code running in browser omit the `global`:

```js
if (window.name === "jsdom")` {
  console.log("running in test");
}
```

sdetweil, 2024-11-09:
hm.. doesn't fail, but doesn't work 
```js 
		// in test mode only
		if (window.name === "jsdom") {
			// check for (undocumented) remote file2 to change test new load
			if(this.config.remoteFile2!== null && this.config.remoteFileRefreshInterval!==null){
	 			 console.log("running in test");
	 			 this.config.remoteFile=this.config.remoteFile2
			}
		}
```
for changing the remote file to prove that the cycle and fetch work , only wanted to fake a new file in test mode
if I comment out the if jsdom, then testcase works.
but fails with it enabled. 

in test and in non-test window.name is empty
console.log("name='+window.name+"'") 
results in name=''

this is in the getDom() callback

closed 2024-11-09 by sdetweil

sdetweil, 2024-11-09:
code in pr #3630