MagicMirrorOrg/MagicMirror #3613: MagicMirror loading fails behind reverse proxy after upgrading to v2.29 due to missing environment variables

XTrandi, 2024-11-01:
**Platform**: Raspbian GNU/Linux 11 (bullseye) on Raspberry Pi 2

**Node Version**: 22.11.0

**MagicMirror² Version**: 2.29.0

**Description**: After upgrading MagicMirror from v2.28.0 to v2.29.0 my web interface won't display anything. Turning on developer tools displays missing /env subpath resulting in an exception in `loader.js` / `main.js` and subsequent loading.
[image]

**Steps to Reproduce**: I use a nginx webserver, which uses a reverse proxy on /apps/MagicMirror to redirect it to hostname:8080. My `config.js` has the reverse proxy included.

```javascript
let config = {
	address: "0.0.0.0",
	port: 8080,
//	basePath: "/",
	basePath: "/apps/MagicMirror/", 	// The URL path where MagicMirror² is hosted. If you are using a Reverse proxy
					// you must set the sub path here. basePath must end with a /
	...
};
``` 

**Expected Results**: As for previous versions socket.io uses the basePath to establish socket connection between client and server. The initial loading for loading environment variables should, too.

**Actual Results**: MagicMirror loading fails. As a workaround I modified [`loader.js`](https://github.com/MagicMirrorOrg/MagicMirror/blob/94c3c699e8bf89e0466580960d5d32b745dd940c/js/loader.js#L18C1-L18C73) to
```javascript
  const res = await fetch(`${location.protocol}//${location.host}/` + config.basePath + `/env`);
``` 
in order to make my application work again. Probably needs further refinement and exception handling.

khassel, 2024-11-01:
thanks, we should have a test for this ...

khassel, 2024-11-01:
> we should have a test for this ...

but I have no idea for this without a reverse proxy ...

sdetweil, 2024-11-12:
so, i created a simple proxy   it works for whatever I code to use the port (config +10) 
but not everything is sent thru the proxy.
    basePath:"/apps/MagicMirror/",

server.js
in testmode (#3631)

```js
			if(process.env.mmTestMode === 'true' ){
				const httpProxy = require('http-proxy')
				var proxy = httpProxy.createProxyServer({});

				//
				//  proxies the request, but removes the basePath as we are in the same location
				// this is just for testing the basePath function
				//
				console.log("setup proxy server")
				http.createServer(function (req, res) {
				  // This simulates an operation that takes 500ms to execute
				    const requestedUrl = req.url;
  					console.log("proxy for="+requestedUrl);
  					if(config.basePath.length>1 && requestedUrl.startsWith(config.basePath)){
  						req.url=requestedUrl.slice(config.basePath.length)
  					}
  					// send the request to the normal port
				    proxy.web(req, res, {
				      target: 'http://localhost:'+port
				    });
				}).listen(port+10);
			}
```

sdetweil, 2024-11-12:
log of requests thru proxy... only socket.io uses the basePath prefix
```text
[2024-11-12 10:27:49.178] [LOG]   proxy for=/apps/MagicMirror/socket.io/?EIO=4&transport=polling&t=o1lq0dc6&sid=mfop4E6RGkZm80rWAAAL 
[2024-11-12 10:27:49.178] [LOG]   proxy for=/modules/MMM-MealieMenu//MMM-MealieMenu.njk?s=1731428869176 
[2024-11-12 10:27:49.179] [LOG]   proxy for=/apps/MagicMirror/socket.io/?EIO=4&transport=polling&t=o1lq0ml2&sid=VVAvAH20rHv4prhHAAAA 
[2024-11-12 10:27:49.180] [LOG]   proxy for=/modules/MMM-MealieMenu//MMM-MealieMenu.njk?s=1731428869177 
[2024-11-12 10:27:49.237] [LOG]   proxy for=/modules/default/newsfeed//newsfeed.njk?s=1731428869236 
[2024-11-12 10:27:54.091] [LOG]   proxy for=/apps/MagicMirror/socket.io/?EIO=4&transport=polling&t=o1pii4km&sid=mfop4E6RGkZm80rWAAAL 
[2024-11-12 10:27:54.092] [LOG]   proxy for=/apps/MagicMirror/socket.io/?EIO=4&transport=polling&t=o1piixhy&sid=mfop4E6RGkZm80rWAAAL 
[2024-11-12 10:27:57.289] [LOG]   proxy for=/socket.io/?EIO=4&transport=polling&t=PCXD0Fd&sid=KlLUkZ_qGwJxsjlTAAAI 
[2024-11-12 10:27:57.289] [LOG]   proxy for=/socket.io/?EIO=4&transport=polling&t=PCXD0Fe&sid=KlLUkZ_qGwJxsjlTAAAI 
[2024-11-12 10:27:59.237] [LOG]   proxy for=/modules/default/newsfeed//newsfeed.njk?s=1731428879235 
[2024-11-12 10:28:09.239] [LOG]   proxy for=/modules/default/newsfeed//newsfeed.njk?s=1731428889237 
[2024-11-12 10:28:09.509] [LOG]   proxy for=/apps/MagicMirror/socket.io/?EIO=4&transport=polling&t=o21er6lf&sid=VVAvAH20rHv4prhHAAAA 
[2024-11-12 10:28:09.510] [LOG]   proxy for=/apps/MagicMirror/socket.io/?EIO=4&transport=polling&t=o21esqiq&sid=VVAvAH20rHv4prhHAAAA 
[2024-11-12 10:28:19.107] [LOG]   proxy for=/apps/MagicMirror/socket.io/?EIO=4&transport=polling&t=o28td0nx&sid=mfop4E6RGkZm80rWAAAL 
[2024-11-12 10:28:19.108] [LOG]   proxy for=/apps/MagicMirror/socket.io/?EIO=4&transport=polling&t=o28tecyy&sid=mfop4E6RGkZm80rWAAAL 
[2024-11-12 10:28:19.237] [LOG]   proxy for=/modules/default/newsfeed//newsfeed.njk?s=1731428899236 
[2024-11-12 10:28:22.295] [LOG]   proxy for=/socket.io/?EIO=4&transport=polling&t=PCXD6MM&sid=KlLUkZ_qGwJxsjlTAAAI 
[2024-11-12 10:28:22.296] [LOG]   proxy for=/socket.io/?EIO=4&transport=polling&t=PCXD6MM.0&sid=KlLUkZ_qGwJxsjlTAAAI 
[2024-11-12 10:28:29.237] [LOG]   proxy for=/modules/default/newsfeed//newsfeed.njk?s=1731428909235 
[2024-11-12 10:28:30.253] [LOG]   proxy for=/apps/MagicMirror/socket.io/?EIO=4&transport=polling&t=o2hezy3u&sid=mfop4E6RGkZm80rWAAAL 
[2024-11-12 10:28:30.255] [INFO]  [MMM-MealieMenu] Week starts: 2024-11-11, next week starts: 2024-11-18 
[2024-11-12 10:28:30.541] [INFO]  Mealie error={} 
[2024-11-12 10:28:30.550] [LOG]   proxy for=/apps/MagicMirror/socket.io/?EIO=4&transport=polling&t=o2hn7iig&sid=mfop4E6RGkZm80rWAAAL 
[2024-11-12 10:28:30.550] [LOG]   proxy for=/modules/MMM-MealieMenu//MMM-MealieMenu.njk?s=1731428910548 
[2024-11-12 10:28:30.551] [LOG]   proxy for=/apps/MagicMirror/socket.io/?EIO=4&transport=polling&t=o2hn8hdp&sid=VVAvAH20rHv4prhHAAAA 
[2024-11-12 10:28:30.552] [LOG]   proxy for=/modules/MMM-MealieMenu//MMM-MealieMenu.njk?s=1731428910549 
[2024-11-12 10:28:34.518] [LOG]   proxy for=/apps/MagicMirror/socket.io/?EIO=4&transport=polling&t=o2kpftyg&sid=VVAvAH20rHv4prhHAAAA 
[2024-11-12 10:28:34.519] [LOG]   proxy for=/apps/MagicMirror/socket.io/?EIO=4&transport=polling&t=o2kpgww0&sid=VVAvAH20rHv4prhHAAAA 
[2024-11-12 10:28:39.237] [LOG]   proxy for=/modules/default/newsfeed//newsfeed.njk?s=1731428919235 
```

PR #3620 (2024-11-02): fix missing basePath
fixes #3613 

wanted to write a test for `basePath` but have no idea at the moment to simulate this without a reverse proxy.

Here my test setup for documentation:
```yaml
networks:
  proxy:
    driver: bridge

services:
  socket-proxy:
    privileged: true
    image: tecnativa/docker-socket-proxy:edge
    container_name: socket-proxy
    restart: unless-stopped
    volumes:
      - /var/run/docker.sock:/var/run/docker.sock:ro
    environment:
      CONTAINERS: 1
    ports:
      - "127.0.0.1:2375:2375"
    networks:
      - proxy

  traefik:
    image: traefik:latest
     [... cut]
commit 9114aefe (2024-11-02): fix missing basePath (#3620)

fixes #3613 

wanted to write a test for `basePath` but have no idea at the moment to
simulate this without a reverse proxy.

Here my test setup for documentation:
```yaml
networks:
  proxy:
    driver: bridge

services:
  socket-proxy:
    privileged: true
    image: tecnativa/docker-socket-proxy:edge
    container_name: socket-proxy
    restart: unless-stopped
    volumes:
      - /var/run/docker.sock:/var/run/docker.sock:ro
    environment:
      CONTAINERS: 1
    ports:
      - "127.0.0.1:2375:2375"
    networks:
      - proxy

  traefik:
    image: traefik:la [... cut]
closed 2025-01-01 by sdetweil