MagicMirrorOrg/MagicMirror #3629: Compatibility with nodejs 23

lsiepel, 2024-11-08:
The installation instructions suggest to install the latest node.js version:
https://docs.magicmirror.builders/getting-started/installation.html#manual-installation

That is currently 23.x, so i installed it and started the mm installation:"

```bash
npm run install-mm

npm error code EBADENGINE
npm error engine Unsupported engine
npm error engine Not compatible with your version of node/npm: magicmirror@2.29.0
npm error notsup Not compatible with your version of node/npm: magicmirror@2.29.0
npm error notsup Required: {"node":">=20.9.0 <21 || 22"}
npm error notsup Actual:   {"npm":"10.9.0","node":"v23.1.0"}
```

So either the text with `latest` needs to be adapted or support for 23.x is needed.

sdetweil, 2024-11-08:
coming next release, jan 1
you can get the develop branch now to test

lsiepel, 2024-11-09:
Solved by the linked PR. Thanks

closed 2024-11-09 by lsiepel