MagicMirrorOrg/MagicMirror #3632: linting/prettier not working anymore on `develop`

khassel, 2024-11-12:
see https://github.com/MagicMirrorOrg/MagicMirror/pull/3630 as example

I'm far away from this topic, what I found: With release `v2.26.0` `*.js` was added to `.prettierignore` so from this point on prettier is ignoring all js files.

Maybe @rejas @KristjanESPERANTO can look at this, thanks!

KristjanESPERANTO, 2024-11-12:
I'm onto it.

KristjanESPERANTO, 2024-11-12:
> linting/prettier not working anymore

The ignore pattern in the ESLint config caused the default modules to no longer be linted.

> *.js was added to .prettierignore so from this point on prettier is ignoring all js files.

Yes this was intended. Since we're using ESLint Stylistic to format the JavaScript code, we don't need to format these files with a second formatter (prettier).

PR #3633 should fix this.

KristjanESPERANTO, 2024-11-12:
That was certainly my fault when I converted the deprecated ESLint config to flat config.

khassel, 2024-11-12:
no problem, thanks for fixing, I didn't see it ...

PR #3633 (2024-11-12): Fix eslint ignores
This will fix #3632.
closed 2024-11-12 by khassel