MagicMirrorOrg/MagicMirror #3635: newbe : using node.js v23.2.0  then npm run install-mm gives npm errorors

tozu-dot, 2024-11-16:
Hello and thank you for opening an issue.

**⚠️ Please make sure that you have read the following lines before submitting your Issue:**

## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [npm(https://forum.magicmirror.builders/category/10/troubleshooting)

A common problem is that your config file could be invalid. Please run in your MagicMirror² directory: `npm run config:check` and see if it reports an error.

## I found a bug in the MagicMirror² installer

If you are facing an issue or found a bug while trying to install MagicMirror² via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror² Docker image

If you are facing an issue or found a bug while running MagicMirror² inside a Docker container please create an issue in the corresponding repository:
[https://gitlab.com/khassel/magicmirror](https://gitlab.com/khassel/magicmirror)

## I'm having troubles installing or configuring foreign modules

Please open an issue in the module repository or ask for help in the [forum](https://forum.magicmirror.builders/)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3/4, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 18 or later (recommended is 20).

**MagicMirror² Version**: Please let us know which version of MagicMirror² you are running. It can be found in the `package.json` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

sdetweil, 2024-11-16:
yes, uninstall node 23

install node 20.9.0, any node 22
 not any node 21

tozu-dot, 2024-11-16:
the errors are:magicmirror@2.29.0 install-mm
> npm install --no-audit --no-fund --no-update-notifier --only=prod --omit=dev

(node:7995) ExperimentalWarning: CommonJS module /usr/lib/node_modules/npm/node_modules/debug/src/node.js is loading ES Module /usr/lib/node_modules/npm/node_modules/supports-color/index.js using require().
Support for loading ES Module in require() is an experimental feature and might change at any time
(Use `node --trace-warnings ...` to show where the warning was created)
npm error code EBADENGINE
npm error engine Unsupported engine
npm error engine Not compatible with your version of node/npm: magicmirror@2.29.0
npm error notsup Not compatible with your version of node/npm: magicmirror@2.29.0
npm error notsup Required: {"node":">=20.9.0 <21 || 22"}
npm error notsup Actual:   {"npm":"10.9.0","node":"v23.2.0"}
npm error A complete log of this run can be found in:

so could someone show in simple words how to get this corrected?

tozu-dot, 2024-11-16:
how do I remove the currrent node.js v23

sdetweil, 2024-11-16:
dont know.

sdetweil, 2024-11-16:
the requirements for MagicMirror  say node 20 or 22

sdetweil, 2024-11-16:
[image]

tozu-dot, 2024-11-16:
the commands are sudo apt remove nodejs and sudo apt remove npm  acoording to internet info ,  will do that and continue thanks

closed 2024-11-16 by tozu-dot