MagicMirrorOrg/MagicMirror #3654: locale is included in the sample config but not the documentation

mmclane, 2024-12-13:
Hello and thank you for opening an issue.

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3/4, Windows, Mac, Linux, System V UNIX).

NA

**Node Version**: Make sure it's version 18 or later (recommended is 20).

NA

**MagicMirror² Version**: Please let us know which version of MagicMirror² you are running. It can be found in the `package.json` file.

v2.29.0

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

I am setting up MM for the first time.  I was going through the sample config that I got here:  https://github.com/MagicMirrorOrg/MagicMirror/blob/master/config/config.js.sample

It has a setting for locale.  That configuration option isn't included in the documentation here: https://docs.magicmirror.builders/configuration/introduction.html

I don't know what it does or how it should be configured.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

See provided URLS

**Expected Results**: Describe what you expected to see.

The documentation includes locale and I know what to do with it.

**Actual Results**: Describe what you actually saw.

NA

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

NA

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

sdetweil, 2024-12-13:
thats cause we don't use it.. it was added in early 2021 for use by third party modules

https://github.com/MagicMirrorOrg/MagicMirror/pull/2461

rejas, 2024-12-14:
So maybe it would make sense to update the docs with this info?

sdetweil, 2024-12-14:
i started down that path, but then realized i dont know what should be there or a good reference

khassel, 2024-12-14:
I think it is sufficient to add a comment in the config sample file that this property is only added for use by third party modules and is not used in mm core (and default modules)

sdetweil, 2024-12-17:
opened pr #3655

closed 2025-01-01 by sdetweil