MagicMirrorOrg/MagicMirror #3675: Weather Module After Update Not loading

Themaster312, 2025-01-02:
Hello After the last Mirror Update the weather says Invalid Date and Nan and it doesnt load

sdetweil, 2025-01-02:
are you using an appropriate apikey. note that openweather changed their business model, requiring an apikey backed by a credit card. v3

the defaults for the module were changed to that requirement

Themaster312, 2025-01-02:
How do I get api 3.0 in openweather? I have a 2.5 and it was running until now

Themaster312, 2025-01-02:
In the last update you just had to add in Condor that 2.5 works

sdetweil, 2025-01-02:
https://openweathermap.org/api

Themaster312, 2025-01-02:
How do I change in the config?

Themaster312, 2025-01-02:
now it shows me something again but somehow everything it shows can't be right in Germany 27 degrees no way

sdetweil, 2025-01-02:
you have to put in lat/lon now.. location/LocationID don't work

Themaster312, 2025-01-02:
Wäre i find lat lon

Themaster312, 2025-01-02:
or just rename in the config

Themaster312, 2025-01-02:
ok, I have it now had a mental block

Can be closed

sdetweil, 2025-01-02:
https://docs.magicmirror.builders/
see the modules section

lat/lon is your latitude and longitude on the world

Themaster312, 2025-01-02:
I have already solved the problem and found out

eskudo12791, 2025-01-03:
Does the lat/ long needed to be added to only the config, or also within the weather module

sdetweil, 2025-01-03:
in the weather module, or any other that needs it

sdetweil, 2025-01-03:
see this post in other issue

https://github.com/MagicMirrorOrg/MagicMirror/issues/3574#issuecomment-2423544713

Themaster312, 2025-01-04:
Issue are solved ,i close it

closed 2025-01-04 by rejas