MagicMirrorOrg/MagicMirror #3676: electron tests are broken

khassel, 2025-01-02:
Before adding a PR I tested it on my fork. Without any changes related to electron and without any dependency updates all electron tests are failing.

Found a hint on the actions site: 
> ubuntu-latest pipelines will use ubuntu-24.04 soon. For more details, see https://github.com/actions/runner-images/issues/10636

After changing the workflow back to ubuntu-22.04 the tests are working again.

Will provide this as hotfix.

I suspect in ubuntu-24.04 dependencies are missing needed by electron. We have enough time to solve the issue in 24.04 because 22.04 is supported for another 2 years.

sdetweil, 2025-01-02:
where do you see the failures?  I get notices for each build action, but haven't had any today

khassel, 2025-01-02:
e.g. https://github.com/khassel/MagicMirror/actions/runs/12589061748

maybe your account or MagicMirrorOrg account are still using 22.04 as latest, they do it step by step, but will be a question of time ...

khassel, 2025-01-03:
I removed the "ready (coming with next release) label because
- the tests are unrelated to releases
- we still have to get the electron test running on ubuntu-24.04

PR #3680 (2025-01-05): electron tests: fixes for running under new github image ubuntu-24.04
and replace xserver with labwc, fixes #3676
commit 8fdd865c (2025-01-05): electron tests: fixes for running under new github image ubuntu-24.04 (#3680)

and replace xserver with labwc, fixes #3676
closed 2025-01-12 by khassel