MagicMirrorOrg/MagicMirror #3678: Calendar events broadcast are lost

sdetweil, 2025-01-03:
2.30. added a function to fix a bug, but clipped the events before broadcast  incorrectly

Pollard-J, 2025-01-14:
Is this why i have no events on my calendar?

sdetweil, 2025-01-14:
yes, did you put on the fix?

Pollard-J, 2025-01-14:
I don't think so. How do I do that?

sdetweil, 2025-01-14:
always look at the troubleshooting section of the forum

https://forum.magicmirror.builders/topic/19282/version-2-30-0-calendar-fix-for-clipping-broadcast-events

Pollard-J, 2025-01-14:
Thank you all fixed now. I didn't know about that. I pretty much leave my MM alone. I only get problems everytime I update it, i think due to whatever I do.

sdetweil, 2025-01-14:
i am going to add this to my upgrade script
if 2.30.0 and an *Ext3 module is installed, install the fix

PR #3679 (2025-01-03): Fix calendar clipping before broadcast
-fixes #3678
commit 75dbe671 (2025-01-03): Fix calendar clipping before broadcast (#3679)

-fixes #3678
closed 2025-04-01 by rejas