MagicMirrorOrg/MagicMirror #3683: enhancement: allow optional new install section for modules for automatic installation

khassel, 2025-01-05:
I wrote the module [MMM-ModInstall](https://gitlab.com/khassel/MMM-ModInstall) which checks for an `install` object beside the module config and does install the module if the module dir not exists.

Example:

```js
let config = {
  modules: [
    {
      module: "MMM-ModInstall",
    },
    {
      module: "MMM-Flights",
      install: {
        url: "https://gitlab.com/khassel/MMM-Flights.git",
        branch: "",
        command: "",
        timeout: 300; // in sec.
      },
      config: {
        // module config ...
      },
    },
  ],
};
```

Problems with this approach
- you need to install `MMM-ModInstall` before you can use it (bad for automation)
- `MMM-ModInstall` must be the first module in `modules` section to install following modules

I think auto installation should be a feature of core mm. This would solve the above problems und the `config.js` would contain all values for (re)installing the modules.

It is an additional feature and will not break any existing stuff.

I would be happy to move the ~40 lines of code into mm core if accepted.

sdetweil, 2025-01-05:
have you tried my mmm-config, auto discovery and config. add/remove from running(disable)
can differentiate between in config or not

the hard part is getting from github to modules folder
and then constructive config

bugsounet, 2025-01-05:
Hi, @khassel 

> @sdetweil says: you are welcome to write in french and i will translate and still ask questions

So, now i will write only in French

----

Je ne suis pas d'accord d'avoir un installateur de module intégré à MagicMirror².
Personnellement, je ne pense pas que la base devrait assumer ce genre de technique. C'est à cela que sert l'écosystème tiers. (meme reaction qu'[ici ](https://github.com/MagicMirrorOrg/MagicMirror/issues/3447#issuecomment-2106311788), je suis les regles)
Comme votre module `MMM-ModInstall` sert a ceci, c'est donc parfait :)

@bugsounet

sdetweil, 2025-01-05:
i think if we were going to do something here, 

- we should use the 3rd party list as a catalog 
- have an install button(if not already installed),  (node helper does this step by reading the 3rd party list and matching names,  insert button, not change UI with a rewrite again
- button fires module url and action to node_helper,
- does the clone/npm ...,   xxx ( which does the batch components)
- then present a form for the config items(defaults set from module defaults)   
- and saves it into config.js and                 
- offers to restart MagicMirror to activate that module

after start , continuing config/update is another problem

xxx I think is what your [MMM-ModInstall](https://gitlab.com/khassel/MMM-ModInstall) module does
I think the info would come from the catalog)

or more simply, use the catalog and do the clone/npm 
and restart mm for access to the module info by another part (like my MMM-Config does, in update mode, detecting changes to the modules folder)

that other part would take on config and continuing config.

khassel, 2025-01-05:
@sdetweil 

my intention is only to get things automated. I want to have a `config.js` with all informations needed to get mm running from scratch with only `config.js`, not more, no gui, there are other modules (MMM-Config, mmpm, ...) for this.

So having a `config.js` with the needed infos (mainly repo url) you could start mm with this and it will automatically start up without doing manual installs.

bugsounet, 2025-01-05:
Je préfère me taire et plus rien dire.
Comme dans 99% des cas, vous ne l'écoutez pas :/
Donc faites comme vous voulez!

sdetweil, 2025-01-05:
i like the idea but adding a new section to many no longer maintained modules makes it hard to get there.

many of the modules won't be visual or functional with the default values.. (our cal and weather are not),
so you started and .... nothing, or the dreaded Loading.....

I know a couple of authors that have had medical problems and will never be back, and don't have the capability to transfer ownership.

don't get me wrong, I want this too.. anyone use HomeAssistant Community Store (not to mention the integrations install......) ,  but HA has a strong UI base

khassel, 2025-01-05:
> i like the idea but adding a new section to many no longer maintained modules makes it hard to get there.

thats not the point, as a mm user I would become the ability to setup such an "automated" `config.js` for (nearly) every module. In 95% you have only to look up and insert the repo url and the rest works out of the box (using default branch and running `npm install --omit=dev` as default install command).

KristjanESPERANTO, 2025-01-05:
The idea of only having to put my config.js into a fresh installation for it to work is tempting. This should work for all my modules.

"no longer maintained modules" should not stop us from introducing improvements. I'm also in favour of not bloating the core unnecessarily, but I think this is worth including.

Something like an app store or module store would of course also be great, but that is a separate topic which should be dealt with independently of this PR.

@bugsounet Sorry, I don't understand French.

sdetweil, 2025-01-05:
@khassel , ok, so the idea is to insert a new module with JUST the URL info(for your module), so that your module can do the clone and install. 

@KristjanESPERANTO   do you save any of the info building the 3rd party list  as json,  name, description, url

khassel, 2025-01-05:
@sdetweil 

yes, example:

normal module config as we do it all the time:

```js
    {
      module: "MMM-Flights",
      config: {
        // module config ...
      },
    },
```

now add a new section if you want it to be installed automatic, so it looks like:

```js
    {
      module: "MMM-Flights",
      install: {
        url: "https://gitlab.com/khassel/MMM-Flights.git",
      },
      config: {
        // module config ...
      },
    },
```

You could do it with most modules including unmaintained ones ...

KristjanESPERANTO, 2025-01-05:
> @KristjanESPERANTO do you save any of the info building the 3rd party list as json, name, description, url

Yes. you can find
- here https://github.com/KristjanESPERANTO/MagicMirror-3rd-Party-Modules/blob/main/docs/data/modules.json
- and here https://kristjanesperanto.github.io/MagicMirror-3rd-Party-Modules/data/modules.json

sdetweil, 2025-01-07:
I decided to put action where my mouth is

new module installer
see
https://forum.magicmirror.builders/topic/19306/new-module-installer

sdetweil, 2025-01-20:
that approach wouldn't work, 
integrated into MMM-Config

KristjanESPERANTO, 2025-03-22:
I still think that this would be a good addition. @rejas, what do you think?

bugsounet, 2025-03-22:
@KristjanESPERANTO :

For me, it would be to integrate an HTTP configuration interface directly into the MM² core for this (on a different port).
This opens up several possibilities:
* display your 3rd-party module site
* install a module
* delete a module
* set the default configuration of the module to be installed
* edit a module's configuration
* restart/shutdown MM²
* restart/shutdown the system
* open a terminal
* consult MM² logs
* provide system information (RAM, disk space, etc.)
* ...

You know what i want to say, I have already code this,
and you have see some screenshot ;)

KristjanESPERANTO, 2025-03-22:
I would definitely agree to go in this direction (let's call it a graphical backend). But this PR is about something else: Simplified module installation based on the config file. This single function would already be helpful now and would not contradict a later introduced graphical backend. Let's try to stick to this single function here and have discussions about a backend there: #3749.

closed 2025-06-07 by github-actions[bot]