MagicMirrorOrg/MagicMirror #3687: Weather modules don't update/grow stale

JHenzi, 2025-01-11:
## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Raspberry Pi 4.

**Node Version**: 22.

**MagicMirror² Version**: latest

**Description**: Weather forecast (pirate weather) doesn't update without relaunching or reloading the page. Today, Tomorrow and the next day are all wrong after one day (today is yesterday, tomorrow is today). I don't think the values are updating either. Believe the current weather / temperature is wrong after it ages as well as it didn't update for a day.

**Steps to Reproduce**: Use Pirate Weather with this config:

```
	{
			module: "weather",
			position: "top_right",
			config: {
				weatherProvider: "pirateweather",
				type: "current",
				lat: 39.132721,
				lon: -84.628098,
				apiKey: "---",
			}
		},
		{
			module: "weather",
			position: "top_left",
			header: "Weather Forecast",
			config: {
				weatherProvider: "pirateweather",
				type: "forecast",
				lat: 39.132721,
				lon: -84.628098,
				apiKey: "---",
			}
		}
```

**Expected Results**: Live data.

**Actual Results**: Stale data

**Configuration**: See above

**Additional Notes**: Recently switched to Pirate Weather. Don't see any errors in the console.

sdetweil, 2025-01-11:
the documentation says there is a 10 minute cycle time, before next fetch of the data

weather is done all in browser, do if you waited more than 10 minutes, open the develoer window
ctrl-shift-i
select the console tab 
and put a unique part of the module name in the filter field (weath for this module) 

see if there are any error messages (usually in red) 
and copy/paste them here

JHenzi, 2025-01-11:
No error messages in console. Happy to find any other logs I guess. Maybe better to explain the weather forecast will say:

- today
- tomorrow
- mon

When today is Monday.

sdetweil, 2025-01-11:
today is saturday

JHenzi, 2025-01-12:
No error messages in console. Happy to find any other logs I guess. Maybe better to explain the weather forecast will say:

today
tomorrow
sun

When today is Sunday.

sdetweil, 2025-01-12:
ok, I changed the title to inlcude pirateweather, 
someone will have to look at it..

JHenzi, 2025-01-13:
I'm not seeing this still happening since I reported it. Users are migrating to pirate weather, maybe someone else will have this happen.

sdetweil, 2025-01-20:
maybe it was a Pirateweather problem

khassel, 2025-01-31:
so this was a temporary problem? If yes, can we close this issue? @JHenzi

Ryanclbryant, 2025-02-03:
I've been experiencing this also; it's not a temporary problem but seemingly intermittent.
It seems related to scheduling updates. When the mirror process starts, updates perform as expected, but over time, the process eventually stops updating without error.

Very painfully slowly working on identifying where or why this is coming up; likely something that Pirateweather is including in the response rarely.

merasil, 2025-02-06:
Also same problem. Def. not temporary

dangherve, 2025-02-27:
For information still have the same problem always show "Loading..."

dangherve, 2025-02-27:
do not know if it will help but i have the same problem with weatherbit and envcanada

Ryanclbryant, 2025-02-27:
For others experiencing this issue; I’m curious if you also use a third party module which consumes calendar and weather notifications?

This problem seemingly disappeared when I stopped dbroadcasting calendar events, and returned when I started them again. I suspect this is related to issue #3678

dangherve, 2025-02-27:
I finaly test several provider the only one woking is openweathermap but the hourly type is not working

dangherve, 2025-02-27:
not what the screenshot says

JHenzi, 2025-03-10:
> For others experiencing this issue; I’m curious if you also use a third party module which consumes calendar and weather notifications?
> 
> This problem seemingly disappeared when I stopped broadcasting calendar events, and returned when I started them again. I suspect this is related to issue [#3678](https://github.com/MagicMirrorOrg/MagicMirror/issues/3678)

I'm not using anything like that. 

The behavior described above makes sense - after "some time" Magic Mirror needs restarted to re-enable weather updates.

papinist, 2025-05-02:
I have this issue too. I'm using pirateweather, data updates when starting MM, and then seems to update for a couple of days.. then it's stuck with days and weather data.

sdetweil, 2025-05-02:
weather is all in browser

can you install MMM-LogExt or MMM-Logging
to send the browser logs to npm start
logs, so we have them in one place

papinist, 2025-05-04:
I just switched to MMM-PirateSkyForecast that I just found. Will see if data in this will remain stuck too.

jbaboval, 2025-05-18:
I have this issue as well. Things are generally fine for a few days, and then it's usually a day or several before I notice the weather stopped updating.

My system is in that state right now. In the error log at the same time as the last successful weather update I have the following:

```
[2025-05-14 11:01:32.757] [ERROR] TypeError: invalid media type
    at Object.parse (/home/pi/MagicMirror/node_modules/content-type/index.js:126:11)
    at setCharset (/home/pi/MagicMirror/node_modules/express/lib/utils.js:252:28)
    at ServerResponse.send (/home/pi/MagicMirror/node_modules/express/lib/response.js:175:32)
    at cors (/home/pi/MagicMirror/js/server_functions.js:58:8)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async /home/pi/MagicMirror/js/server.js:82:41
[2025-05-14 11:01:32.764] [ERROR] (node:1296) UnhandledPromiseRejectionWarning: TypeError: invalid media type
    at Object.parse (/home/pi/MagicMirror/node_modules/content-type/index.js:126:11)
    at setCharset (/home/pi/MagicMirror/node_modules/express/lib/utils.js:252:28)
    at ServerResponse.send (/home/pi/MagicMirror/node_modules/express/lib/response.js:175:32)
    at ServerResponse.json (/home/pi/MagicMirror/node_modules/express/lib/response.js:279:15)
    at ServerResponse.send (/home/pi/MagicMirror/node_modules/express/lib/response.js:163:21)
    at cors (/home/pi/MagicMirror/js/server_functions.js:62:7)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async /home/pi/MagicMirror/js/server.js:82:41
(Use `electron --trace-warnings ...` to show where the warning was created)
[2025-05-14 11:01:32.765] [ERROR] (node:1296) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
[2025-05-14 11:01:53.124] [ERROR] TypeError: invalid media type
    at Object.parse (/home/pi/MagicMirror/node_modules/content-type/index.js:126:11)
    at setCharset (/home/pi/MagicMirror/node_modules/express/lib/utils.js:252:28)
    at ServerResponse.send (/home/pi/MagicMirror/node_modules/express/lib/response.js:175:32)
    at cors (/home/pi/MagicMirror/js/server_functions.js:58:8)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async /home/pi/MagicMirror/js/server.js:82:41
[2025-05-14 11:01:53.125] [ERROR] (node:1296) UnhandledPromiseRejectionWarning: TypeError: invalid media type
    at Object.parse (/home/pi/MagicMirror/node_modules/content-type/index.js:126:11)
    at setCharset (/home/pi/MagicMirror/node_modules/express/lib/utils.js:252:28)
    at ServerResponse.send (/home/pi/MagicMirror/node_modules/express/lib/response.js:175:32)
    at ServerResponse.json (/home/pi/MagicMirror/node_modules/express/lib/response.js:279:15)
    at ServerResponse.send (/home/pi/MagicMirror/node_modules/express/lib/response.js:163:21)
    at cors (/home/pi/MagicMirror/js/server_functions.js:62:7)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async /home/pi/MagicMirror/js/server.js:82:41
[2025-05-14 11:01:53.125] [ERROR] (node:1296) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
```

sdetweil, 2025-05-19:
generally this means that the data fetch failed
and the contents were not as expected, and no effective error recovery was in place.

it appears the module tried again, and encountered the same error again, and again

sdetweil, 2025-05-19:
looking at the MM code for handling this request 
```
[2025-05-14 11:01:32.757] [ERROR] TypeError: invalid media type
    at Object.parse (/home/pi/MagicMirror/node_modules/content-type/index.js:126:11)
    at setCharset (/home/pi/MagicMirror/node_modules/express/lib/utils.js:252:28)
    at ServerResponse.send (/home/pi/MagicMirror/node_modules/express/lib/response.js:175:32)
    at cors (/home/pi/MagicMirror/js/server_functions.js:58:8)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async /home/pi/MagicMirror/js/server.js:82:41
```
in performWebRequest   (modules/default/utils.js)

we don't check for an error from fetch , and ASSUME it is good
```
async function performWebRequest (url, type = "json", useCorsProxy = false, requestHeaders = undefined, expectedResponseHeaders = undefined, basePath = "/") {
	const request = {};
	let requestUrl;
	if (useCorsProxy) {
		requestUrl = getCorsUrl(url, requestHeaders, expectedResponseHeaders, basePath); <--- using cors proxy
	} else {
		requestUrl = url;
		request.headers = getHeadersToSend(requestHeaders);
	}
	const response = await fetch(requestUrl, request);  <----- after here (via proxy url)
	const data = await response.text();

	if (type === "xml") {
		return new DOMParser().parseFromString(data, "text/html");
	} else {
		if (!data || !data.length > 0) return undefined;  <----- would cause error on res.send in cors

		const dataResponse = JSON.parse(data);
		if (!dataResponse.headers) {
			dataResponse.headers = getHeadersFromResponse(expectedResponseHeaders, response);
		}
		return dataResponse;
```

cors also doesn't check 
```
			Log.log(`cors url: ${url}`);
			const response = await fetch(url, { headers: headersToSend }); <------

			for (const header of expectedReceivedHeaders) {
				const headerValue = response.headers.get(header);
				if (header) res.set(header, headerValue);
			}
			const data = await response.text();
			res.send(data);  <--- doesn't return fetch error 
```

sdetweil, 2025-05-19:
and we didn't check the response header for accept/json or xml matching the type 

cors got invoked because it was in the url but it didnt throw, etc,,,