MagicMirrorOrg/MagicMirror #3704: enhancement: allow specifying a location name when using the /onecall endpoint in the OpenWeatherMap module

efkann, 2025-01-25:
### What problem do you want to solve with this change?

Hi,

With the recent changes in OpenWeatherMap, version 2.5 has been deprecated, and version 3.0 is now the default.

But the `/onecall` endpoint returns a timezone rather than a location name. Currently, the returned timezone is used to set the location in the module.

https://github.com/MagicMirrorOrg/MagicMirror/blob/9c9a5359dd80eb92332849ac54279e374fabb3ef/modules/default/weather/providers/openweathermap.js#L36-L38

https://github.com/MagicMirrorOrg/MagicMirror/blob/9c9a5359dd80eb92332849ac54279e374fabb3ef/modules/default/weather/providers/openweathermap.js#L56-L58

### What do you think is the correct solution?

Since the user already provides the `lat` and `lon` values in the configuration, I see no harm in adding an optional field like `locationName` to use when the endpoint is set to `/onecall`.

What do you think ?

I’d be happy to submit a PR to implement these changes.

### Participation

- [x] I am willing to submit a pull request for this change.

### Additional comments

I understand that there's already a `location` field in the config, but this field is consumed with other endpoints.  So, I believe it wouldn't be a good practice to reuse the same field for a different purpose in the `/onecall `endpoint.

sdetweil, 2025-01-25:
just use the module header option
{
module:"weather",
position:"....",
header:"whatever you want",
config:{
   appendLocationToHeader:false,
    other config parms
  }
)

efkann, 2025-01-25:
> just use the module header option { module:"weather", position:"....", header:"whatever you want", config:{ appendLocationToHeader:false, other config parms } )

Didn't know this was possible, thanks.

closed 2025-01-25 by efkann