MagicMirrorOrg/MagicMirror #3707: Jährliche Termine werden nicht angezeigt

chris47803, 2025-01-29:
### Environment

Hallo.

Ich habe einen neuen MagicMirror für einen Freund aufgesetzt.

Dort werden einzelne, wöchentlich und monatliche Termine angezeigt.

Jährliche Termin werden nicht angezeigt.

Ich habe meine calendar.js auf seinen PI kopiert (auf meinem PI wird alles korrekt angezeigt).

Trotzdem sehe ich keine jährlichen Termine.

Auch nicht wenn ich meinen Outlook Kalnder einbinde.

Ich weiß nicht mehr weiter.

LG Chris

### Which start option are you using?

npm run start

### Are you using PM2?

Yes

### Module

calendar

### Have you tried disabling other modules?

- [x] Yes
- [ ] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

<details>
<summary>Configuration</summary>

```
<!-- Paste your configuration here. Don't forget to remove any sensitive information! -->
```
</details>

```js
<!-- Paste relevant code here -->
```

Steps to reproduce the issue:

### What did you expect to happen?

Hallo.

Ich habe einen neuen MagicMirror für einen Freund aufgesetzt.

Dort werden einzelne, wöchentlich und monatliche Termine angezeigt.

Jährliche Termin werden nicht angezeigt.

Ich habe meine calendar.js auf seinen PI kopiert (auf meinem PI wird alles korrekt angezeigt).

Trotzdem sehe ich keine jährlichen Termine.

Auch nicht wenn ich meinen Outlook Kalnder einbinde.

Ich weiß nicht mehr weiter.

LG Chris

### What actually happened?

Hallo.

Ich habe einen neuen MagicMirror für einen Freund aufgesetzt.

Dort werden einzelne, wöchentlich und monatliche Termine angezeigt.

Jährliche Termin werden nicht angezeigt.

Ich habe meine calendar.js auf seinen PI kopiert (auf meinem PI wird alles korrekt angezeigt).

Trotzdem sehe ich keine jährlichen Termine.

Auch nicht wenn ich meinen Outlook Kalnder einbinde.

Ich weiß nicht mehr weiter.

LG Chris

### Participation

- [x] I am willing to submit a pull request for this change.

sdetweil, 2025-01-29:
you didn't paste the configuration, AND started conversation in the forum

sdetweil, 2025-01-29:
we need to see the calendar configuration. 
xxx out the private urls, but don't change anything else.

if this is v2.30.0 did you put on the fixes for cslendar?
https://forum.magicmirror.builders/topic/19282/version-2-30-0-calendar-fix-for-clipping-broadcast-events?page=6

khassel, 2025-01-29:
@chris47803 can you please edit the issue content and translate it to english? You can use google translate or a similar tool if needed.

Thanks.

khassel, 2025-03-18:
- no reaction
- no translation
- typical trouble shooting should be done in the forum

closed 2025-03-18 by khassel