MagicMirrorOrg/MagicMirror #3708: [Bug] if a node_helper has an init function, the 'this' pointer is not set

sdetweil, 2025-01-30:
### Environment

System information:
### SYSTEM:   manufacturer: Apple Inc.; model: Macmini; virtual: false
### OS:       platform: darwin; distro: macOS; release: 14.5; arch: arm64; kernel: 23.5.0
### VERSIONS: electron: 32.2.1; used node: 20.18.0; installed node: 18.20.5; npm: 10.8.2; pm2: 5.4.2
### OTHER:    timeZone: America/Denver; ELECTRON_ENABLE_GPU: undefined 

### Which start option are you using?

npm run start

### Are you using PM2?

No

### Module

None

### Have you tried disabling other modules?

- [x] Yes
- [ ] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

```
        init: function(){
                Log.log(this.name + " is in init!");
        },
```

from the doc 
```text
init()
This method is called when a module gets instantiated. In most cases you do not need to subclass this method.
```

### What did you expect to happen?

I expected a log message (using my brithdaylist module)

birthdaylist is in init

### What actually happened?

undefined is in init

### Participation

- [x] I am willing to submit a pull request for this change.

khassel, 2025-01-30:
Had same problem in a module last week. Found a solution and can provide a PR (but it's o.k. for me if you want to implement this) ...

sdetweil, 2025-01-30:
no, if you got it, go

sdetweil, 2025-01-31:
i tried to open this issue from my phone, but it wouldnt let me NOT fill in the system info section. so i had to get out my laptop, run MagicMirror again to get it

KristjanESPERANTO, 2025-01-31:
> i tried to open this issue from my phone, but it wouldnt let me NOT fill in the system info section. so i had to get out my laptop, run MagicMirror again to get it

Which browser did you use? I just checked with Chrome on Android and the system info section was editable.

sdetweil, 2025-01-31:
I did not HAVE  the info on my phone, and my computer was put away due to travel. 

I figured I could edit it later to paste in the info, which is not relevant  to this problem.

closed 2025-04-12 by github-actions[bot]