MagicMirrorOrg/MagicMirror #3718: [Bug] Calendar module: maximumEntries not applied from global level

vdborght, 2025-02-05:
### Environment

MagicMirror² version: 2.30
Node version: 23.7
npm version: 10.9.2
Platform: Raspberry Pi OS 6.6.74+rpt-rpi-v8

### Which start option are you using?

npm run start

### Are you using PM2?

Yes

### Module

calendar

### Have you tried disabling other modules?

- [ ] Yes
- [x] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

After a few years, I decided to upgrade my MagicMirror instance to the latest.
To have a "clean" system, I reinstalled the whole system (from OS to MM to modules) and restored my MM configs and CSS.

Doing this I noticed that only 10 events were being retrieved per calendar.
A bit of debug shows that I had maximumEntries only defined on the "config" level and this doesn't get applied

After defining maximumEntries on the "config/calendar" level (for each calendar separately), it now works. 

### What did you expect to happen?

Defining maximumEntries on the "config" level of the calendar module is enough to apply to all calendars and 100 events are shown and defined.

### What actually happened?

Only 10 entries are shown instead of the defined 100.

### Participation

- [ ] I am willing to submit a pull request for this change.

vdborght, 2025-02-05:
Seems that "fix clippings" does a lot more than just fixing clipped titles ;-)
I'll be closing this bug report.

Thank you!
Peter

closed 2025-02-05 by vdborght

sdetweil, 2025-02-05:
it does NOT clip off the events collected before broadcast