MagicMirrorOrg/MagicMirror #3720: [Feature Request] Calendar: Wait / delay module

Coernel82, 2025-02-10:
### Prerequisites

- [x] I am running the latest version of MagicMirror², and know that this feature is not available now.
- [x] I know my issue is not related to a third-party module.
- [x] I have searched for [existing issues](https://github.com/MagicMirrorOrg/MagicMirror/issues) that already include this feature request, without success.

### Describe the Feature Request

# My usecase for the default calendar module is as follow:

I load the module twice:

- Calendar module 1: shows daily routine appointments and all other appointments out of 4 or 5 WebDav calendars on my Nextcloud for the current day only
- Calendar module 2: includes all calendars except the routine appointments and displays 8 entries

Also installed I have the [MMM-NextCloud-Tasks](https://github.com/Coernel82/MMM-NextCloud-Tasks) module (which I forked and improved).

The problem which regularly occurs is that the calendar modul displays an error. Admittedly I have not looked in the logs but I am pretty sure that this is caused by a timeout (lost socket connection) as this is what happened when loading my tasks from multiple urls on the same Nextcloud.

For [MMM-NextCloud-Tasks](https://github.com/Coernel82/MMM-NextCloud-Tasks) I solved this by retrying to fetch the data up to x times (I think 5 or 8) in a row with a delay of 20 seconds each time. This solves the problem.

# The cause of the error
I have the suspicion that too many WebDav request overload the Nextcloud.

# My idea / solution
Make the modules somehow wait for each other. This could be done with a simple delay which has to be set up manually in each config. What I have done as another workaround is to set the fetchInterval to different high prime numbers - so with longer runtime the point of time where data is fetched drifts apart and therefore solves the problem for some weeks or months till the prime numbers hit the same timespot again.
Of course the better solution is to send a socket notification: MMM-BLOCK_WEBDAV, fetch data, and then MMM-UNBLOCK_WEBDAV - and each module can receive this and only do its job when there is no MMM-BLOCK_WEBDAV. However an unloaded module cannot receive a socket notification I guess.

### Describe the Use Case

see above

### Describe Preferred Solution

see above

### Describe Alternatives

see above

### Participation

- [ ] I am willing to submit a pull request for this change.

sdetweil, 2025-02-10:
calendar should retry. (fetchInterval time)

but this is a bigger problem over many different sources. weather has the same potential

sdetweil, 2025-02-10:
is your nextcloud self hosted?

Coernel82, 2025-02-10:
Yes it retries however that is all 30 minutes so the error could persist for 30 minutes. I could set it for 5 Minutes but of course I want to keep the Nextcloud workload down.
My Nextcloud is self hosted on a raspberry Pi 4 using the Nextcloud pi project.

closed 2025-04-26 by github-actions[bot]