MagicMirrorOrg/MagicMirror #3722: [Bug] GBM Errors on Raspberry Pi 5

Dyptorden, 2025-02-16:
### Environment

MagicMirror² version: [not shown in logs]
Node version: 22.14.0
npm version: 11.1.0
Platform: Raspberry Pi 5 (Debian GNU/Linux 12, arm64, kernel: 6.6.74+rpt-rpi-2712)

### Description
When running MagicMirror on Raspberry Pi 5, the application logs multiple GBM (Generic Buffer Management) related errors. While the application continues to function (calendar and news fetchers work), the console is flooded with these errors.

### Environment
- **Hardware**: Raspberry Pi 5
- **OS**: Debian GNU/Linux 12 (arm64)
- **Kernel**: 6.6.74+rpt-rpi-2712
- **Electron**: 32.2.7
- **Node**: 22.14.0
- **npm**: 11.1.0

### Error Messages
```
[ERROR:gbm_wrapper.cc(74)] Failed to get fd for plane.: No such file or directory (2)
[ERROR:gbm_wrapper.cc(257)] Failed to export buffer to dma_buf: No such file or directory (2)
```
These errors repeat multiple times in the log.

### Full Log
```
[2025-02-16 20:38:16.080] [LOG]   Launching application. 
[2025-02-16 20:38:16.948] [INFO]  System information:
### SYSTEM:   manufacturer: ; model: ; virtual: false
### OS:       platform: linux; distro: Debian GNU/Linux; release: 12; arch: arm64; kernel: 6.6.74+rpt-rpi-2712
### VERSIONS: electron: 32.2.7; used node: 22.14.0; installed node: 22.14.0; npm: 11.1.0; pm2: 5.4.3
### OTHER:    timeZone: Europe/Bucharest; ELECTRON_ENABLE_GPU: undefined
```

### Expected Behavior
MagicMirror should start without GBM-related errors.

### Actual Behavior
Application starts and functions, but continuously logs GBM errors related to buffer management and DMA operations.

### Impact
While the application appears to function normally, the log files are being filled with these errors, which could make it difficult to identify other potential issues and might impact system resources.

### Attempted Workarounds
1. Added the following to `config.js` after the `units` section and before `modules`:
```javascript
electronOptions: {
    args: ["--disable-gpu"]
}
```
This modification did not resolve the GBM errors.

2. Tried running with GPU disabled via environment variable:
```bash
ELECTRON_DISABLE_GPU=1 npm run start
```
The GBM errors persisted despite this change.

### Additional Notes
This might be related to how Electron 32.2.7 handles graphics on the new Raspberry Pi 5 architecture. Similar issues have been observed in other Electron applications on the RPi 5.

### Would appreciate help with
1. Confirming if this is a known issue with Electron 32.2.7 on RPi 5
2. Guidance on whether this needs to be addressed at the MagicMirror or Electron level
3. Suggested fixes or workarounds beyond disabling GPU acceleration

### Which start option are you using?

npm run start

### Are you using PM2?

No

### Module

None

### Have you tried disabling other modules?

- [ ] Yes
- [x] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

I did no change in the config as I just cloned the repo.

Steps to reproduce the issue:
Just started the app using **_npm run start_**

### What did you expect to happen?

Expected to have no errorrs in the app log (terminal)

### What actually happened?

got the errors:
```
[2025-02-16 20:47:19.656] [LOG]   Launching application. 
[2025-02-16 20:47:20.369] [INFO]  System information:
### SYSTEM:   manufacturer: ; model: ; virtual: false
### OS:       platform: linux; distro: Debian GNU/Linux; release: 12; arch: arm64; kernel: 6.6.74+rpt-rpi-2712
### VERSIONS: electron: 32.2.7; used node: 22.14.0; installed node: 22.14.0; npm: 11.1.0; pm2: 5.4.3
### OTHER:    timeZone: Europe/Bucharest; ELECTRON_ENABLE_GPU: undefined 
[25894:0216/204720.829091:ERROR:gbm_wrapper.cc(74)] Failed to get fd for plane.: No such file or directory (2)
[25894:0216/204720.829242:ERROR:gbm_wrapper.cc(257)] Failed to export buffer to dma_buf: No such file or directory (2)
[25894:0216/204720.829413:ERROR:gbm_wrapper.cc(74)] Failed to get fd for plane.: No such file or directory (2)
[25894:0216/204720.829489:ERROR:gbm_wrapper.cc(257)] Failed to export buffer to dma_buf: No such file or directory (2)
[25894:0216/204720.829562:ERROR:gbm_wrapper.cc(74)] Failed to get fd for plane.: No such file or directory (2)
[25894:0216/204720.829621:ERROR:gbm_wrapper.cc(257)] Failed to export buffer to dma_buf: No such file or directory (2)
[25894:0216/204720.829710:ERROR:gbm_wrapper.cc(74)] Failed to get fd for plane.: No such file or directory (2)
[25894:0216/204720.829766:ERROR:gbm_wrapper.cc(257)] Failed to export buffer to dma_buf: No such file or directory (2)
[25894:0216/204720.829837:ERROR:gbm_wrapper.cc(74)] Failed to get fd for plane.: No such file or directory (2)
[25894:0216/204720.829907:ERROR:gbm_wrapper.cc(257)] Failed to export buffer to dma_buf: No such file or directory (2)
[25894:0216/204720.829993:ERROR:gbm_wrapper.cc(74)] Failed to get fd for plane.: No such file or directory (2)
[25894:0216/204720.830029:ERROR:gbm_wrapper.cc(257)] Failed to export buffer to dma_buf: No such file or directory (2)
[25894:0216/204720.830119:ERROR:gbm_wrapper.cc(74)] Failed to get fd for plane.: No such file or directory (2)
[25894:0216/204720.830191:ERROR:gbm_wrapper.cc(257)] Failed to export buffer to dma_buf: No such file or directory (2)
[25894:0216/204720.831676:ERROR:gbm_wrapper.cc(74)] Failed to get fd for plane.: No such file or directory (2)
[25894:0216/204720.831741:ERROR:gbm_wrapper.cc(257)] Failed to export buffer to dma_buf: No such file or directory (2)
[25894:0216/204720.831814:ERROR:gbm_wrapper.cc(74)] Failed to get fd for plane.: No such file or directory (2)
[25894:0216/204720.831887:ERROR:gbm_wrapper.cc(257)] Failed to export buffer to dma_buf: No such file or directory (2)
[25894:0216/204720.831975:ERROR:gbm_wrapper.cc(74)] Failed to get fd for plane.: No such file or directory (2)
[25894:0216/204720.832011:ERROR:gbm_wrapper.cc(257)] Failed to export buffer to dma_buf: No such file or directory (2)
[25894:0216/204720.832097:ERROR:gbm_wrapper.cc(74)] Failed to get fd for plane.: No such file or directory (2)
[25894:0216/204720.832166:ERROR:gbm_wrapper.cc(257)] Failed to export buffer to dma_buf: No such file or directory (2)
[25894:0216/204720.832252:ERROR:gbm_wrapper.cc(74)] Failed to get fd for plane.: No such file or directory (2)
[25894:0216/204720.832306:ERROR:gbm_wrapper.cc(257)] Failed to export buffer to dma_buf: No such file or directory (2)
[25894:0216/204720.832375:ERROR:gbm_wrapper.cc(74)] Failed to get fd for plane.: No such file or directory (2)
[25894:0216/204720.832442:ERROR:gbm_wrapper.cc(257)] Failed to export buffer to dma_buf: No such file or directory (2)
[25894:0216/204720.832528:ERROR:gbm_wrapper.cc(74)] Failed to get fd for plane.: No such file or directory (2)
[25894:0216/204720.832565:ERROR:gbm_wrapper.cc(257)] Failed to export buffer to dma_buf: No such file or directory (2)

```

### Participation

- [ ] I am willing to submit a pull request for this change.

sdetweil, 2025-02-16:
yes, these appear to be errors in electron which have no negative side effect other than the message

we have disabled gpu by default, as most users dont need it, and it eliminates a lot of error messages

the environment variable to turn it on is so changed 
``` 
export ELECTRON_ENABLE_GPU=1
```

bugsounet, 2025-02-16:
@sdetweil: Really ... you make me despair...
You can't even read this issue anymore...

HEY: The guy has already tested it!

bugsounet, 2025-02-16:
@sdetweil : retry again (same player)

sdetweil, 2025-02-16:
well, he tried DISABLE_GPU  which is the default

bugsounet, 2025-02-16:
@Dyptorden 

Unfortunately, this problem is not related to MagicMirror.
It is related to [electron](https://github.com/electron/electron) (MagicMirror's main engine)

be aware that according to electron coders, this is more of a warning than an error.
On the other hand, by disabling the use of the GPU (DISABLE_GPU), your MagicMirror will be without hardware acceleration and only with the CPU.

This is very good for use without the use of 3rd party modules
in other cases, your MagicMirror will run very very slowly
(so I advise you to enable GPU acceleration)

KristjanESPERANTO, 2025-02-16:
@Dyptorden On a Raspberry Pi 4 I can confirm this behaviour when running `npm run start`. But when I run `npm run start:wayland` this warning does not appear. I have been using the wayland start option by default for some time now and it works fine for me.

Can you check this on your system?

KristjanESPERANTO, 2025-05-18:
@sdetweil Since using the Wayland start option seems to solve the problem and there is no feedback from @Dyptorden we can close this issue, right?

sdetweil, 2025-05-18:
yes

closed 2025-05-18 by sdetweil