MagicMirrorOrg/MagicMirror #3723: [Bug] broadcastPastEvents=true leads to past events resulting as happening TODAY

SimoneMSR, 2025-02-19:
### Environment

MagicMirror² version: 2.29.0
Node version: 22.12.0
npm version: 10.9.0
Platform: Windows

### Which start option are you using?

npm run server

### Are you using PM2?

No

### Module

calendar

### Have you tried disabling other modules?

- [ ] Yes
- [x] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

<details>
<summary>Configuration</summary>

```
var config
= {
  address: "0.0.0.0",
  ipWhitelist: [],
  modules: [
    {
      module: "clock",
      hiddenOnStartup: true,
      position: "top_left",
      config: {}
    },
    {
      module: "MMM-CalendarExt3Agenda",
      hiddenOnStartup: true,
      position: "top_left",
      header: "My Agenda",
      config: {
        instanceId: "basicCalendar",
        showMiniMonthCalendar: false,
        firstDayOfWeek: 1,
        startDayIndex: -1,
        calendarSet: [
          "us_holiday"
        ]
      }
    },
    {
      module: "weather",
      hiddenOnStartup: true,
      position: "bottom_left",
      config: {
        maxNumberOfDays: 4,
        weatherProvider: "openmeteo",
        lat: 40.776676,
        lon: -73.971321
      }
    },
    {
      module: "weather",
      hiddenOnStartup: true,
      instanceId: "weather-forecast",
      classes: "weather-forecast",
      position: "bottom_left",
      config: {
        type: "forecast",
        weatherProvider: "openmeteo",
        lat: 40.776676,
        lon: -73.971321
      }
    },
    {
      module: "calendar",
      position: "bottom_left",
      hiddenOnStartup: true,
      config: {
        calendars: [
          {
            name: "us_holiday",
            url: "https://ics.calendarlabs.com/76/mm3137/US_Holidays.ics",
            color: "#6f1283"
          }
        ],
        broadcastPastEvents: true,
        pastDaysCount : 30
      }
    },
    {
      module: "MMM-CalendarExt3",
      hiddenOnStartup: true,
      position: "top_center",
      header: "",
      config: {
        mode: "month",
        headerWeekDayOptions: {
          weekday: "narrow"
        }
      }
    },
    {
      module: "MMM-Pir",
      hiddenOnStartup: true,
      position: "top_left",
      config: {
        debug: false,
        Display: {
          timeout: 612000,
          animate: true,
          style: 1,
          colorFrom: "#FF0000",
          colorTo: "#00FF00",
          mode: 5,
          ecoMode: false,
          counter: true,
          lastPresence: true,
          lastPresenceTimeFormat: "LL H:mm",
          availability: true,
          autoDimmer: false,
          xrandrForceRotation: "normal",
          wrandrForceRotation: "normal",
          wrandrForceMode: null,
          waylandDisplayName: "wayland-1",
          relayGPIOPin: 16,
          ddcutil: {
            powerOffCode: "02",
            skipSetVcpCheck: true,
            setPowerRetries: 3
          }
        },
        Pir: {
          mode: 1,
          gpio: 21,
          triggerMode: "H"
        },
        Motion: {
          deviceId: 0,
          captureIntervalTime: 1000,
          scoreThreshold: 100
        },
        Cron: {
          ON: [],
          OFF: [],
          mode: 3
        },
        Touch: {
          mode: 3
        },
        Governor: {
          sleeping: 4,
          working: 2
        },
        Sounds: {
          on: "open.mp3",
          off: "close.mp3"
        }
      }
    },
    {
      module: "MMM-ip",
      position: "bottom_left",
      config: {}
    }
  ],
  useHttps: false,
  httpsPrivateKey: "",
  httpsCertificate: "",
  locale: "it-IT",
  language: "it"
}

/** ************* DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") { module.exports = config }

```
</details>

```js
```

Steps to reproduce the issue:
npm run client

open a browser window in localhost:8080

### What did you expect to happen?

Past events are showed with their own date, on TODAY

### What actually happened?

All past events are showed  as happening TODAY

### Participation

- [x] I am willing to submit a pull request for this change.

sdetweil, 2025-02-20:
there was extensive change done to calendar in 2.30
please test 2.30

note the clipping fix posted in the Troubleshooting section on the forums

note there is one change since then for recurring issue viewed from more western timezones than scheduled

SimoneMSR, 2025-03-04:
I have tested with version 2.30.0, with clipping enabled, but I see the same bahaviour

sdetweil, 2025-03-04:
can you show me the output of

git branch

in the MagicMirror folder

sdetweil, 2025-03-04:
also can you find the event in your ics data

from terminal window 
curl -sL the_cslendar_url > somefile.txt

edit somefile.txt and find the vevent that describes it
paste vevent:begin to vevent:end (all in caps)

SimoneMSR, 2025-03-05:
$ git fetch origin pull/3679/head:_fix_clipping
```
remote: Enumerating objects: 1, done.
remote: Counting objects: 100% (1/1), done.
remote: Total 1 (delta 1), reused 1 (delta 1), pack-reused 0 (from 0)
Unpacking objects: 100% (1/1), 113 bytes | 11.00 KiB/s, done.
From https://github.com/MagicMirrorOrg/MagicMirror
 * [new ref]           refs/pull/3679/head -> _fix_clipping
```
$ git switch _fix_clipping
```
Switched to branch '_fix_clipping'
M       .gitignore

```
$ restart MM
```
bash: restart: command not found
```

Here's an example of event
```
BEGIN:VEVENT
SUMMARY:Presidents' Day
DTSTART;VALUE=DATE:20240219
DTEND;VALUE=DATE:20240219
LOCATION:United States
DESCRIPTION:Visit https://calendarlabs.com/holidays/us/presidents-day.php to know more about Presidents' Day. \n\n Like us on Facebook: http://fb.com/calendarlabs to get updates
UID:6704d93cca27d1728371004@calendarlabs.com
DTSTAMP:20241008T070324Z
STATUS:CONFIRMED
TRANSP:TRANSPARENT
SEQUENCE:0
END:VEVENT
```

sdetweil, 2025-03-05:
@SimoneMSR 

you must also do
npm run install-mm 

>restart MM

however you run MagicMirror , restart it
npm start, pm2 restart x

thanks for event
what timezone are you in?

SimoneMSR, 2025-03-05:
> [@SimoneMSR](https://github.com/SimoneMSR)
> 
> thanks for event what timezone are you in?

I am in CET time zone

sdetweil, 2025-03-06:
hm...
I don't see that event added 
even when maximumNumberOfDays:500 is set

can I see your config for calendar,  xxx out the url

SimoneMSR, 2025-03-07:
> hm... I don't see that event added even when maximumNumberOfDays:500 is set
> 
> can I see your config for calendar, xxx out the url

That's what I am saying in this DOC [issue](https://github.com/MagicMirrorOrg/MagicMirror-Documentation/issues/290). I had to use `pastDaysCount` option

here's my config

```
    {
      module: 'calendar',
      position: 'bottom_left',
      hiddenOnStartup: true,
      config: {
        calendars: [
          {
            name: 'us_holiday',
            url: 'https://ics.calendarlabs.com/76/mm3137/US_Holidays.ics',
            color: '#6f1283'
          }
        ],
        broadcastPastEvents: true,
        maximumNumberOfDays : 1000,
        pastDaysCount : 30,
      }
    },
```

SimoneMSR, 2025-03-07:
The strange thing is that broadcasted past events seem to have the correct time, as rendered by CalendarExt3 (which uses calendar notification). 

In the image attached, you see a lot of "Today" on le left, but events are not all in "today", in the calendar on the right.

[image]

sdetweil, 2025-03-07:
the event you gave me is last year , so it should never be shown

SimoneMSR, 2025-03-08:
I ubderstand. This is strange. Isn't that a recurring event? That's however a holyday...

sdetweil, 2025-03-08:
while the holiday is recurring, the date is different every year. the event shown is not recurring, as it has no RRULE clause

calendar labs makes a new ics

SimoneMSR, 2025-03-08:
the problem howecer happens also with event from this other calendar (as shown by the imae above)

https://gcatholic.org/calendar/ics/2025-it-General-F.ics?v=3

sdetweil, 2025-03-08:
i will look at the other calendar source content

sdetweil, 2025-03-08:
this is what I  see from that calendar, US MST and then CET, no change with your config for calendar
(using the develop branch, which has the clipping fix and symbols fix on calendar.. nothing else

broadcastPastEvents: true is the default

[image]
[image]

sdetweil, 2025-03-08:
does that icon mean hmmmm??  never seen it before

SimoneMSR, 2025-03-09:
To me (🎉) is a celebration, in this case of the possibility to rendere past events correctly in MM!

SimoneMSR, 2025-03-09:
Really? So pastDayCount does not have a meaning anymore?

sdetweil, 2025-03-09:
well, the doc is incomplete

pastDaysCount is only used to advise receivers of the event broadcast that the user wants to limit the total past days used

sdetweil, 2025-03-16:
I have added text to the documentation for this property (pastDaysCount) in the next release, closing now as this was a misunderstanding of the the property use

closed 2025-03-16 by sdetweil