MagicMirrorOrg/MagicMirror #3734: [Bug] MM does not show up via pm2 since git pull update

Denn2705, 2025-03-03:
### Environment

MagicMirror² version: 2.30.0
Node version: 22.14.0
npm version: 10.9.2
Platform: Raspberry pi zero 2W , Bookworm OS latest Update/Upgrade package

Hi all, since i did the update git-pull command, my MM Installation is just booting up into the desktop environment. The PM2 logs show the following Issue. Could someone help my what is going wrong since the update?

Error Output PM2 Log:
`/home/pi/.pm2/logs/mm-out.log last 15 lines:
0|mm       | > magicmirror@2.30.0 start
0|mm       | > npm run start:x11
0|mm       |
0|mm       |
0|mm       | > magicmirror@2.30.0 start:x11
0|mm       | > DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js
0|mm       |
0|mm       |
0|mm       | > magicmirror@2.30.0 start
0|mm       | > npm run start:x11
0|mm       |
0|mm       |
0|mm       | > magicmirror@2.30.0 start:x11
0|mm       | > DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js
0|mm       |

0|mm  | [2025-03-03 20:09:58.493] [ERROR] App threw an error during load
0|mm  | [2025-03-03 20:09:58.503] [ERROR] Error: Cannot find module 'undici'
0|mm  | Require stack:
0|mm  | - /home/pi/MagicMirror/js/app.js
0|mm  | - /home/pi/MagicMirror/js/electron.js
0|mm  |     at Module._resolveFilename (node:internal/modules/cjs/loader:1055:15)
0|mm  |     at s._resolveFilename (node:electron/js2c/browser_init:2:115476)
0|mm  |     at Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
0|mm  |     at Module._load (node:internal/modules/cjs/loader:908:27)
0|mm  |     at c._load (node:electron/js2c/node_init:2:13672)
0|mm  |     at Module.require (node:internal/modules/cjs/loader:1122:19)
0|mm  |     at require (node:internal/modules/helpers:130:18)
0|mm  |     at Object.<anonymous> (/home/pi/MagicMirror/js/app.js:15:40)
0|mm  |     at Module._compile (node:internal/modules/cjs/loader:1256:14)
0|mm  |     at Module._extensions..js (node:internal/modules/cjs/loader:1311:10)
0|mm  |     at Module.load (node:internal/modules/cjs/loader:1098:32)
0|mm  |     at Module._load (node:internal/modules/cjs/loader:945:12)
0|mm  |     at c._load (node:electron/js2c/node_init:2:13672)
0|mm  |     at Module.require (node:internal/modules/cjs/loader:1122:19)
0|mm  |     at require (node:internal/modules/helpers:130:18)
0|mm  |     at Object.<anonymous> (/home/pi/MagicMirror/js/electron.js:4:14)
0|mm  | A JavaScript error occurred in the main process
0|mm  | Uncaught Exception:
0|mm  | Error: Cannot find module 'undici'
0|mm  | Require stack:
0|mm  | - /home/pi/MagicMirror/js/app.js
0|mm  | - /home/pi/MagicMirror/js/electron.js
0|mm  |     at Module._resolveFilename (node:internal/modules/cjs/loader:1055:15)
0|mm  |     at s._resolveFilename (node:electron/js2c/browser_init:2:115476)
0|mm  |     at Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
0|mm  |     at Module._load (node:internal/modules/cjs/loader:908:27)
0|mm  |     at c._load (node:electron/js2c/node_init:2:13672)
0|mm  |     at Module.require (node:internal/modules/cjs/loader:1122:19)
0|mm  |     at require (node:internal/modules/helpers:130:18)
0|mm  |     at Object.<anonymous> (/home/pi/MagicMirror/js/app.js:15:40)
0|mm  |     at Module._compile (node:internal/modules/cjs/loader:1256:14)
0|mm  |     at Module._extensions..js (node:internal/modules/cjs/loader:1311:10)
0|mm  |     at Module.load (node:internal/modules/cjs/loader:1098:32)
0|mm  |     at Module._load (node:internal/modules/cjs/loader:945:12)
0|mm  |     at c._load (node:electron/js2c/node_init:2:13672)
0|mm  |     at Module.require (node:internal/modules/cjs/loader:1122:19)
0|mm  |     at require (node:internal/modules/helpers:130:18)
0|mm  |     at Object.<anonymous> (/home/pi/MagicMirror/js/electron.js:4:14)
0|mm  | [1157:0303/201003.563428:ERROR:object_proxy.cc(576)] Failed to call method: org.freedesktop.DBus.StartServiceByName: object_path= /org/freedesktop/DBus: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
`

Thanks in advance!
BR Dennis

### Which start option are you using?

npm run start:x11

### Are you using PM2?

Yes

### Module

None

### Have you tried disabling other modules?

- [x] Yes
- [ ] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

<details>
<summary>Configuration</summary>

```
<!-- Paste your configuration here. Don't forget to remove any sensitive information! -->
```
</details>

```js
<!-- Paste relevant code here -->
```

Steps to reproduce the issue:

### What did you expect to happen?

Run the MM same like before the git-pull update

### What actually happened?

PM2 does not start up, pi just boot into the desktop env.

### Participation

- [ ] I am willing to submit a pull request for this change.

Denn2705, 2025-03-03:
running "npm run install-mm" again fixed the issue

closed 2025-03-03 by Denn2705

sdetweil, 2025-03-03:
yes, if you fo a git pull, code changes and a new
npm install is required 

see https://docs.magicmirror.builders/getting-started/upgrade-guide.html