MagicMirrorOrg/MagicMirror #3736: add @electron/rebuild to the dependencies

sdetweil, 2025-03-05:
### What problem do you want to solve with this change?

modules which require hardware access (pir, sensors, ...)
use low level libraries that require binary compatibility with the node engine running. either 
nodejs in serveronly mode
or electron in full mode

during module install the author must choose which mode the module will be used in. most 'assume' full electron mode

the older versions of er(electron rebuild) worked when loaded into the module node_modules tree, it somehow found the electron version. you could also specify a parameter.

the newer er versions do not successfully find the electron version unless it is located in the same node_modules folder

so module install fails, and things change every quarter on our release, and the default upgrade doesnt take this impact into account

my upgrade script does an npm install for all  modules with dependencies. i wrote a post-install script used by most of the modules, which checks for er in the MagicMirror node_modules folder, and if not present installs it (now with the wrong npm name )

its not our users fault we keep whacking the system every quarter. and its not the module authors fault either

interestingly, in the serveronly mode, i haven't seen reported issues with chromium/firefox ering used as the browser, even when er done on latest electron level

### What do you think is the correct solution?

ship the matching electron-rebuild  library with magicmirror

we use it in test, so only have to move from devdependencies list to dependencies list

### Participation

- [x] I am willing to submit a pull request for this change.

khassel, 2025-03-05:
> have to move from devdependencies list to dependencies list

if we do this (I'm still not convinced ...) we should move it beside `electron` to `optionalDependencies`:

```json
	"optionalDependencies": {
		"electron": "^35.0.0"
	},
```

sdetweil, 2025-03-05:
yes, you are correct. that is really the right place

sdetweil, 2025-03-05:
most of our module developers are not developers,  and our users know even less.

whats the harm in trying to add a little stability while we keep current

KristjanESPERANTO, 2025-03-05:
Which modules have the mentioned problem?

> the newer er versions do not successfully find the electron version unless it is located in the same node_modules folder

Modules like [MMM-KeyBindings](https://github.com/shbatm/MMM-KeyBindings/blob/master/postinstall.sh) already solved it 2018 (maybe even before) this way:

```
ELECVERSION=$(node -pe "require('../../node_modules/electron/package.json').version")
./node_modules/.bin/electron-rebuild -f -w evdev,udev -v $ELECVERSION
```

This works with old and new versions of electron-rebuild. I'm using this module with an evdev device without any issues.

Reading the electron version and than use it as parameter for electron-rebuild should work for other modules too, right? So maybe we only should tell the developers how to do it?

sdetweil, 2025-03-05:
find the modules w node_helpers, and use onoff,gpio

pir, touch, navigate,

sdetweil, 2025-03-05:
i would never have figured out your env 
and i dont think it fair to put that on every 
module maker to figure out each release

KristjanESPERANTO, 2025-03-05:
> keybinding has my postinstall

Does it? It looks very different to yours.

> the wrong version of eb

What's the right one? MMM-KeyBindings already uses @electron/rebuild.

> and screws any module later

If someone (like me) uses only one module with electron-rebuild there is no problem?

> find the modules w node_helpers, and use onoff,gpio

Yeah, but on my setup I don't use the gpio. So it's difficult to test it by myself.

> whats the harm in trying to add ...

The harm could be that if we introduce it, module developers will build on it. If it turns out later that it wasn't the best solution, we'll have problems taking it back.

I am not against improving the situation. But before discussing a concrete solution, I would just like to understand better in which scenarios and why the problems occur. Perhaps there are other solutions.

But it shouldn't be up to me, I'm a non-expert when it comes to electron-rebuild. So your decision shouldn't depend on me. I'm just asking questions to understand it better 😀

sdetweil, 2025-04-14:
in the meantime, I am adding on install, upgrade and restore