MagicMirrorOrg/MagicMirror #3738: [Feature Request] {{ Calendar - events: ad a specific CSS class according to expression or condition }}

SimoneMSR, 2025-03-07:
### Prerequisites

- [x] I am running the latest version of MagicMirror², and know that this feature is not available now.
- [x] I know my issue is not related to a third-party module.
- [x] I have searched for [existing issues](https://github.com/MagicMirrorOrg/MagicMirror/issues) that already include this feature request, without success.

### Describe the Feature Request

I would like to customize the calendar so to apply specific CSS to events, via config.js, or via a module method, or notification.

### Describe the Use Case

Say that I can add the class "past" to all past events, I could hide past events in the original calendar widget, but still have them rendered in the CalendarExt3 (which uses broadcasts from calendar module)

### Describe Preferred Solution

`config.js`
```
....calendars: [
    {
      name : "my_calendar"
      url: "...",
      cssClasses : {
        past : "e.startDate < new Date()"
      }
    }
}
```
using a module method:
```
MM.getModules().find(m=>m.name=="calendar").setStyleCallback(event=>{
  var classes = {}
  classes.past = event.startDate < new Date()
  return classes
})
```

using a module motification:
```
myModule.sendNotification("calendar_setStyleCallback", {callback : event=>{
  var classes = {}
  classes.past = event.startDate < new Date()
  return classes
}})
```

### Participation

- [x] I am willing to submit a pull request for this change.

sdetweil, 2025-03-07:
how is this different from customEvents?

sdetweil, 2025-03-07:
you can also modify another modules content in code

SimoneMSR, 2025-03-08:
I guess customEvents configuration cannot access the startDate of an event. Am I right?

closed 2025-05-17 by github-actions[bot]