MagicMirrorOrg/MagicMirror #3743: [Bug] {{ Calendar: broadcasted events number is always limited by maximumEntries  }}

SimoneMSR, 2025-03-18:
### Environment

MagicMirror² version:  2.30
Node version: 22.12.0
npm version: 10.9.0
Platform: Linux

### Which start option are you using?

npm run server

### Are you using PM2?

Yes

### Module

calendar

### Have you tried disabling other modules?

- [ ] Yes
- [x] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

<details>
<summary>Configuration</summary>

```
/*************** AUTO GENERATED BY ROMEO MODULE ***************/

var config = 
{
  address: '0.0.0.0',
  ipWhitelist: [],
  modules: [
    {
      module: 'clock',
      hiddenOnStartup: true,
      position: 'top_left',
      config: {}
    },
    {
      module: 'MMM-CalendarExt3Agenda',
      hiddenOnStartup: true,
      position: 'top_left',
      header: 'My Agenda',
      config: {
        instanceId: 'basicCalendar',
        showMiniMonthCalendar: false,
        firstDayOfWeek: 1,
        startDayIndex: -1,
        calendarSet: [
          'us_holiday'
        ]
      }
    },
    {
      module: 'weather',
      hiddenOnStartup: true,
      position: 'bottom_left',
      config: {
        maxNumberOfDays: 4,
        weatherProvider: 'openmeteo',
        lat: 40.776676,
        lon: -73.971321
      }
    },
    {
      module: 'weather',
      hiddenOnStartup: true,
      instanceId: 'weather-forecast',
      classes: 'weather-forecast',
      position: 'bottom_left',
      config: {
        type: 'forecast',
        weatherProvider: 'openmeteo',
        lat: 40.776676,
        lon: -73.971321
      }
    },
    {
      module: 'calendar',
      position: 'bottom_left',
      hiddenOnStartup: true,
      config: {
        calendars: [
          {
            name: 'us_holiday',
            url: 'https://ics.calendarlabs.com/76/mm3137/US_Holidays.ics',
            color: '#6f1283'
          },
          {
            name: 'religious',
            url: 'https://gcatholic.org/calendar/ics/2025-it-General-F.ics?v=3'
          }
        ],
        broadcastPastEvents: true
      }
    },
    {
      module: 'MMM-CalendarExt3',
      hiddenOnStartup: true,
      position: 'top_center',
      header: '',
      config: {
        mode: 'month',
        headerWeekDayOptions: {
          weekday: 'narrow'
        }
      }
    },
    {
      module: 'MMM-Pir',
      hiddenOnStartup: true,
      position: 'top_left',
      config: {
        debug: false,
        Display: {
          timeout: 612000,
          animate: true,
          style: 0,
          colorFrom: '#FF0000',
          colorTo: '#00FF00',
          mode: 0,
          ecoMode: false,
          counter: true,
          lastPresence: true,
          lastPresenceTimeFormat: 'LL H:mm',
          availability: true,
          autoDimmer: false,
          xrandrForceRotation: 'normal',
          wrandrForceRotation: 'normal',
          wrandrForceMode: null,
          waylandDisplayName: 'wayland-1',
          relayGPIOPin: 16,
          ddcutil: {
            powerOffCode: '02',
            skipSetVcpCheck: true,
            setPowerRetries: 3
          }
        },
        Pir: {
          mode: 1,
          gpio: 21,
          triggerMode: 'H',
          skipToggleScreen: true
        },
        Motion: {
          deviceId: 0,
          captureIntervalTime: 1000,
          scoreThreshold: 100
        },
        Cron: {
          ON: [
            {
              dayOfWeek: [
                1
              ],
              hour: 1,
              minute: 0
            }
          ],
          OFF: [
            {
              dayOfWeek: [
                1
              ],
              hour: 1,
              minute: 9
            }
          ],
          mode: 0
        },
        Touch: {
          mode: 3
        },
        Governor: {
          sleeping: 4,
          working: 2
        },
        Sounds: {
          on: 'open.mp3',
          off: 'close.mp3'
        }
      }
    },
    {
      module: 'MMM-ip',
      position: 'bottom_left',
      config: {}
    }
  ],
  useHttps: false,
  httpsPrivateKey: 'modules/MMM-Styler/ssl/domain.key',
  httpsCertificate: 'modules/MMM-Styler/ssl/domain.crt',
  locale: 'en-US'
}

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}

```
</details>
<details>
<summary>Steps</summary>
Steps to reproduce the issue:
npm run server
navigate to localhost:8080
</details>

### What did you expect to happen?

The payload of module notification `CALENDAR_EVENTS` would contain all the events in the calendar.

### What actually happened?

 I get only 10 events per calendar. And they are all in the past, thus, in the "calendar" module I see the coming events, but in CalendarExt3 I do not see anything

[image]

### Participation

- [ ] I am willing to submit a pull request for this change.

sdetweil, 2025-03-18:
did you install the clipping fix? 

https://forum.magicmirror.builders/topic/19282/version-2-30-0-calendar-fix-for-clipping-broadcast-events

SimoneMSR, 2025-03-18:
Thank you @sdetweil . This fixed the issue!

closed 2025-03-18 by SimoneMSR