MagicMirrorOrg/MagicMirror #3744: [Bug] {{ useHttps - clientOnly won't start }}

SimoneMSR, 2025-03-18:
### Environment

MagicMirror² version: 2.31
Node version: 22.12.0
npm version: 10.9.0
Platform: Linux

### Which start option are you using?

node clientonly --address ... --port ...

### Are you using PM2?

No

### Module

None

### Have you tried disabling other modules?

- [ ] Yes
- [x] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

<details>
<summary>Configuration</summary>

```
/*************** AUTO GENERATED BY ROMEO MODULE ***************/

var config = 
{
  address: '0.0.0.0',
  ipWhitelist: [],
  modules: [
    {
      module: 'clock',
      hiddenOnStartup: true,
      position: 'top_left',
      config: {}
    },
    {
      module: 'MMM-CalendarExt3Agenda',
      hiddenOnStartup: true,
      position: 'top_left',
      header: 'My Agenda',
      config: {
        instanceId: 'basicCalendar',
        showMiniMonthCalendar: false,
        firstDayOfWeek: 1,
        startDayIndex: -1,
        calendarSet: [
          'us_holiday'
        ]
      }
    },
    {
      module: 'weather',
      hiddenOnStartup: true,
      position: 'bottom_left',
      config: {
        maxNumberOfDays: 4,
        weatherProvider: 'openmeteo',
        lat: 40.776676,
        lon: -73.971321
      }
    },
    {
      module: 'weather',
      hiddenOnStartup: true,
      instanceId: 'weather-forecast',
      classes: 'weather-forecast',
      position: 'bottom_left',
      config: {
        type: 'forecast',
        weatherProvider: 'openmeteo',
        lat: 40.776676,
        lon: -73.971321
      }
    },
    {
      module: 'calendar',
      position: 'bottom_left',
      hiddenOnStartup: true,
      config: {
        calendars: [
          {
            name: 'us_holiday',
            url: 'https://ics.calendarlabs.com/76/mm3137/US_Holidays.ics',
            color: '#6f1283'
          },
          {
            name: 'religious',
            url: 'https://gcatholic.org/calendar/ics/2025-it-General-F.ics?v=3'
          }
        ],
        broadcastPastEvents: true
      }
    },
    {
      module: 'MMM-CalendarExt3',
      hiddenOnStartup: true,
      position: 'top_center',
      header: '',
      config: {
        mode: 'month',
        headerWeekDayOptions: {
          weekday: 'narrow'
        }
      }
    },
    {
      module: 'MMM-Pir',
      hiddenOnStartup: true,
      position: 'top_left',
      config: {
        debug: false,
        Display: {
          timeout: 612000,
          animate: true,
          style: 0,
          colorFrom: '#FF0000',
          colorTo: '#00FF00',
          mode: 0,
          ecoMode: false,
          counter: true,
          lastPresence: true,
          lastPresenceTimeFormat: 'LL H:mm',
          availability: true,
          autoDimmer: false,
          xrandrForceRotation: 'normal',
          wrandrForceRotation: 'normal',
          wrandrForceMode: null,
          waylandDisplayName: 'wayland-1',
          relayGPIOPin: 16,
          ddcutil: {
            powerOffCode: '02',
            skipSetVcpCheck: true,
            setPowerRetries: 3
          }
        },
        Pir: {
          mode: 1,
          gpio: 21,
          triggerMode: 'H',
          skipToggleScreen: true
        },
        Motion: {
          deviceId: 0,
          captureIntervalTime: 1000,
          scoreThreshold: 100
        },
        Cron: {
          ON: [
            {
              dayOfWeek: [
                1
              ],
              hour: 1,
              minute: 0
            }
          ],
          OFF: [
            {
              dayOfWeek: [
                1
              ],
              hour: 1,
              minute: 9
            }
          ],
          mode: 0
        },
        Touch: {
          mode: 3
        },
        Governor: {
          sleeping: 4,
          working: 2
        },
        Sounds: {
          on: 'open.mp3',
          off: 'close.mp3'
        }
      }
    },
    {
      module: 'MMM-ip',
      position: 'bottom_left',
      config: {}
    }
  ],
  useHttps: true,
  httpsPrivateKey: 'modules/MMM-Styler/ssl/domain.key',
  httpsCertificate: 'modules/MMM-Styler/ssl/domain.crt',
  locale: 'en-US'
}

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}

```
</details>

Steps to reproduce the issue:
Produce a self-signed certificate with `openssl`, expiring in 3035.

```
node serveronly/ &
DISPLAY=:0 node clientonly/ --address 0.0.0.0 --port 8080 --use-tls
```

### What did you expect to happen?

The client working

### What actually happened?

I get this error

```
Unable to connect to server: (Error: Unable to read config from server (https://0.0.0.0:8080/config/ (self-signed certificate)
```

If I run the above, with `useHttps:false`, everything works.

### Participation

- [ ] I am willing to submit a pull request for this change.

sdetweil, 2025-03-18:
try client w localhost not 0.0.0.0

sdetweil, 2025-03-18:
if you do this from diff machines does it work?
can you use remote browser to server in https?

SimoneMSR, 2025-03-18:
I have updated the `config.js` in this issue.
What I am trying to do is to access the MM from devices (android, iOS)

I can successfully connect to https://www.raspberrypi.local:8080 from browser

But I still get the error

khassel, 2025-03-18:
if I google `unable to verify the first certificate` the problem seems to be an incomplete cert setup, see e.g. https://stackoverflow.com/questions/31673587/error-unable-to-verify-the-first-certificate-in-nodejs

SimoneMSR, 2025-03-19:
This is strange, since In chrome i get the ✅ icon for https. However, to do this I added another certificate CA certofocate to tje browser. 
Should I add the same to MM, in some way?

I have 3 files:
- server.cert (use in MM)
- server.key (use in MM)
- CA.cert (add in browser to avoid "page insecure" alert)

khassel, 2025-03-19:
don't know, try it.

I would add the contents of `CA.pem` into the `server.cert` and try again.

closed 2025-05-31 by github-actions[bot]