MagicMirrorOrg/MagicMirror #3749: [Feature Request] Introduction of a graphical backend

KristjanESPERANTO, 2025-03-22:
### Prerequisites

- [x] I am running the latest version of MagicMirror², and know that this feature is not available now.
- [x] I know my issue is not related to a third-party module.
- [x] I have searched for [existing issues](https://github.com/MagicMirrorOrg/MagicMirror/issues) that already include this feature request, without success.

### Describe the Feature Request

I would like to suggest a feature that could improve the accessibility and usability of MagicMirror²: the introduction of a **graphical backend**. I would like to hear your opinions on this.

My suggestion is that we try to list the advantages and disadvantages of such a backend here. To then weigh up whether we want to go down this route.

Perhaps we could also implement a module system for the backend, just like in the frontend. Then most important features like module management and configuration would be in the core - and the rest could be done by 3rd party modules.

Of course, all of this would be a huge project. That's why I could also imagine that we first agree on the introduction of module management (via a graphical backend). A feature that should definitely be part of the core.

Once this is implemented, we could then discuss whether we should extend the backend further.

Here's a list of arguments and features I've spontaneously collected. Please let me know if there are any important points missing. But we don't have to go into too much detail yet, I'm mainly interested in summarising the discussion (which has already flared up in various places) and its arguments in one place.

We could certainly also be inspired by other good approaches/features from MMM-GoogleAssistant, MMM-Remote-Control, mmpm or MMM-Config.

## Disadvantages

- Increased complexity
- Increased maintenance effort (requires additional maintenance, such as ensuring compatibility, updating features, fixing bugs and writing docs ...)

## Advantages

- Accessibility: A graphical backend would make the platform more user-friendly, especially for non-coding users who might struggle with manual file editing or terminal commands.
- Reduced user errors: By automating certain processes through a graphical backend, common mistakes made by users during manual configuration could be eliminated. This would lead to fewer support requests and issue reports.
- Efficiency: Tasks like module installation, configuration, and system diagnostics would become faster and more streamlined.
- Remote administration: A graphical backend would simplify the ability to manage MagicMirror² remotely over a LAN connection.

## Possible features

### Must have features

- Module Management (install, uninstall and configure modules ...)
- System Control (restart or shut down MagicMirror² or the entire system ...)

### Other features

- Display system information (such as versions, RAM usage, CPU load, and disk space ...)
- Access and review MagicMirror² logs for troubleshooting.
- Open a terminal directly from the graphical backend for advanced operations.

### Describe the Use Case

.

### Participation

- [ ] I am willing to submit a pull request for this change.

khassel, 2025-03-22:
I'm not sure if we should start such a project. You already mentioned some disadvantages and some projects which are doing similar things.

Right out of my head:
- from the manifesto of the project one point was to bring users to "coding" by "forcing them to make changes" and see what happens, with such a graphic UI you give them a blackbox where they "don't have to think"
- who should code this? Looking at the current 3 collaborators I see only @rejas having enough skills but probably not the time (maybe you want code such a project?)
- `This would lead to fewer support requests and issue reports.` well, probably we will get similar requests `I did this in the graphical backend but does not work`
- in any case this should be done in an own git repo (I was already thinking about moving the default modules to an own repo but found no good way so far without breaking things) so such a project could start outside of mm and moved to mm-org later (btw. I would like to have your 3rd-party project under mm-org)

KristjanESPERANTO, 2025-03-23:
> ... you give them a blackbox ...

Yes, I can also see that the (deliberately forced) learning effect for users would be much less. That's why I'm also divided on the subject myself.

Note: I just added "Reducing the learning effect for users" as a disadvantage.

> who should code this?

If the goal is a modular backend, that would not be achievable for me at the moment, especially in terms of time. If the solution is to be “just” module management instead, I think I could do it.

> moving the default modules to an own repo

Interesting, I can see the advantages.

>  I would like to have your 3rd-party project under mm-org

Okay, let's do this.

bugsounet, 2025-03-23:
[image]

sorry :/

KristjanESPERANTO, 2025-03-23:
MMM-GoogleAssistant is a very nice example of a graphical backend, but it also shows how the complexity of the core can increase (and thereby raise the entry barrier for newcomers (devs)).

bugsounet, 2025-03-23:
it's not `MMM-GoogleAssistant` (it's deleted on dead), it's an unknow module (MMM-Bugsounet) ;)

You have right, it's very complex, 50% in core and other in js.
You have to mange translations, access, make an API, ...

I work on it since ~ 3 years now.
Result: 
 * It's very cool to use it 👍 
 * It's like MM² ... it's never finish ;)

mpuff, 2025-03-29:
it would be much easier for new users to use magicmirror.
Not all users have the skills to use nodejs and install modules.

KristjanESPERANTO, 2025-04-07:
> Not all users have the skills to use nodejs and install modules.

[One goal of the project is to motivate people to learn](https://docs.magicmirror.builders/about/manifesto.html). Many cool modules were written by people who have no professional IT background at all. That's a great effect and success.

----

I'm closing the issue for now, as many good arguments for and against have been made. This is not meant to end the discussion. But I don't think we're going to get anywhere unless someone comes up with a very clear way of making configuration more convenient **without** significantly reducing the effect of encouraging learning and engagement.

closed 2025-04-07 by KristjanESPERANTO