MagicMirrorOrg/MagicMirror #3750: [Bug] after install develop on Macos, only back screen from electron, works npm run server

sdetweil, 2025-03-22:
### Environment

[2025-03-22 08:12:49.356] [INFO]  System information:
### SYSTEM:   manufacturer: Apple Inc.; model: Macmini9,1; virtual: false
### OS:       platform: darwin; distro: macOS; release: 15.3.2; arch: arm64; kernel: 24.3.0
### VERSIONS: electron: 32.3.3; used node: 20.18.1; installed node: 20.18.1; npm: 10.8.2; pm2: 5.4.3
### OTHER:    timeZone: America/Phoenix; ELECTRON_ENABLE_GPU: undefined 

### Which start option are you using?

npm run start

### Are you using PM2?

No

### Module

None

### Have you tried disabling other modules?

- [x] Yes
- [ ] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

Steps to reproduce the issue:

used n to install required node version, 22.14.1

checked out develop branch
remove node_modules
npm install

npm start
electron starts, black screen., no index.html load
get this error 
[34408:0322/081249.790421:ERROR:network_service_instance_impl.cc(608)] Network service crashed, restarting service.
reported
https://github.com/electron/electron/issues/43400

checkout master
use n to install 20.18.1
remove node_modules
npm install
npm start, same problem 

develop ok on Windows 11 and Ubuntu 24.04

connect to server from other browser on same or other system ok,
(using sample config) 

### What did you expect to happen?

expected to work 

### What actually happened?

black screen 

[image]

### Participation

- [ ] I am willing to submit a pull request for this change.

khassel, 2025-03-22:
Macos is not mentioned in our documentation (fortunately).

I had similar issues when running on a pi zero 2w where meanwhile the 512MB memory are not sufficient anymore (depending on used os and installed modules).

I think this issue is not related to mm but electron.

sdetweil, 2025-03-23:
something in the save electron config folder
removing
```
~/Library/Application Support/Electron
```
fixes the problem

sdetweil, 2025-03-26:
this also fails on ubuntu same network service failed error.. 

I don't see this in the electron release issues..

sdetweil, 2025-03-26:
on windows, install used electron 35.0.3 and it worked ok
tried on mac, it installed 35.1.0
```
### SYSTEM:   manufacturer: Apple Inc.; model: Macmini9,1; virtual: false
### OS:       platform: darwin; distro: macOS; release: 15.3.2; arch: arm64; kernel: 24.3.0
### VERSIONS: electron: 35.1.0; used node: 22.14.0; installed node: 22.14.0; npm: 10.9.2; pm2: 5.4.3
### OTHER:    timeZone: America/Phoenix; ELECTRON_ENABLE_GPU: undefined 
```

but works ok

sdetweil, 2025-03-29:
possibly this fixes
https://github.com/MagicMirrorOrg/MagicMirror/pull/3754

closed 2025-04-01 by rejas