MagicMirrorOrg/MagicMirror #3759: sceneNext not hiding current scene when wrapping around

themox, 2025-04-04:
### Environment

MagicMirror² version: 2.31.0
Node version: v23.11.0
npm version: 10.2.0
Platform: Ubuntu 20.04 LTS

### Which start option are you using?

npm run start

### Are you using PM2?

Yes

### Module

calendar

### Have you tried disabling other modules?

- [x] Yes
- [ ] No

### Have you searched if someone else has already reported the issue on the forum or in the issues?

- [x] Yes

### What did you do?

<details>
<summary>Configuration</summary>

```
/* Config Sample
 *
 * For more information on how you can configure this file
 * see https://docs.magicmirror.builders/configuration/introduction.html
 * and https://docs.magicmirror.builders/modules/configuration.html
 *
 * You can use environment variables using a `config.js.template` file instead of `config.js`
 * which will be converted to `config.js` while starting. For more information
 * see https://docs.magicmirror.builders/configuration/introduction.html#enviromnent-variables
 */
let config = {
	address: "0.0.0.0",	// Address to listen on, can be:
							// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
							// - another specific IPv4/6 to listen on a specific interface
							// - "0.0.0.0", "::" to listen on any interface
							// Default, when address config is left out or empty, is "localhost"
	port: 8080,
	basePath: "/",	// The URL path where MagicMirror² is hosted. If you are using a Reverse proxy
									// you must set the sub path here. basePath must end with a /
	ipWhitelist: [],	// Set [] to allow all IP addresses
									// or add a specific IPv4 of 192.168.1.5 :
									// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
									// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
									// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false,			// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "",	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "",	// HTTPS Certificate path, only require when useHttps is true

	language: "en",
	locale: "en-US",   // this variable is provided as a consistent location
			   // it is currently only used by 3rd party modules. no MagicMirror code uses this value
			   // as we have no usage, we  have no constraints on what this field holds
			   // see https://en.wikipedia.org/wiki/Locale_(computer_software) for the possibilities

	logLevel: ["INFO", "LOG", "WARN", "ERROR", "DEBUG"], // Add "DEBUG" for even more logging
	timeFormat: 24,
	units: "imperial",

	modules: [

		{
			module: 'MMM-pages',
			config: {
				modules:
					[
					["clock","MMM-CalendarExt2",], 
					["clock", "MMM-CalendarExt2", "newsfeed"], 
					["clock", "MMM-CalendarExt2", "weather"], 
					["MMM-CalendarExt2"],
					],
				fixed: ["updatenotification"],
				rotationTime: 1000*3,
			}
		},

		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_right"
		},
		{
			module: "weather",
			position: "top_right",
			config: {
				weatherProvider: "openmeteo",
				type: "current",
				lat: 39.0068,
				lon: -76.7791
			}
		},
		{
			module: "weather",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				weatherProvider: "openmeteo",
				type: "forecast",
				lat: 39.0068,
				lon: -76.7791
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
		{
			module: 'MMM-CalendarExt2',
			config: {
				scanInterval: 1000*60*60*10,
				rotateInterval: 0,
				deduplicateEventsOn: ["title", "startDate"],
				defaultScene: "scene1",
				
				defaultSet: {
					  view: {
						slotCount: 0,
						display: false,
					}
				},
				scenes: [
					{
						name: "scene1",
						views: ["dayview","weekview"],
					}, 
					{
						name: "scene2",
						views: ["dayview","weekview"],
					},
					{
						name: "scene3",
						views: ["dayview","weekview"],
					},
					{
						name: "scene4",
						views: ["monthview"],
					},
				],
				views:[
					{
						name: "shownone",
						calendars: [""],
					},
					{
						name: "dayview",
						mode: "daily",
						position: "top_left",
						fromNow: "0",
						slotCount: "2",
						hideOverflow: true,
						calendars: ["holidays"],
						
					},
					{
						name: "weekview",
						mode: "week",
						fromNow: "0", 
						slotCount: "2",
						calendars: ["holidays"],
						position: "bottom_bar",
						slotMaxHeight: "150px",
						hideOverflow: true,	
						
					},
					{
						name: "monthview",
						mode: "week",
						title: "Long Range Calendar",
						fromNow: "0",
						slotCount: "4",
						slotMaxHeight: "180px",
						calendars: ["holidays"],
						hideOverflow:true,
						position: "middle_center",
						
					
					},
				],
				calendars: [
		
					{
						name: "holidays",
						url: 'https://calendar.google.com/calendar/ical/en.usa%23holiday%40group.v.calendar.google.com/public/basic.ics',
						color: '#5555FF',
						className: "holidays",
						icon: "bx:bxs-calendar-star",
						},
				],
				notifications: {
					"PAGE_INCREMENT" : {
						exec: "sceneNext",
					},
					"PAGE_DECREMENT" : {
						exec: "scenePrevious",
					},
				}, // end notifications						
			},
		}, // end mmcalenderext2
	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") { module.exports = config; }
```
</details>

```js
<!-- Paste relevant code here -->
```

Steps to reproduce the issue:

I am running this with MMM-Pages and rotating between different calendar scenes with each page rotation.  When pages rotates, it emits a PAGE_INCREMENT notification.  I am catching that with CalendarEXT2, and calling "sceneNext".  If I have n number of scenes, it works fine for the transtion from 1->n, but when transitioning from n to wrap back around to 1, the nth scene is not getting hidden, and every scene thereafter has that nth scene overlayed.

### What did you expect to happen?

I expect that when sceneNext is called from the last scene in a series, the scenes automatically wrap, the last scene is hidden, and the first scene in the series is displayed.

### What actually happened?

If I have n number of scenes, it works fine for the transtion from 1->n, but when transitioning from n to wrap back around to 1, the nth scene is not getting hidden, and every scene thereafter has that nth scene overlayed.

### Additional comments

This worked great for years in this exact configuration but stopped working in the last couple of months.  I hadn't updated in a while, and when I pulled down the latest updates from git, it no longer works.  I have done extensive testing and it appears to be an issue with the Calendar module and how it performs sceneNext, but I haven't figured out which commit version broke it. I have confirmed this behavior does not occur in commit 46fc9bc8e42c8025b743968720abeb7bd048d83a dated in August of 2024.  

### Participation

- [ ] I am willing to submit a pull request for this change.

sdetweil, 2025-04-04:
that may be a problem w MMM-Scenes
not a part  of the base MagicMirror 

i would open isdue epithet the scenes module
also note the MMM-CalendarExt2 is no longer being developed , MMM-CalendarExt3 takes its place

sdetweil, 2025-04-04:
the base calendar foes not send any scene or
page notifications

themox, 2025-04-04:
Calendar isn't sending the notification, it is receiving it and executing sceneNext.  That behavior is clearly in MMM-CalendarExt2.js.  If you don't want to address it, that is one thing.  But after several days of troubleshooting this, it is definitely an issue with this module.

themox, 2025-04-04:
Ok, I just realized I put this on the wrong repo - that is on me.  I will close this and place it on MMM-CalendarExt2.

closed 2025-04-04 by themox

sdetweil, 2025-04-04:
the default calendar only sends one notification

[image]

themox, 2025-04-04:
You are correct.  As I posted above, I clearly did not pay enough attention and accidentally posted the issue to the wrong repo.  You and I were discussing two very different things.